wills106/homeassistant-solax-modbus #815: [Bug]: 'Solis' - RTC sync button doesn't not work

SzosszeNET, 2024-03-19:
### Describe the bug

Latest, ha, solax
Pressing RTC results in an error message.
unsupported type for timedelta seconds component: NoneType

### Integration Version

2024.03.6

### Homeassistant core version

2024.3.1

### Inverter brand

Solis 

### Plugin used

plugin_solis.py

### Serial prefix

000000

### Connection Method

RS-485

### Detailed Error Log

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:239
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 22:07:01 (4 occurrences)
Last logged: 22:11:27

[140055611350592] unsupported type for timedelta seconds component: NoneType
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2319, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2356, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 905, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 139, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 89, in async_press
    res = self.button_info.value_function(0, self.button_info, self._hub.data )
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/const.py", line 254, in value_function_sync_rtc_ymd
    now = datetime.now() + timedelta(seconds=datadict.get('sync_rtc_offset', 0))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: unsupported type for timedelta seconds component: NoneType

simondrake, 2024-03-22:
+1 I'm seeing the same.

fboundy, 2024-03-24:
PR submitted to fix this.

closed 2024-03-25 by wills106

wills106, 2024-03-25:
Just released [2024.03.7b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7b4) with the fix.

PR #819 (2024-03-25): Fix RTC Sync Bug
Fixes #815 by checking that `sync_rtc_offset` returns a number