wills106/homeassistant-solax-modbus #816: [Bug]: Integration stops updating after some time

ruimmsimoes, 2024-03-21:
### Describe the bug

Last few weeks once/ twice a day the integration stops updating values.

This week the problems started occurring more often (many times a day).
I created an automation to reload the integration, and after the reload the data starts flowing again.

The integration had been working without any problem for several months.

### Integration Version

2024.03.6

### Homeassistant core version

2024.3.1

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

ARM 1.35 and DSP 1.33

### Connection Method

Generic USB RS-485 adapter

### Dongle firmware

Serial connection

### Detailed Error Log

```
2024-03-19 15:31:06.273 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2024-03-19 15:31:06.273 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2024-03-19 15:31:06.277 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2024 3
2024-03-19 15:31:06.277 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2024-03-19 15:31:06.277 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax

2024-03-19 15:31:56.682 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-03-19 15:31:56.684 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at /dev/ttyUSB0:0
2024-03-19 15:31:56.684 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type

2024-03-19 15:31:56.781 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number before potential swap: **************
2024-03-19 15:31:56.807 INFO (MainThread) [homeassistant.components.button] Setting up solax_modbus.button
2024-03-19 15:31:56.808 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: SolaxModbusButtonEntityDescription(key='battery_awaken', device_class=None, entity_category=None, entity_registry_enabled_default=False, entity_registry_visible_default=True, force_update=False, icon='mdi:battery-alert-variant', has_entity_name=False, name='Battery Awaken', translation_key=None, translation_placeholders=None, unit_of_measurement=None, allowedtypes=6144, register=86, command=1, blacklist=None, write_method=1, value_function=None, autorepeat=None)
2024-03-19 15:31:56.811 INFO (MainThread) [homeassistant.components.number] Setting up solax_modbus.number
2024-03-19 15:31:56.819 INFO (MainThread) [homeassistant.components.select] Setting up solax_modbus.select
2024-03-19 15:31:56.828 INFO (MainThread) [homeassistant.components.sensor] Setting up solax_modbus.sensor
2024-03-19 15:31:56.845 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x0 seriesnumber to block with start 0x0
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x1d safety_code to block with start 0x0
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x1e matebox_enabled to block with start 0x0
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0x7d
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x7d firmware_DSP_minor_version to block with start 0x7d
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x7e firmware_DSP_hardware_version to block with start 0x7d
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x7f firmware_DSP_major_version to block with start 0x7d
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x80 firmware_ARM_major_version to block with start 0x7d
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x81 firmwareversion_modbustcp_major to block with start 0x7d
2024-03-19 15:31:56.846 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x82 firmwareversion_modbustcp_minor to block with start 0x7d
[... 185 lines cut ...]
2024-03-19 15:35:14.435 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Connection] Not connected[AsyncModbusSerialClient /dev/ttyUSB0:0] : SolaX cannot read holding registers at device 1 position 0x0

2024-03-19 16:54:50.585 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Input/Output] ERROR: No response received after 3 retries : SolaX cannot read holding registers at device 1 position 0x0
NoneType: None
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0x9c: charger_start_time_2
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0x9d: charger_end_time_2
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0x9e: discharger_start_time_2
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0x9f: discharger_end_time_2
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xa1: hotstandby
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xa2: extend_bms_setting
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xa3: battery_heating
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xa4: battery_heating_start_time_1
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xa5: battery_heating_end_time_1
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xa6: battery_heating_start_time_2
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xa7: battery_heating_end_time_2
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xb2: pgrid_bias
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xb5: export_control_factory_limit
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xb6: export_control_user_limit
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xb8: eps_min_soc
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xb9: eps_frequency_test
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xba: inverter_rate_power
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xbb: language
2024-03-19 16:55:27.329 WARNING (MainThread) [custom_components.solax_modbus] SolaX: read failed at 0xbc: mppt_toggle

```

simondrake, 2024-03-22:
+1 same here. I take some solace in the fact I'm not alone, as I was going mad trying to figure out what was wrong on my end. 

@ruimmsimoes - can you share the automation that does the reload, please?

simondrake, 2024-03-22:
I don't see that in the list of versions, on mobile at least, did you download it manually?

ruimmsimoes, 2024-03-22:
Here is the automation I created, it also notifies me every time the integration is restarted:
```
alias: Reiniciar integração SolaX
description: ""
trigger:
  - platform: template
    value_template: >-
      {{ now() - states.sensor.solax_house_load.last_updated >=
      timedelta(minutes=15) }}
condition: []
action:
  - service: homeassistant.reload_config_entry
    metadata: {}
    data:
      entry_id: 277a443271216e367913be7bf7e5794b
  - service: notify.mobile_app_iphonex_de_rui
    metadata: {}
    data:
      title: Reiniciada integração SolaX
      message: >-
        Dados SolaX sem actualização à mais de 15 minutos, integração
        reinicializada.
mode: single
```

To find out the entry_id of your integration, using the file editor, go to file `/homeassistant/.storage/core.config_entries` and search for 'solax_modbus'.

Unfortunately the automation is executing more than twice every hour.

ruimmsimoes, 2024-03-22:
> I had to revert back to 2024.02.7b2 and that version is working fine.

I also can't revert back to that version.
The earliest it lets me revert to is 2024.03.2.
@anton4 How did you manage to revert?

If that version is working then the problem is probably with the changes put in place in version 2024.03.1.

wills106, 2024-03-22:
By default HACS only shows the previous 5 https://hacs.xyz/docs/configuration/options

simondrake, 2024-03-22:
@ruimmsimoes - thanks. 

I changed to using [config_entry_id](https://www.home-assistant.io/docs/configuration/templating/#config-entries). For anybody else seeing this, if you want to validate it's correct- you can run the following:

```bash
cat homeassistant/.storage/core.config_entries | jq -r '.data.entries | .[] | select(.title == '<integration name>' | .entry_id'
```

Thanks @wills106 - I see it now. I've just updated to the newest version, so I'll see if the issue happens again and, if it does, I'll downgrade.

anton4, 2024-03-22:
> I don't see that in the list of versions, on mobile at least, did you download it manually?

Yes, I did it manually, but @wills106 pointed out you can modify the default HACS (5 previous versions) to show earlier versions also.

PatrikTrestik, 2024-03-22:
It also happened to me. Can you confirm that you are using any "remotecontrol" automation?

I had impression that after disabling my remote control it works without issue.
I specifically blame "Enabled No Discharge". Checking the code.

ruimmsimoes, 2024-03-24:
Downgraded to integration version 2024.02.6 and since downgrading the integration only had to be restarted 4 times in two days.
Although not perfect it is much more stable.

Has anyone been able to find out which changes are provoking this behavior?

If you need someone to test any changes I’m available.

@PatrikTrestik I am not using any automations.

lazy-pete, 2024-03-26:
Hi - I e also been seeing a lot of failures. I’m using the solax_Modbus integration as comms for fboundy’s PV_opt integration and found over the past few days that the integration is failing overnight - most evident by a huge drift between the clock sync times I see in the pv_opt dashboard, so I’m guessing the inverter time last seen is when it fails. 

Logs from last night below I think - I’ll try and put in the reload automation and downgrade as above meantime. 

[image]

Logger: homeassistant
Source: helpers/update_coordinator.py:261
First occurred: 03:26:12 (2 occurrences)
Last logged: 03:26:16

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 417, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 183, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 497, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 992, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1113, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1050, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 998, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 548, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/devolo_home_network/sensor.py", line 229, in native_value
    next(
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 261, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
RuntimeError: coroutine raised StopIteration

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:388
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 25 March 2024 at 23:48:14 (61 occurrences)
Last logged: 03:27:55

Failed to connect
Failed to connect [Errno 113] Connect call failed ('192.168.1.25', 502)

This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:393
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 25 March 2024 at 23:48:14 (61 occurrences)
Last logged: 03:27:55

Unable to connect to Inverter at 192.168.1.25:502

lazy-pete, 2024-03-26:
I’m wondering from some of the above whether it’s actually a local networking fail to communicate with the EW11 dongle

simondrake, 2024-03-28:
Downgrading to `2024.03.7b4` didn't fix the issue for me, it would occasionally zero out all the stats and only a HA restart fixes the issue - integration reload didn't. 

I've upgraded to `2024.03.7b4` again, but if anyone has a confirmed working version please share it. 

The fact a HA restart fixes the problem, I'd hazard a guess it's an issue inside HA.

matomatusov, 2024-03-28:
@simondrake 
I think version 2024.02.5 is stable.

mlj11, 2024-03-28:
I have the same problem. The integration stops working and the only soulution is to restart HA. It stops working usually when the remote control is activated and within a few seconds/minutes it stops updating (not always, but very likely). There is probably a delay when reading data, because the entities are not updated at 15s interval, but with some delays (15-25s). I've gone back a few versions and can confirm that I don't see this issue with version 2024.02.6. The remote control works correctly and the updates are done exactly at 15s intervals.

johny-mnemonic, 2024-03-29:
I can confirm the same behavior. Integration works just fine till you try to use `Remotecontrol`. No matter whether it's from automation or manually.

Seems like the integration gets completely stuck as only restart of HA helps. Trying to reload the integration just gets the service stuck as well.

Here is my debug log output:
```
2024-03-29 04:16:56.041 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] Evaluated remotecontrol_trigger: corrected/clamped values: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', 3000.0), ('remotecontrol_reactive_power', 0), ('remotecontrol_duration', 20.0)]
2024-03-29 04:16:56.042 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0x7c: [1, 1, 3000, 0, 0, 0, 20]
2024-03-29 04:17:02.477 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] Evaluated remotecontrol_trigger: corrected/clamped values: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', 3000.0), ('remotecontrol_reactive_power', 0), ('remotecontrol_duration', 20.0)]
2024-03-29 04:17:02.477 DEBUG (MainThread) [custom_components.solax_modbus] ready to repeat button remotecontrol_trigger data: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', 3000.0), ('remotecontrol_reactive_power', 0), ('remotecontrol_duration', 20.0)]
2024-03-29 04:23:04.534 WARNING (MainThread) [homeassistant.components.automation.solax_not_updated] Solax not updated : Already running
```
It basically sets the  `Remotecontrol` and immediately gets stuck. Probably it is somewhere in the autorepeat function.

I have an automation to detect stuck Solax integration and reload the integration, but as you can see on the last line, it reports `Already running` as the service which tries to reload the integration got stuck as well trying to reload the integration...

Downgrade to 2024.02.6 fixed it for me as well.

PatrikTrestik, 2024-03-29:
It also crashed for me while using Remote control. But it works when I use only "Enable Power Control" which does no calculation and only sets "Active Power Target".

I'm going to check code if there is something obvious.

PatrikTrestik, 2024-03-29:
Hang in debugger, good. It seams that there is some locking problem. 
I would like to fix it today if I find a time.

adamhf, 2024-03-29:
@PatrikTrestik I'm seeing the same issue as you, issuing a PowerControl command such as 'No Discharge' causes the integration to lock up.  Reverting to 2024.02.6 seems to have resolved the issue.  I saw it for the first time last night after a reboot, I've kept up to date with HA core releases and versions of this plugin so unsure which combination caused the issue to surface.

PatrikTrestik, 2024-03-29:
https://github.com/wills106/homeassistant-solax-modbus/pull/827 should help.

If you have time please test from my branch even before it is merged.

closed 2024-03-29 by wills106

simondrake, 2024-04-01:
[image]

Still happening for me. @wills106 - can we reopen this?

lazy-pete, 2024-04-01:
> 28120F89-654E-4FC5-81C7-CBBC38048AAB.jpeg (view on web)
>
> Still happening for me. @wills106 <https://github.com/wills106> - can we
> reopen this?

Also still having to reboot HA periodically

P

PR #827 (2024-03-29): Reduce locking to minimum scope (Modbus client only)
Should resolve  #816 
I moved all locks down to self._client....
Checked and remove possible deadlocks.

Tests are good, but if it was deadlock problem it is really hard to test. Please report all problems.