wills106/homeassistant-solax-modbus #817: [Bug]:  Missing information And invalid value on Solax Boost X1

ioniks, 2024-03-24:
### Describe the bug

There is invalid information and missing information :
Grid Export (W) = Measured Power = nearly Pw Power total
Grid Import (W) = always 0
Total Grid Export = Total Grid Import = Inconnu (in french)
Mesured Power L1 to L3 not needed on X1
Inverter Power : "Inconnu"
Inverter Current 
Missing entities : Today Export energy, Today Import energy, House load

### Integration Version

2024.03.6

### Homeassistant core version

2024.3.3

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

 XB333

### Inverter firmware versions

ARM: 147, DSP : 229

### Connection Method

Solax powket Wifi 3.0

### Dongle firmware

Can't check it at this time

### Detailed Error Log

No error

wills106, 2024-03-24:
I can correct Measured power. But there aren't any additional register to add.
There hasn't been an update to the Modbus docs since 2021 for this Inverter.
I have already tried to reach out to SolaX to see if there has been an update to the documentation https://github.com/wills106/homeassistant-solax-modbus/issues/743#issuecomment-1970613253

wills106, 2024-03-25:
I have corrected Measured Power's in [2024.03.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7)

ioniks, 2024-03-26:
@wills106 hi, i have received a doc from solax

[SolaxPower Single Phase External Communication Protocol - X1 Series V3.0_20240326165841.pdf](https://github.com/wills106/homeassistant-solax-modbus/files/14755591/SolaxPower.Single.Phase.External.Communication.Protocol.-.X1.Series.V3.0_20240326165841.pdf)

wills106, 2024-03-26:
That's correct, it's a Non standard protocol https://github.com/wills106/homeassistant-solax-modbus/issues/743#issuecomment-1963746436

ioniks, 2024-04-12:
I have a response from solax no other documentation.

I can help test or retrieve modbus info to find them? If so how can I help?

closed 2025-10-03 by github-actions[bot]