wills106/homeassistant-solax-modbus #824: [Bug]: 'Solis' - 'S6-EH3P10K-H-EU' Could not exclusively lock port

Gregy, 2024-03-27:
### Describe the bug

After upgrade to 2024.03.07 I started getting `Failed to connect [Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable` error. The data gathering becomes spotty eventually stopping.

Reloading the integration doesn't fix the problem. Restarting HASS does and interesting error message is generated during restart.
```
2024-03-27 22:19:02.566 WARNING (Thread-2 (_do_shutdown)) [homeassistant.util.executor] Thread[SyncWorker_27] is still running at shutdown: File "/usr/local/lib/python3.12/threading.py", line 1030, in _bootstrap
    self._bootstrap_inner()
  File "/usr/local/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 89, in _worker
    work_item = work_queue.get(block=True)
```
Is it possible there is some runaway thread holding the lock?

### Integration Version

2024.03.07

### Homeassistant core version

2024.3.3

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

10330602

### Connection Method

RS485 USB adapter

### Detailed Error Log

https://gist.github.com/Gregy/dd5e09717575696f8229bc3241c838a5

Gregy, 2024-03-28:
After downgrade to 2024.03.6 I can still see communication errors in the logs but not locking issues and the integration seems to be able to recover.

https://gist.github.com/Gregy/8f982e87e9c0839bcf86b58f93aa8f4e

Gregy, 2024-03-31:
I have tested `2024.03.8` which doesn't solve the issue.

wills106, 2024-03-31:
Are you trying to access the USB device with anything else at the same time? Like the built in Modbus Integration?

Gregy, 2024-03-31:
No, there is nothing else. But I use relatively short update interval of 4s. Could that be the trigger?

I am testing what is the last version I can run without any errors in the logs. The 2023.12.5 is rock solid, I will continue to go up until I hit issues.

Gregy, 2024-04-01:
The first release which causes random connection dropouts and modbus errors is 2024.02.6. 2024.02.5 is stable.

Gregy, 2024-09-24:
Sorry, I don't know I am still running 2024.02.5. I will try to do an upgrade in near future and let you know if there are still problems.

closed 2024-09-24 by Gregy