wills106/homeassistant-solax-modbus #825: No data being received following update to 2024.4.0b1

markjsmith, 2024-03-28:
### Describe the bug

The integration reports successfully loaded but no entities are available
All was fine in all 2024.3.x releases

### Integration Version

2024.03.07

### Homeassistant core version

2024.4.0b0

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

XAC362

### Inverter firmware versions

Can get later

### Connection Method

Waveshare RS485 to ETH (B)

### Dongle firmware

Can get later 

### Detailed Error Log

```
Logger: homeassistant
Source: core.py:354
First occurred: 08:30:26 (1 occurrences)
Last logged: 08:30:26

Error doing job: Exception in callback <_OneTimeListener <coroutine object SolaXModbusHub.async_init at 0x7fd7c082c310>>(<Event homeas...nt_started[L]>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/core.py", line 1318, in __call__
    self.hass.async_run_hass_job(self.listener_job, event)
  File "/usr/src/homeassistant/homeassistant/core.py", line 830, in async_run_hass_job
    if hassjob.job_type is HassJobType.Callback:
       ^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/backports/functools.py", line 68, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 322, in job_type
    return self._job_type or get_hassjob_callable_job_type(self.target)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 354, in get_hassjob_callable_job_type
    raise ValueError("Coroutine not allowed to be passed to HassJob")
ValueError: Coroutine not allowed to be passed to HassJob
```

### Additional context

I see there is some sort of change to the modbus interface but could not follow what needed to do

Thrasher2020, 2024-03-28:
I have the same issue so it's not just your install. I also looked at the Modbus change as the reason for failure.

Here's the release note : 

The Modbus integration do not allow a configuration with the device only (no entities), this will create an issue, and the entry will not be loaded.

For the special cases (like some custom_components) where it is wanted, please add a dummy entity.

([@janiversen](https://github.com/janiversen) - [#113516](https://github.com/home-assistant/core/pull/113516)) ([documentation](https://rc.home-assistant.io/integrations/modbus))

markjsmith, 2024-03-28:
Yes, I wasn’t sure if it applied here, as we enter the same info into the configuration of the integration as it asks for to create a modbus entry in the config file.

wills106, 2024-03-31:
I have just tried HA 2024.4.0b3 with Integration 2024.03.8 and I don't get any of these issues.

Have you added anything to do with Modbus into configuration.yaml?

As this Integration doesn't use the built in Modbus integration.

markjsmith, 2024-03-31:
No, just upgraded the core. I’ll check no one has disturbed anything in the garage just in case, but it was all good until the upgrade. I’ve since upgraded to b3 too and still the same

Thrasher2020, 2024-03-31:
B3 and latest 2024.03.08 and the situation remains the same. I have no modbus entries in my configuration yaml.

nickedw, 2024-04-02:
+1 - mine appears to have the same issue now too

markjsmith, 2024-04-02:
Latest update fixed it for me! Thanks!
I’ll leave it open while Thrasher confirms his system is fixed too

Thrasher2020, 2024-04-02:
Works for me too 👍

closed 2024-04-02 by markjsmith