wills106/homeassistant-solax-modbus #826: [Bug]: Sofar - HYD 15KTL-3PH Description of Error

wrfz, 2024-03-29:
### Describe the bug

Warning is thrown when HA is restarted

### Integration Version

2024.03.7

### Homeassistant core version

2024.3.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP1ES115N

### Connection Method

Waveshare

### Detailed Error Log

2024-03-29 11:33:52.498 WARNING (MainThread) [homeassistant.core] Task <Task pending name='track time interval 30.0' coro=<SolaXModbusHub.async_refresh_modbus_data() running at /config/custom_components/solax_modbus/__init__.py:330> wait_for=<Future pending cb=[Task.task_wakeup()]> cb=[set.remove()]> was still running after final writes shutdown stage; Integrations should cancel non-critical tasks when receiving the stop event to prevent delaying shutdown

wills106, 2024-09-20:
Are you still getting this issue on newer Integration versions? Or can it be closed down?

wrfz, 2024-09-20:
No more reproducible

closed 2024-09-20 by wrfz