wills106/homeassistant-solax-modbus #831: [Bug]: ? Growatt SPH 10000TL BH-UP No registers beyond 1023

Repsionu, 2024-03-30:
### Describe the bug

Since some days ago all registers greater than 1023 can not be altered anymore. There was a power outage caused by system itself thinking DC voltage from panel was too high- but this time it cut out all house supply :-(

### Integration Version

Latest

### Homeassistant core version

Latest stable

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

TPJ4CH

### Connection Method

RS485-USB

### Detailed Error Log

`2024-03-30 16:43:30.378 INFO (MainThread) [custom_components.solax_modbus.select] writing Growatt select register 1031 value 1
2024-03-30 16:43:40.773 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module 
2024-03-30 16:43:50.730 INFO (MainThread) [custom_components.solax_modbus.select] writing Growatt select register 1029 value 1
2024-03-30 16:43:55.772 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module 
2024-03-30 16:43:57.678 INFO (MainThread) [custom_components.solax_modbus.select] writing Growatt select register 1028 value 1
2024-03-30 16:44:03.942 INFO (MainThread) [custom_components.solax_modbus.select] writing Growatt select register 1026 value 256
2024-03-30 16:44:10.766 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module 
2024-03-30 16:44:25.770 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module 
2024-03-30 16:44:31.661 INFO (MainThread) [custom_components.solax_modbus.select] writing Growatt select register 1082 value 1
2024-03-30 16:44:40.769 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module 
2024-03-30 16:44:43.183 INFO (MainThread) [custom_components.solax_modbus.select] writing Growatt select register 1034 value 1
2024-03-30 16:44:55.774 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module 
2024-03-30 16:45:03.906 INFO (MainThread) [custom_components.solax_modbus.select] writing Growatt select register 1082 value 0
2024-03-30 16:45:10.828 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module `

### Additional context

Also this : read failed at 0x1c is flooding logs. Otherwise all is OK and thanks for the superb component :-)
Best, JR

wills106, 2024-03-31:
Try [2024.03.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.8) as that addresses some communication issues.

Repsionu, 2024-03-31:
This did not fix it but no big deal :-) Waiting for technician to test the inverter as it exports ca. 60W all the time to the grid despite of settings in HA or app either from battery or panels when not needed nor allowed :-( Just was curious as one of not RW registers was **Grid First Time 6**  selector which has state unavailable from 23. march- go figure out if it is related or not...
Thanks for a prompt reply.

DorwinDT, 2024-03-31:
@Repsionu quite offtopic, but please, I have the same inverter and I am not able to connect to it from Solax. 
via "Simply modbus TCP client" I am able to read registers, but solax is not able to connect - no devices are added.
and I don't know if something is missing or so..

Repsionu, 2024-04-01:
@DorwinDT  Sorry but no clue as I use direct rs485<->USB connection. Which ETH port on inverter and which cable wires You use? Did You enable RS485 VPP mode on inverter?  I use ETH3 and cable pins 1 and 5.
Best, JR

DorwinDT, 2024-04-01:
@Repsionu  
So problem solved. it was in inverter settings. 485 poprt has to be set to 'unused' and nopt as recommended in one blog. thank yoiu.

98jan, 2024-08-23:
Hi,

I am using the Growatt Hybrid SPH 10000 with Modbus and an USB to RS485 adapter on my Raspberry Pi 5.

And the log is flooded with following error message for the Growatt Inverter. For the solax integration it is working as a charm.

```
2024-08-23 19:59:40.250 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 712, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf8 in position 2: ordinal not in range(128)
2024-08-23 19:59:55.248 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 712, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf8 in position 2: ordinal not in range(128)
2024-08-23 20:00:10.256 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 712, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf8 in position 2: ordinal not in range(128)
2024-08-23 20:00:25.250 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 712, in treat_address
    val = str(decoder.decode_string(descr.wordcount * 2).decode("ascii"))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf8 in position 2: ordinal not in range(128)
2024-08-23 20:00:40.254 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
2024-08-23 20:00:55.258 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
2024-08-23 20:01:10.257 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
2024-08-23 20:01:25.251 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
2024-08-23 20:01:40.254 WARNING (MainThread) [custom_components.solax_modbus] Growatt: read failed at 0x1c: inverter_module
```

I am using the latest version of the plugin (2024.08.10) and HomeAssistant (2024.8.2).

wills106, 2024-08-25:
Just released [2024.08.12](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.12) removing "Inverter Module" as it seems to be returning something else on some Inverters such as #977 and possibly https://github.com/wills106/homeassistant-solax-modbus/issues/735#issuecomment-2308190992 maybe #979 as well?

98jan, 2024-08-25:
fixed it for me, the error messages stopped and still working as expected.

wills106, 2024-09-15:
Closing down as complete.

closed 2024-09-15 by wills106