wills106/homeassistant-solax-modbus #838: [FR]: Solis S5-EH1P-L - Allow setting of max charge/discharge current

mooky31, 2024-04-02:
### Description

Hello,

I'd like to forbid discharge of the battery to the load when my EV is plugged.
I'm in contact with a Solis engineer, he told me that I should set the discharge current to 0A.
But as far as I can see sensor.solis_battery_charge_current_limit and sensor.solis_battery_discharge_current_limit are only read from the inverter, and not settable from HASS.

### Suggested Solution

Make them settable like number.solis_battery_minimum_soc

### Alternatives

I tried to used the timed mode, but the discharge time is a forced one, eventually leading to discharge to the grid

Gregy, 2024-04-02:
I believe that is already supported. I can set it on my Solis using this integration. You are looking for "Battery Discharge Current". It is defined here: https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_solis.py#L694

mooky31, 2024-04-02:
Indeed it works : when you move this slider it changes the current_limit.
Thank you for the tip, and sorry for the useless FR.

This integration is great, I thought I will have to code myself to manage my inverter.
It is already done, and well better than what I could have made myself !
Many thanks to all the contributors.