wills106/homeassistant-solax-modbus #840: [Bug]: Register 0x111 used twice

martintothgit, 2024-04-03:
### Describe the bug

The register 0x111 seems to be used twice in code:
1) battery_to_ev_charger
2) disch_cut_off_point_different

In Home Assistant Core Log is following message: 

==================================
Logger: custom_components.solax_modbus.sensor
Quelle: custom_components/solax_modbus/sensor.py:117
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 17:40:46 (1 Vorkommnisse)
Zuletzt protokolliert: 17:40:46

holding register already used: 0x111 disch_cut_off_point_different
=====================================

### Integration Version

2024.04.1

### Homeassistant core version

2024.3.3

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM 1.39 DSP 1.40

### Connection Method

SolaX Pocket Wifi 3.0

### Dongle firmware

12.01

### Detailed Error Log

Logger: custom_components.solax_modbus.sensor
Quelle: custom_components/solax_modbus/sensor.py:117
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 17:40:46 (1 Vorkommnisse)
Zuletzt protokolliert: 17:40:46

holding register already used: 0x111 disch_cut_off_point_different

nickedw, 2024-04-04:
I have the same error message. I don’t know if it’s related, but I’m also seeing a 0 value today for Solax house load, everything else is feeding through correctly. I’m seeing correct values on the Solax cloud app for this parameter.

wills106, 2024-04-05:
Just released [2024.04.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.04.2) fixing the duplicate register.

martintothgit, 2024-04-06:
Thanks a lot for quick fixing!
Error message is now gone!

closed 2024-04-06 by martintothgit

HellixFix, 2025-04-05:
After last update got

holding register already used: 0x13bb bms_2_monitoring_version

Growatt MOD 10KTL3-XH(BP)

uspino2, 2025-05-19:
Same here, also Growatt:
```

Logger: custom_components.solax_modbus.sensor
Source: custom_components/solax_modbus/sensor.py:256
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 19:27:31 (1 occurrence)
Last logged: 19:27:31

holding register already used: 0x13bb bms_2_monitoring_version
```