wills106/homeassistant-solax-modbus #841: [Bug]: Periodically energy Reset

MiJo112, 2024-04-03:
### Describe the bug

Since March 17, 0 values have been repeatedly transmitted for energy and power values.
It doesn't matter for the power values, but the energy values naturally mess up the statistics.
I suspect it occurred after an update.
[image]

### Integration Version

2024.04.1

### Homeassistant core version

2024.3.3

### Inverter brand

Solax (Qcells)

### Plugin used

plugin_solax.py

### Serial prefix

H34B15I2

### Inverter firmware versions

ARM 1.38 DSP 1.39

### Connection Method

Waveshare LAN

### Dongle firmware

?

wills106, 2024-04-05:
Do you get any errors in the [error log](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log)?

MiJo112, 2024-04-06:
[error_filter_solax.txt](https://github.com/wills106/homeassistant-solax-modbus/files/14893075/error_filter_solax.txt)

pymodbus:
[image]

There is another energy meter connected to the same modbus, which runs without any problems.

closed 2025-10-03 by github-actions[bot]