wills106/homeassistant-solax-modbus #843: [Bug]: SOFAR - HYD-15KTL

julienbourgainpro, 2024-04-03:
### Describe the bug

Everytime i try to change the inverter mode, i have the error "No response received after 3 retries"
But sometimes the inverter mode change and sometimes not. 
It's very instable, and home assistant doen't like errors in automatisation.

I don't know if the stacktrace can help you, let me know how i can provide more information to fix that

### Integration Version

2024.4.1

### Homeassistant core version

2024.4.0

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP1ES1

### Connection Method

LSE-3

### Detailed Error Log

2024-04-03 23:33:42.795 ERROR (MainThread) [homeassistant.components.script.passive_mode] Passive mode: Error executing script. Unexpected error for call_service at pos 1: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 504, in _async_step
await getattr(self, handler)()
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 733, in _async_call_service_step
response_data = await self._async_run_long_action(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 696, in _async_run_long_action
return await long_task
^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2542, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2579, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 971, in entity_service_call
single_response = await _handle_entity_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1043, in _handle_entity_call
result = await task
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
await self.async_select_option(option)
File "/config/custom_components/solax_modbus/select.py", line 106, in async_select_option
await self._hub.async_write_registers_single(unit=self._modbus_addr, address=self._register, payload=payload)
File "/config/custom_components/solax_modbus/__init__.py", line 543, in async_write_registers_single
resp = await self._client.write_registers(address, payload, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 180, in async_execute
raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2024-04-03 23:33:42.815 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [281473359042240] Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
response = await hass.services.async_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2542, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2579, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/script/__init__.py", line 672, in _service_handler
response = await self._async_start_run(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/script/__init__.py", line 630, in _async_start_run
script_result = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/script/__init__.py", line 661, in _async_run
return await self.script.async_run(script_vars, context)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1650, in async_run
return await asyncio.shield(create_eager_task(run.async_run()))
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 454, in async_run
await self._async_step(log_exceptions=False)
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 506, in _async_step
self._handle_exception(
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 559, in _handle_exception
raise exception
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 504, in _async_step
await getattr(self, handler)()
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 733, in _async_call_service_step
response_data = await self._async_run_long_action(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 696, in _async_run_long_action
return await long_task
^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2542, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2579, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 971, in entity_service_call
single_response = await _handle_entity_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1043, in _handle_entity_call
result = await task
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
await self.async_select_option(option)
File "/config/custom_components/solax_modbus/select.py", line 106, in async_select_option
await self._hub.async_write_registers_single(unit=self._modbus_addr, address=self._register, payload=payload)
File "/config/custom_components/solax_modbus/__init__.py", line 543, in async_write_registers_single
resp = await self._client.write_registers(address, payload, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 180, in async_execute
raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2024-04-03 23:35:24.604 ERROR (MainThread) [pymodbus.logging] General exception: unpack requires a buffer of 4 bytes
2024-04-03 23:35:24.605 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1017, in _read_ready__data_received
self._protocol.data_received(data)
File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 303, in data_received
self.datagram_received(data, None)
File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 337, in datagram_received
cut = self.callback_data(self.recv_buffer, addr=addr)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 201, in callback_data
self.framer.processIncomingPacket(data, self._handle_response, slave=0)
File "/usr/local/lib/python3.12/site-packages/pymodbus/framer/base.py", line 139, in processIncomingPacket
self.frameProcessIncomingPacket(single, callback, slave, **kwargs)
File "/usr/local/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 106, in frameProcessIncomingPacket
raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request

### Additional context

inverter software version 0V1100510V1100510V110051
v110051 is the latest version for this inverter

julienbourgainpro, 2024-04-11:
On my side, i have made a script in node-red to observe a "wanted storage mode select" and change inverter mode with catch error and retries in a while loops. but it would be simplier without this error

[image]

In config.yaml
```
sofar_wanted_storage_mode:
  name: Mode de stockage voulu
  options:
    - Self Use
    - Time of Use
    - Passive Mode
  initial: Time of Use
  icon: mdi:panda

```
In node-red
`[{"id":"6652012ab3fc4571","type":"api-call-service","z":"616154d2fa885784","name":"Set Passive Mode","server":"d467981a.05fb88","version":5,"debugenabled":false,"domain":"script","service":"passive_mode","areaId":[],"deviceId":[],"entityId":[],"data":"","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":670,"y":700,"wires":[["94ce86d81423a43a"]]},{"id":"0397d651202298c5","type":"api-current-state","z":"616154d2fa885784","name":"If is not Passive Mode","server":"d467981a.05fb88","version":3,"outputs":2,"halt_if":"Passive Mode","halt_if_type":"str","halt_if_compare":"is_not","entity_id":"select.sofar_energy_storage_mode","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"payload","propertyType":"msg","value":"","valueType":"entityState"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"}],"for":"0","forType":"num","forUnits":"minutes","override_topic":false,"state_location":"payload","override_payload":"msg","entity_location":"data","override_data":"msg","x":420,"y":700,"wires":[["6652012ab3fc4571"],["4a9ddd77c5b674e3"]]},{"id":"f524f04b1989e25d","type":"catch","z":"616154d2fa885784","name":"","scope":["6652012ab3fc4571"],"uncaught":false,"x":640,"y":740,"wires":[["94ce86d81423a43a"]]},{"id":"f8b6c0bf1b5b36d8","type":"api-call-service","z":"616154d2fa885784","name":"Set Self Use","server":"d467981a.05fb88","version":5,"debugenabled":false,"domain":"script","service":"self_use","areaId":[],"deviceId":[],"entityId":[],"data":"","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":650,"y":900,"wires":[["7dcbab87ec175374"]]},{"id":"dfb88a4fd921c30f","type":"api-current-state","z":"616154d2fa885784","name":"If is not Self Use","server":"d467981a.05fb88","version":3,"outputs":2,"halt_if":"Self Use","halt_if_type":"str","halt_if_compare":"is_not","entity_id":"select.sofar_energy_storage_mode","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"payload","propertyType":"msg","value":"","valueType":"entityState"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"}],"for":"0","forType":"num","forUnits":"minutes","override_topic":false,"state_location":"payload","override_payload":"msg","entity_location":"data","override_data":"msg","x":400,"y":900,"wires":[["f8b6c0bf1b5b36d8"],["05899f6953be5e1a"]]},{"id":"e95a0033b649ecb0","type":"catch","z":"616154d2fa885784","name":"","scope":["f8b6c0bf1b5b36d8"],"uncaught":false,"x":640,"y":960,"wires":[["7dcbab87ec175374"]]},{"id":"84ca676d87499c9d","type":"api-call-service","z":"616154d2fa885784","name":"Set Time of Use","server":"d467981a.05fb88","version":5,"debugenabled":false,"domain":"script","service":"time_of_use","areaId":[],"deviceId":[],"entityId":[],"data":"","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":680,"y":1100,"wires":[["e3bc4ceadb885d64"]]},{"id":"9fdeb28c94d70c28","type":"api-current-state","z":"616154d2fa885784","name":"If is not Time of Use","server":"d467981a.05fb88","version":3,"outputs":2,"halt_if":"Time of Use","halt_if_type":"str","halt_if_compare":"is_not","entity_id":"select.sofar_energy_storage_mode","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"payload","propertyType":"msg","value":"","valueType":"entityState"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"}],"for":"0","forType":"num","forUnits":"minutes","override_topic":false,"state_location":"payload","override_payload":"msg","entity_location":"data","override_data":"msg","x":420,"y":1100,"wires":[["84ca676d87499c9d"],["6768f46e788a2bce"]]},{"id":"2bcedec4c5a8b523","type":"catch","z":"616154d2fa885784","name":"","scope":["84ca676d87499c9d"],"uncaught":false,"x":660,"y":1140,"wires":[["e3bc4ceadb885d64"]]},{"id":"c4a308f732938417","type":"trigger-state","z":"616154d2fa885784","name":"Storage Mode change","server":"d467981a.05fb88","version":4,"inputs":0,"outputs":2,"exposeAsEntityConfig":"","entityId":"input_select.sofar_wanted_storage_mode","entityIdType":"exact","debugEnabled":true,"constraints":[],"customOutputs":[],"outputInitially":false,"stateType":"str","enableInput":false,"x":120,"y":660,"wires":[["c24432efb914523f"],[]]},{"id":"36d848b8c5376faf","type":"api-current-state","z":"616154d2fa885784","name":"If wanted is Passive mode","server":"d467981a.05fb88","version":3,"outputs":2,"halt_if":"Passive Mode","halt_if_type":"str","halt_if_compare":"is","entity_id":"input_select.sofar_wanted_storage_mode","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"payload","propertyType":"msg","value":"","valueType":"entityState"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"}],"for":"0","forType":"num","forUnits":"minutes","override_topic":false,"state_location":"payload","override_payload":"msg","entity_location":"data","override_data":"msg","x":140,"y":780,"wires":[["0397d651202298c5"],["54c35dd11b80aec8"]]},{"id":"54c3 [... cut ...] :"min","reset":"","bytopic":"all","topic":"topic","outputs":1,"x":880,"y":900,"wires":[["dfb88a4fd921c30f"]]},{"id":"e3bc4ceadb885d64","type":"trigger","z":"616154d2fa885784","name":"Wait 1 minutes","op1":"","op2":"30","op1type":"nul","op2type":"str","duration":"1","extend":false,"overrideDelay":false,"units":"min","reset":"","bytopic":"all","topic":"topic","outputs":1,"x":880,"y":1100,"wires":[["9fdeb28c94d70c28"]]},{"id":"d467981a.05fb88","type":"server","name":"Home Assistant","addon":true}]`

wills106, 2024-04-17:
This is possibly the same error as #801 

I have just released [2024.04.4b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.04.4b1) pushing pyModbus to v3.6.8

julienbourgainpro, 2024-04-26:
I think you're right, you can close this issue.

When #801 will be fixed, if i'm still in trouble, i will open another issue

closed 2024-08-07 by wills106

cschlipf, 2024-08-07:
Link to FAQ: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-faq/#using-the-lse-3-logger-stick-write-requests-fail-with-errors

PR #986 (2024-08-07): Update sofar-faq.md: Invalid return codes for write attempts using the LSE-3 logger.
Added description of invalid return codes on write attempts, when using the LSE-3 logger and how to workaround this issue.

Close #843