wills106/homeassistant-solax-modbus #848: [Bug]: 'Inverter Brand' - 'Inverter Model' Description of Error

chirs-51, 2024-04-06:
### Describe the bug

after some days sometimes only one its strops working a reload of the integration its fine but its had happen for some time now. 

### Integration Version

2024.04.1

### Homeassistant core version

2024.3.3

### Inverter brand

growatt

### Plugin used

plugin_growatt.py

### Serial prefix

RA1.0 ZCBA

### Connection Method

rs484  usb

### Detailed Error Log

its a growatt 3600 first half a year its wars working fine. after some updates ago thes startet to happen. form thes to the last one i total uninstalled the integration and restartet home assistent before reinstalled it Again, 

Logger: custom_components.solax_modbus
Kilde: custom_components/solax_modbus/__init__.py:614
integration: SolaX Inverter Modbus (dokumentation, problemer)
Første forekomst: 3. april 2024 kl. 20.26.31 (1 forekomster)
Senest logget: 3. april 2024 kl. 20.26.31

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 614, in async_read_modbus_data
    res = await self.async_read_modbus_registers_all()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 790, in async_read_modbus_registers_all
    self.data[descr.key] = descr.value_function(0, descr, self.data)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 111, in value_function_today_solar_energy
    return  datadict.get('today_pv1_solar_energy', 0) + datadict.get('today_pv2_solar_energy',0) + datadict.get('today_pv3_solar_energy',0) + datadict.get('today_pv4_solar_energy',0)
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for +: 'NoneType' and 'NoneType'

a nother one 

Logger: homeassistant.components.websocket_api.http.connection
Kilde: components/websocket_api/commands.py:239
integration: Home Assistant WebSocket API (dokumentation, problemer)
Første forekomst: 30. marts 2024 kl. 00.42.39 (7 forekomster)
Senest logget: 00.42.49

[140688892443200] Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2319, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2356, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 905, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solax_modbus/select.py", line 109, in async_select_option
    await self._hub.async_write_register(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 422, in async_write_register
    if awake: return await self.async_lowlevel_write_register(unit, address, payload)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 416, in async_lowlevel_write_register
    return await self._client.write_register(address, payload[0], **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 180, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

wills106, 2024-09-20:
Are you still getting this on newer versions of the Integration or can it be closed down?

chirs-51, 2024-09-21:
> versions

The issue seems to be gone so yes just close it 😊

closed 2024-09-21 by wills106