wills106/homeassistant-solax-modbus #849: [Bug]: Solax working fine, then suddenly unrecognized inverter type - serial number : unknown

LnK-Holly, 2024-04-07:
### Describe the bug

The plugin was working beautifully, then last week suddenly wouldn't connect with error: unrecognized inverter type - serial number : unknown.

The IP address hasn't changed, the Waveshare adaptor LED changes from green to blue to say connected, but plug in just presents above error and no entities available.

I don't think it happened at any particular update, but can't be sure. I've tried rolling back to previous versions of both HA core and this plug in but neither seem to work either. 

No idea what's gone wrong! Any help / guidance very much appreciated.

### Integration Version

2024.04.2

### Homeassistant core version

2024.4.1

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

MPT10T

### Inverter firmware versions

Unknown

### Connection Method

Ethernet Waveshare Adaptor

### Dongle firmware

N/A

### Detailed Error Log

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:7215
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 07:22:32 (2 occurrences)
Last logged: 07:54:20

unrecognized inverter type - serial number : unknown

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:7138
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 07:22:32 (2 occurrences)
Last logged: 07:54:20

SolaX: cannot find serial number, even not for MIC

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:80
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 07:22:12 (4 occurrences)
Last logged: 07:54:20

SolaX: reading serial number from address 0x0 failed; other address may succeed
SolaX: reading serial number from address 0x300 failed; other address may succeed

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:68
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 07:22:12 (4 occurrences)
Last logged: 07:54:20

SolaX: attempt to read serialnumber failed at 0x0 data: None
SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 68, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 482, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 180, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

tsunami-ad, 2024-04-08:
I have exactly the same problem.

LnK-Holly, 2024-04-08:
Any idea what caused yours? I'm still none the wiser, completely stumped!

tsunami-ad, 2024-04-08:
I dont know. Maybe updated another integration. But i restored from backup and its still not working so i'm not sure.

klubalo, 2024-04-09:
I have seen this issue right after update to 2024.03.7 and triggering remotecontrol (it was instant failure). I had to downgrade to 2024.03.6 as that's last version that works correctly with remotecontrol for me. Maybe some issue with locking? Not sure. I don't have much time to mess with it.

LnK-Holly, 2024-04-11:
Thanks for info @klubalo , unfortunately downgrading to previous versions doesn't seem to fix it for me. Can't for the life of me work out what's gone wrong?!

jimbaxter, 2024-04-11:
I have been having issues like this for the last month after 6 months of stable use with the PocketWifi 3.0

Using independent modbus tools I can still see there is a modbus connection but I need to work out how to use modbus commands and work out if it is working well enough to provide the required data. so I am unsure if the modbus connection has changed?

vitjurasek, 2024-04-14:
The same problem after todays upgrade :-(

Trevbil, 2024-04-16:
Same here. Stopped all of a sudden after upgrade. Downgrading does not help. Mine has progressed to showing “no entities or devices”

chwari, 2024-04-17:
Hello,
it stopped yesterday. 
ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] QCELLS: cannot find serial number, even not for MIC
ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

It´s sad cause it´s my favorite integration and the reason I installed HA in the first place. Did someone try to reinstall?

Hopefully there will be a fix...

ulldins, 2024-04-17:
I managed to resolve my issues. I found the IP address of the modbus was conflicting another device on my network.

I had forgotten I have a wifi range extender which some devices connect to, upon checking that, I found another device with the same IP address.

Having now changed the IP, the problem has resolved.

LnK-Holly, 2024-04-17:
Unfortunately IP conflict isn't the issue for me, nothing else has same IP.
I can also log in to the Waveshare device settings webpage at the correct
IP address.

I did try completely removing and reinstalling the plugin, but had the same
error.

Not sure if this helps unpick it, but on my Waveshare device the connection
LED changes from green to blue when I try to reload the plugin, and the Act
LED flashes green a few times, then it stops. The connection LED remains
blue.

@wills106, any ideas??

Trevbil, 2024-04-17:
Thank you @ulldins! 

Although I did not have a duplicated IP like you did. It gave me an idea.

- I went back to the pocket wifi settings and moved from my reserved IP and created a new static IP. Went to my router settings and reserved that IP as well. 
- Deleted the integration from HACS and rebooted (note I have tried this a lot of times before, I had even restored from a backup without success)
- Reinstalled SolaX Modbus Integration and restarted Home Assistant 
- Added the integration and voila - up and running and all previous dashboards came alive again.

I reckon others' setups may be different. Perhaps it can help them think of a solution.

ulldins, 2024-04-17:
> Thank you @ulldins!
>
> Although I did not have a duplicated IP like you did. It gave me an idea.
>
> * I went back to the pocket wifi settings and moved from my reserved IP and created a new static IP. [...]

Great news although monitor it for up to a few hours, as mine would connect and communicate fine and then stop working again.

LnK-Holly, 2024-04-18:
Yeh, I've just tried all above as well, and hasn't worked for me. I'm still getting same error as original post.

chwari, 2024-04-18:
Same here,
when setting up a new hub and connecting to the new IP I get the message "success" but ther are no devices or entitys shown.

chwari, 2024-04-19:
Hallo,
it´s working again!

This ist what I did:
In the state of deep frustration I ordered a Waveshare Industrial Serial Server RS485 to RJ45 Ethernet TCP/IP to Serial Module, a power adapter and 10m Cat 6 cable with RJ45 connector. With this I am going to use the COM Port on the Inverter.

But today I saw the the inverter IP had changed **again** (I forgot to set it to static when I left it yesterday) and the device appears under its old name "espressif". This had changed when I gave it a different IP yesterday. 

So I deleted the old hub in SolaX and added a new with the new IP - **again**. And this time it worked. The energy dashboard of HA ist not yet working but my own energydashboard works. All entitys and the numbers from the history log are back.

So if you have a AVM router (I have a fritz!box cable) you can try the following:
Got to Homenetwork (Heimnetz)
Network (Netzwerk)
Look up the inverter (in my case "espressif") and click "edit"
Uncheck "static IP" (Diesem Netzwerkgerät immer die gleiche IPv4-Adresse zuweisen) if set.
Scroll down to "reset settings" (Einstellungen zurücksetzen) an click reset
Confirm

Then just wait for the IP adress to change automaticly - maybe overnight. And try again.

Good luck!

Note:
The device was connected via my WiFi mesh-repeater (slave). Now it´s connected directly to the router. Why? No idea and I don´t think that it is possible to change this manually. 

Tomorrow I will try to set up the com port with the RS485 to RJ45. Just to see if I can ;-)
Hopefully that connection will be more stable.

And thank you wills106 for this tool.

djenson2000, 2024-04-21:
Unfortunate to see this hasn't been ironed out yet. I'm still on version 2024.02.6 (working) and not updating anything else unless there is a security issue. 

I believe i first raised the issue [here](https://github.com/wills106/homeassistant-solax-modbus/discussions/796#discussion-6346021) but wasn't sure if it was specific to my environment.

Trevbil, 2024-04-21:
From the comments of those that could sort this out (including me), it seems to be related to an IP/router issue. 

If anyone else has sorted the issue, it’ll be great to post how they did it for others.

LnK-Holly, 2024-04-23:
I'm still unable to get this working, which is such a shame. I've restored previous (admittedly partial) backups, uninstalled and reinstalled older versions of this plugin, changed IP address of Modbus device, restarted my router, and still none of it works. Just getting the same error as per my original post.

Anyone have any other ideas?? @wills106 , do you have any thoughts?

LnK-Holly, 2024-04-23:
OK, really embarrassingly I've fixed mine.. 1 of the 2 cables between Waveshare device and inverter had snapped. It was still in place, but not making contact.

Rewiring this has resolved all issues! 

Sorry everyone, and thanks for people's suggestions/ help. Hope you guys who haven't fixed yours can find a solution soon!

wranglatang, 2024-04-23:
Im also having this problem, just got home and its stopped! been working find all day!

arnohoff, 2024-04-23:
I have the same problem and have not figured out yet what to do... How can I restart the WiFi dongle?

Thanks

wranglatang, 2024-04-23:
> I have the same problem and have not figured out yet what to do... How can I restart the WiFi dongle?
> 
> Thanks

have you tried just unplugging the dongle and plugging it back in?

arnohoff, 2024-04-23:
Hi there, 

I solved it now after apprm. 5h of trying. 

What did I do? 
- connect via ip to the Web interface to the converter. 
- search WiFi, select WiFi
- enter password and save. 
- wait till fritxbox gives me up
- to to frizbox and freeze the new ip to static
- to in home assistant and go to modbus, delete old device
- add new device with me IP 
- all values are back and running!!!

Awesome

wranglatang, 2024-04-24:
I just restarted home assistant after making some changes to things and noticed the issue has occurred again. My Previous fix or reconnecting the device in my UDR has not solved the issue.

Birkenw25, 2024-05-31:
I have the samme or similar issue. HA forgot the solax. It is connected with wifi dongle. I have tried a lot! But up to know I am unsure about the reason. Restart of Integration, HA and the VM does not help. But the last two times I always disconnected the wifi dongle, waited a few seconds and reconnect it and it worked again. All entities are back again as nothing was happend.
But out of a sudden, it happend again. But again I get it done just by disconnect/reconnect the dongle.
does anybody know, how to automate this bugfix ;-) ?

wills106, 2024-09-07:
Has this been resolved in later Integration versions?
Looking to see if I can close this down.

tsunami-ad, 2024-09-07:
Hey, no. Still cant recognize the entity. I put the right ip and no entity

wills106, 2024-09-08:
> pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

@tsunami-ad are you getting the above error or just unrecognised Inverter? Have you tried a newer version of the Integration?

tsunami-ad, 2024-09-08:
I get few errors:
1. SolaX: attempt to read serialnumber failed at 0x0 data: None
2. unrecognized inverter type - serial number : unknown
3. SolaX: cannot find any serial number(s)

And the ip of the inverter is right

Birkenw25, 2024-09-09:
> Has this been resolved in later Integration versions? Looking to see if I can close this down.

Yes, I haven’t had this error since weeks.
I am running 2024.05.07 and it works fine. 
Many thanks!

LnK-Holly, 2024-09-09:
> Has this been resolved in later Integration versions? Looking to see if I can close this down.

Mine is working beautifully, thanks Will!

closed 2024-09-15 by wills106