wills106/homeassistant-solax-modbus #851: [FR]: 'No grid export'

wonko, 2024-04-07:
### Description

We have, when the energy price is negative, a penalty when you export electricity to the grid. As long as the battery isn't fully charged, excess PV energy is sent to the battery.

Is there a mode in "Modbus Power Control" where excess PV power isn't sent to the grid, but "discarded"? The only way I currently can do this is to turn of the inverter completely, but the side-effect is that we then use grid-energy for the house load, while there actually is PV power.

### Suggested Solution

No idea - either some documentation or a good example on this...

### Alternatives

Turning the inverter off. If the energy price is significantly negative, this has no downside. Due to the way grid export is calculated when the price is close to 0, this comes at a cost.

### Additional Context

Solax inverter, Gen4, already using modbus power control for charging/discharging etc.

wonko, 2024-04-08:
Thanks @romankysely, I didn't know this setting. I assume this writes to the EEPROM, thus should we avoid setting this too much? Is there a equal setting or mode using the remote power control?

romankysely, 2024-04-08:
Yes I suppouse there is EEPROM write unfortunately but I did not find any relevant option via Remotecontrol. May be change YAML code to write to registere after minus/plus state not each hour.

romankysely, 2024-04-08:
We have aboute 1.000.000 hits to EEPROM I suppose :-))
I have another problem. I have setup Setting/Advanced/Export Control/User Value = 15000W. I can not to rise to 22000W. Maybe there   is limit on inverter. I have 2 invertors 12kW each. Master has = 15000 and Slave = 0W set. I suppose there is one value for tandem for Export Limit. Integration not allow me change value more than 15000. Any idea?

wonko, 2024-04-08:
Seems like a bit off-topic for this feature request, but I guess you could take a look at https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_solax.py#L329 - seems like the limits are defined over there. I only have a single 6kW Gen4 setup, so no help from my side.

romankysely, 2024-04-11:
Thx  for interesting source of info. 
Back to topic "prohibit export": 
First I have to correct myself - not 1.000.000 writes to EEPROM alow but 100.000 only (source https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-G4-operation-modes/). 
I have read documentation regarding Remotecontrol (as you mentioned - thx). I am little bit confuse with this part: "remotecontrol_export_limit: (ATTENTION: not working as intended - removed)". Does not work at all or is it replaced by some mode?
I supouse all of us who has a risk of penalty for export energy out must automaticly manage it. So how do you all guys resolve it? Everybody do not care to write to EEPROM? Or any other idea how to manage it?
PS: BTW I resolved my question regarding Maximum Export Limit there #858

wonko, 2024-04-15:
Closing this, as the "solax_export_control_user_limit" is the parameter needed to limit the export towards the grid.

closed 2024-04-15 by wonko