wills106/homeassistant-solax-modbus #857: [Bug]: Export control slider applying 10x less than real value on the inverter

pedromfa, 2024-04-10:
### Describe the bug

When i select for example 100 w export on the integration, on the inverter config it changes to 10 w. This causes to then when i select the max 7400w on the integration then in reality im just selecting 740w on the inversor. Can this be fixed? Appreciated.

[image]
[image]

### Integration Version

2024.04.2

### Homeassistant core version

2024.4.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H4372A

### Inverter firmware versions

ARM 2.03 and DSP 2.07

### Connection Method

SolaX Pocket WIFI 3.0

### Dongle firmware

NOT SURE WHERE TO LOOK FOR THIS

hypnum, 2024-04-10:
It also happens to me with a poker v3.0. 
 Core 2024.4.1
Supervisor 2024.04.0
Operating System 12.1

jkoekemoer, 2024-04-11:
Same issue for me on the last 4 versions of the modbus plugin as well as HASS versions (has always done this since first integration)
Core
2024.4.1
Supervisor
2024.04.0
Operating System
12.1

hypnum, 2024-04-12:
It seems that I have fixed the problem by modifying the configuration indicated in #418 . Changing the values in
EXPORT_LIMIT_SCALE_EXCEPTIONS = [
     ('H4', 100 ), # assuming all Gen4s
     ('H34', 100), # assuming all Gen4s
     ('H4502T', 1), # more specific entry comes last and wins
     ('H3UE10', 100), # bug report #339
# ('H1E', 100 ), # assuming all Gen4s

pedromfa, 2024-04-12:
> Same issue for me on the last 4 versions of the modbus plugin as well as HASS versions (has always done this since first integration) Core 2024.4.1 Supervisor 2024.04.0 Operating System 12.1
> 
> Edit Based on issue #418 enabling config_export_control_limit_readscale does mostly fix the issue but value still does not reflect one to one

Exactly. Same here. It reflects 1 to 1.1. Its strange.
[image]

blueshadownet, 2024-04-25:
I have the same issue. You can go into the SolaX app and set it. For example 4kw will show as 40,000W but once you change it on the SolaX Modbus integration, it will not allow you to set it that high.

pedromfa, 2024-04-26:
> In MAX_EXPORT you look for your investor and modify your export limit and then you can modify correctly. ![Captura](https://private-user-images.githubusercontent.com/71152234/325914815-45944603-46ed-452d-b7b7-eb8836bf1895.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxMjQzMDIsIm5iZiI6MTcxNDEyNDAwMiwicGF0aCI6Ii83MTE1MjIzNC8zMjU5MTQ4MTUtNDU5NDQ2MDMtNDZlZC00NTJkLWI3YjctZWI4ODM2YmYxODk1LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI2VDA5MzMyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZkMzhhYjNiMDc5YmVhNmNiOGVjOTgxODc5MjdiNTNhNTQxZjhjZWJjM2Y1ZmQ0MmU0NGQ4OGE4ZTQwODgxOGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tHpfdoouKjvqNKonFEZGsrmFy_hBk-ggKQ-1Mf-T6Hk)

I have no idea how to do that. :) I was hoping for a new version release to fix this.

hypnum, 2024-04-26:
You have to edit the file plugin_solax.py which is in
\config\custom_components\solax_modbus. Find the max_export.

billmars206, 2024-05-05:
I have this problem.
The Solax Export control slider shows 10x less than the real value on the inverter.
Inverter set to 3680 W, Solax export control shows 368 W.
Set Home Assistant Solax Export control to 3680 W, Inverter is set to 36800 W.

I solved this by adding:-

    ('H4502A', 1),

to  the file plugin_solax.py which is in
\config\custom_components\solax_modbus.

so it looks like this:-

EXPORT_LIMIT_SCALE_EXCEPTIONS = [
    ('H4', 10),     # assuming all Gen4s
    ('H4502A', 1), # added for my inverter serial number H4502AI60*****
    ('H34', 10),     # assuming all Gen4s
    ('H4502T', 1),   # more specific entry comes last and wins
    ('H3UE10', 10),  # bug report #339
#    ('H1E', 10 ), # assuming all Gen4s
]

The only problem is that when the integration is updated I have to add it again.
Can this setting be included in any integration update?

Thanks.

wills106, 2024-05-05:
Feel free to submit a PR

wills106, 2024-05-11:
Should be corrected in [2024.05.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.3) for 'H4372A' & 'H4502A'

closed 2024-05-12 by pedromfa