wills106/homeassistant-solax-modbus #858: [Bug]:  number.solax_export_control_user_limit maximum  in  multiple inverters master slave setup

a1bert01, 2024-04-11:
### Describe the bug

solax_export_control_user_limit maximum is 16500 (hardcoded) but as I have 3 invertors connected (in master - slave setup)  real maximum should be 3x16500 = 49500

it can be set by original application (and is read corectly by integration) but it cannot be set by integration itself

### Integration Version

2023.09.3

### Homeassistant core version

 2023.7.3

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

33

### Connection Method

-

### Dongle firmware

-

romankysely, 2024-04-11:
I resolved the same trouble that way: 
Look there first: https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-G4-operation-modes/
Then Enable SolaX Config Max Export at Integration (at my case is it max. 50kW). Be aware that enabled entity is showen at Diagnostic section! Setup power as you need it a than change slider SolaX Export Control User Limit to requested value (or automations...).  Initial maximum ( in my case 2x X3-Gen4 12kW is set 15.000) because this https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_solax.py#L329

I am very interesting of your experince with PM (Parallel Mode) because there is not so much instalations. I am going to open topic with sensor sensor.solax_pm_i2_battery_current_charge which has often shoots value to 64.000Amper!! And other issues with PM mode...

a1bert01, 2024-04-11:
oops I have not noticed  "SolaX Config Max Export", thanks!

closed 2024-04-11 by a1bert01