wills106/homeassistant-solax-modbus #860: [Bug]: Parallel Mode setup sensor.solax_pm_i2_battery_current_charge value is often at 6400A

romankysely, 2024-04-11:
### Describe the bug

Very often is value of the current on Slave Invertor shoots to value 6400Amps. 
[image]

The same situation is with power entity: sensor.solax_pm_i2_battery_power_charge = 65kW (there is probably linked calculation from corrupted current entity) 
[image]

Is the problem with integration or wrong values coming through registry from invertor? @wills106 what do you think about? thx for investigation. I know You dont have possibility to test PM but I can do. I am prepare for help and cooperation with investigation on my HW. There are more problems with PM so we can try to resolve them gradually. Thx a lot

### Integration Version

 2024.03.7b4

### Homeassistant core version

2024.3.3

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A12

### Inverter firmware versions

ARM 1.34 DSP 1.35

### Connection Method

Waveshare adapter

### Dongle firmware

LAN dongle (I do know where to find FW)

### Detailed Error Log

There si no relevant error in log.

### Additional context

Thx all participants for help. Cofee is warming up :-)

a1bert01, 2024-04-11:
just checked on our 3xH34A15

[image]

wills106, 2024-04-12:
I know the SolaX doc doesn't mention anything about termination resistors. But have you tried adding them?

https://github.com/wills106/homeassistant-solax-modbus/issues/743#issuecomment-1951408142

Have you tried the Bus conflict option mentioned https://github.com/wills106/homeassistant-solax-modbus/issues/804#issuecomment-2004783893 also the Multi-Host setting could help to remove the spikes.

romankysely, 2024-04-13:
Thx wills106 for interesting hints. I do not have resistors either Mulit-host settings. I am going to test it gradualy.
I am going to test it gradualy. 
1. step Mulit-host settings. I changed Waveshare settings and I am waiting for results.
[image]

To be  sure: I am physicaly connected ( 2 wires without GND) with Waveshare ONLY to Master and ALL data is coming from Master(?). So Master Invertor is containing (collect) data also for Slave via Solax "internal Modbus CAN2-CAN1" or Waveshare is communicating also with Slave directly via Modbus and data integration is inside HA Solax Modbus Integration? In the other words who is integrator?  Master internaly or Solax Invertor Modbus Integration?
@a1bert01 are you going to test it too?

romankysely, 2024-04-15:
Continuation and results:

1. step: completed - no success
2. step: used 120 Ohm resistor and terminate Modbus on the Waveshare only - no success
3. step: used 120 Ohm resistor and terminate Modbus on the Waveshare and Invertor Slave port COM - no success

When You look to graph it doesnt look like random spike problem on the electric level. During night is long state of the peek uninterupted. It seems like coincident with PV activity or charging battery generally. Any idea pls?
[image]

ladaan, 2024-07-12:
Hi Roman,
Have you resolved the issue? I am in charge of 2 Solax Gen4 15kW in parallel mode and using this Solax Modbus integration where in both are Solax Pocket Wifi 3.0. I am not installer of the Solax parallel solution but just a guy who wants to configure several automations (e.g. Export limit to 0 W when the price of the electricity on the market is low etc.). Not sure why there are two dongles since they are connected via Modbus (CAN ports). Possibly this is due to monitoring on Solax Cloud..
However, I am not happy with a lot of values shown in comparison with single installation.
Cheers,
Lada

ladaan, 2024-07-15:
Couldn't it be related to Solax firmware (DSP and ARM) versions than to this plugin itself? plugin_solax.py just reads data from registers. I have spikes e.g. in PM PV Power (reach value 2^32) on versions (Firmware: DSP v1.33 ARM v1.31). The latest are DSP v1.42 and ARM v1.41.

ladaan, 2024-07-19:
[image]
This is what I see.

romankysely, 2024-07-24:
> Hi Roman, Have you resolved the issue? I am in charge of 2 Solax Gen4 15kW inverters in parallel mode and using this Solax Modbus integration where in both are Solax Pocket Wifi 3.0. I am not installer of the Solax parallel solution but just a guy who wants to configure several automations (e.g. Export limit to 0 W when the price of the electricity on the market is low etc.). Not sure why there are two dongles since they are connected via Modbus (CAN ports). Possibly this is due to monitoring on Solax Cloud.. However, I am not happy with a lot of values shown in comparison with single installation. Cheers, Lada

Hi Lada, sorry for delay.
I have now ARM 1.42 and DSP 1.43 but nothing new, same issue. Regarding 2 Inverters setup: Yes there is mess. I am using LAN dongles on both inverters. I am preparing for exchange Master LAN dongle for Pocket WiFi because I need release COM port (Modbus) for parallel wallboxes. My plan is leave LAN dongle on SLave invertor. I suppose that Slave communicate via LAN dongle only wich Solax Cloud. Master and Slave are talking betveen each other via CANbus and Master is keeping most of information from Slave on site. But not all and I have to some of them create via custom sensors. 
BTW if you have 2 invertors, you probably can utilise this topic #858. Default is only 15kW Export limit. 
I am using simple automation for controling exports and watching negative price on the SPOT. But I have to resolve multiple write to EEPROM yet.....

ladaan, 2024-07-25:
> Hi Lada, sorry for delay. I have now ARM 1.42 and DSP 1.43 but nothing new, same issue. Regarding 2 Inverters setup: Yes there is mess. I am using LAN dongles on both inverters. I am preparing for [...]

I will give it a try today. Remove one of the dongle (master or slave) and will see what happens with updates to cloud or to HA.

Regarding the export limit. I am already aware of this issue for a few months. You can just manually modify it within plugin_solax.py (but of course this is not ideal as it necessary to do it after each update of solax modbus integration).

romankysely, 2024-07-25:
> Regarding the export limit. I am already aware of this issue for a few months.
Thgere is better solution:
https://github.com/wills106/homeassistant-solax-modbus/issues/858#issuecomment-2049353876

ladaan, 2024-07-26:
> > Regarding the export limit. I am already aware of this issue for a few months.
> > Thgere is better solution:
> > [#858 (comment)](https://github.com/wills106/homeassistant-solax-modbus/issues/858#issuecomment-2049353876)

Thank you for the tip. However, once I enabled it in solax modbus integration (disabled by default0 the modbus communication became unavailable. As I did it remotely I had to ask someone on site to pull out and put in the Solax Pocket wifi dongle. Not sure whether there is any remotely possible way to make it available again (maybe by uploading a new or current firmware to dongles resulting in reboot). Do you have this experience?

extazy173, 2024-08-14:
Hi, @romankysely any updates? Are you from czech?

romankysely, 2024-08-17:
Hi @extazy173, yes I am Czech. No update. We can switch directly in czech
at roman(at)kysely.name

romankysely, 2025-02-24:
I am responding to wills106's call to clean up old, outdated issues. 
The problem still persists. The new finding is that this only happens when the battery is discharging and negative values ​​appear. The same applies to the Battery Current Charge sensor. It seems that the problem is probably in the current sensor and is transferred to the Battery Power Charge through the power calculation. Then the peaks occur. I will try to solve this problem by reading data from the Slave inverter directly, not via the ModbusRTU Master.
[image]

[image]

wills106, 2025-02-24:
> Hi [@extazy173](https://github.com/extazy173), yes I am Czech. No update. We can switch directly in czech

Thanks for looking into old issues. By the way, feel free to communicate on here in your native language if it's easier for both of you or with other users while trying to troubleshoot an issue.

a1bert01, 2025-02-24:
we should definitely summarize the current state/best practice of master-slave setup in the documentation.  there is only a short and not very clear FAQ section right now

If I understand correctly, there is no way to get sum counters (import/export/solar energy) just from master

to get all available data from master & slaves, I need to register all converters as separate hubs in HA and connect them with their own  rs485 links (to avoid simultaneous read collisions ) and compute sums in HA itself?

a1bert01, 2025-02-24:
so  the problem is just modbus read serialization  (between multiple hub instances)? 

https://github.com/wills106/homeassistant-solax-modbus/discussions/868

romankysely, 2025-02-24:
yes because Modbus master suppot only one TCP relation with clinet in the same time as I know.. #868 is interesting

closed 2025-10-03 by github-actions[bot]