wills106/homeassistant-solax-modbus #863: [Bug]: Parallel system power levels incorrect on individual inverters

Cybgak, 2024-04-17:
### Describe the bug

In a parallel system when you send the remotecontrol values to the master inverter it also applies them to the slave inverter.
The problem is the power limit values allowed are based on the values for a single inverter (15000W) however when the master inverter receives the value it divides it over the available inverters so for instance with two inverters each one is limited to 7500W.

### Integration Version

latest

### Homeassistant core version

2024.4.0

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H34A12

### Inverter firmware versions

ARM 1.39 DSP 1.4

### Connection Method

Pocket WiFi 3.0

### Dongle firmware

3.001.02

### Detailed Error Log

No error

romankysely, 2024-04-17:
Do you use solax_export_control_user_limit or remotecontrol_export_limit ?

Cybgak, 2024-04-17:
> Do you use solax_export_control_user_limit or remotecontrol_export_limit ?

solax_export_control_user_limit is set to 12000W on both inverters
I think remotecontrol_export_limit has been disabled because it wasn't working properly.

romankysely, 2024-04-18:
maybe help #858

wills106, 2024-09-20:
Did #858 help? Can this be closed down?

Cybgak, 2024-09-23:
I'm not sure the setting suggested in #858 is required any more as I seem to be able to set the remote control limits to 30,000 (2x15000) anyway now which is what I needed.

closed 2024-09-23 by Cybgak