wills106/homeassistant-solax-modbus #872: [Bug]:  SolaX modbus integration no longer gets data

JAFC-BandO, 2024-04-24:
### Describe the bug

After updating to the new software (2024.04.3)  I'm no longer able to get any data from my SolaX modbus integration. all sensors are just unavailable. Also I tried to roll back to 
2024.04.2 previous release. It didn't fix anything 
[image]

### Integration Version

2024.04.3

### Homeassistant core version

2024.4.4

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

NA

### Inverter firmware versions

NA

### Connection Method

WIFI 3.0 Adapter

### Dongle firmware

New version

thomasrumpf, 2024-04-25:
After update to HA 2024.4.4 I have the same Problem.

simowNgithub, 2024-04-25:
I have this issue always after fully restarting HA. Could you check what happens if you completly delete the device and add it again? I think then everything will work.

tweltner, 2024-04-25:
> After update to HA 2024.4.4 I have the same Problem.

I have same problem after updating to HA 2024.4.4

**Home Assistant Installation:**
Core: 2024.4.4
Supervisor: 2024.04.0
Operating System: 12.2
Frontend: 20240404.2

**SolaX Integration:**
2024.04.3

**Log:**
Logger: pymodbus.logging
Quelle: custom_components/solax_modbus/__init__.py:482
Integration: SolaX Inverter Modbus ([Dokumentation](https://homeassistant-solax-modbus.readthedocs.io/), [Probleme](https://github.com/wills106/homsassistant-solax-modbus/issues))
Erstmals aufgetreten: 20:36:36 (2 Vorkommnisse)
Zuletzt protokolliert: 20:36:41

Cancel send, because not connected!

vitjurasek, 2024-04-26:
the same problem :-(

vitjurasek, 2024-04-26:
Tato chyba pochází z vlastní integrace.

Logger: custom_components.solax_modbus.plugin_solax
Zdroj: custom_components/solax_modbus/plugin_solax.py:7215
integrace: SolaX Inverter Modbus (dokumentace, problémy)
První výskyt: 07:47:27 (2 výskyty)
Naposledy logováno: 08:02:33

unrecognized inverter type - serial number : unknown

ReneMronet, 2024-04-27:
same problem.

Marc-dApice, 2024-04-27:
Same problem

ZaphodB72, 2024-04-27:
Same problem

tweltner, 2024-04-27:
I luckily solved the problem. After delete the configured solax integration including it's entities and re-add them directly did not help i tried yesterday this changes:

I reconfigure the existing solax integration and changed connection on second page from "Modbus TCP" to "Modbus RTU over TCP" (which is definitely not correct) - do not forget to save - and then changed this configuration back to "Modbus TCP" the connection works and all entities are back.

Maybe this helps you guys, too! :))

ZaphodB72, 2024-04-27:
I solved it by blocking the SolaX WiFi Adapter from the Internet in my Router and then enabling Internetconnection again. 
After that all entities came back🤷‍♂️

ReneMronet, 2024-04-28:
> I reconfigure the existing solax integration and changed connection on second page from "Modbus TCP" to "Modbus RTU over TCP" (which is definitely not correct) - do not forget to save - and then changed this configuration back to "Modbus TCP" the connection works and all entities are back.

Thanks @tweltner!
That helps for me :-)

Best regards

he-tim, 2024-04-28:
Same issue here after rebooting HA. 
SolaX integration was running fine for several weeks before.
I am using 2 SolaX systems with Wifi Dongles - both connections failed at the same time.
Reconfiguring the integration from "Modbus TXP" to "Modbus RTU (or ASCII) over TCP" and back did not change anything.
HA:
  - Core 2024.4.4
  - Supervisor 2024.04.0
  - Operating System 12.2
  - Frontend 20240404.2

Codiwr, 2024-05-01:
Same problem here -

Hard reset of the inverter seems to have worked
- unplugged dongle, plugged back in then turned off and on. Then reloaded (not reinstalled) integration.

- Other suggestions above failed as well as restoring HA from backup from a few days ago.
Hope this helps

JAFC-BandO, 2024-05-01:
**Workaround**:

I fixed my issue by rebooting my main router and retriggering the HA SolaX integration setup with the "new" IP that was assigned to the inverter.

Lake292, 2024-05-02:
Same here yesterday my HA was updated and today I have hard time to reestablishing the connection again... Solax Cloud works also other applications that use Modbus TCP are working fine.

ernst10, 2024-05-03:
Just a short update.
I did rollback to core 2024.4.0 now the Modbus is working again. I used Terminal and the following command:
ha core update --version 2024.4.0. I had to wait a few minutes as i use a Microsd card for HA.
on another test system having the same problem i did also rollback to 2024.4.0 and also worked. Then i did update to 2024.4.4 which as the failing core version before and surprise it worked again with 2024.4.4
don´t know what is there going on

Ernst10

Codiwr, 2024-05-03:
Broken again :(. Not found a solution yet

hansaugler, 2024-05-04:
Same problem with With Waveshare RS485 Server

ernst10, 2024-05-04:
was anyone able to verify my "solution" ? for me it´s still working. Yes i will not do any update to 2024 .5.x. What a surprise. 
Ernst10

johnnyhero83, 2024-05-04:
I kind of have the same problem. Sometimes I am able to load the entities and sometimes not. While doing nothing differently. It seems like it is randomly going through once and second time you just cant load them. Plus it seems like after you can finsky read the data, you cant write over modbus. The entity Solax.Lock.State seems to be Locked and not Unlocked. Then you cant write.

johnnyhero83, 2024-05-05:
Have any of you guys has it solved by updating to Ha core 5.1.?
Or updating to version 5.2. Of this custom integration?
Thanks

Codiwr, 2024-05-05:
I upgraded to core 5.1.0 - and it's working now, not sure if that was what fixed it directly though.

ernst10, 2024-05-05:
Hi 
Is there a way for rollback to previous version of the modbus custom integration. Just to know before i will test with latest version 5.2

thanks
Ernst10

Strinzer, 2024-05-05:
Fallback to 2024.4.0 and update to 2024.4.4 (like ernst10) worked for HA OS instance. It did not work for Supervised Version.

ernst10, 2024-05-05:
Strange. I´m running supervised version of HA

ernst10, 2024-05-05:
Update
using supervised HA
upgrade from Core 2024.4.4 to core 2024.5.1 
Modebus still working
now upgrade modbus solax 2024.4.3 to 2024.5.2
all is working again as expected

what ever has caused the break it is gone now for me

ernst10

simowNgithub, 2024-05-06:
> Update using supervised HA upgrade from Core 2024.4.4 to core 2024.5.1 Modebus still working now upgrade modbus solax 2024.4.3 to 2024.5.2 all is working again as expected
> 
> what ever has caused the break it is gone now for me
> 
> ernst10

core 2024.4.4
solax 2024.5.2

working also on my system.

Codiwr, 2024-05-07:
Problem has returned again. On HA 5.1. Upgraded to 5.2 but to no avail. Restarting/rebooting jas had no effect.

purewhitedesign, 2024-05-10:
I don't know, whats going on. After update to solax inverter modbus 2024.05.2 all entities are not available. So I restore a backup and it works. Today same issue. No entities available.
Solutions like change TCP to RTU and back or block / allow internet access doesn't work for me.
I had to delete the solax interverter modbus integration completely. Then I installed and configured it again.
Now it works.
Solax Version: 2024.05.2
Core: 2024.5.2
Supervisor: 2024.05.1
Operating System: 12.3
Frontend: 20240501.1

Codiwr, 2024-05-10:
Has happened 3 times over the past week. Only thing that works reliably is to Unbind in Solax app, turn off hard switch on Inverter and on again and set up WiFi again in app. 

Then restart Solax integration and restart HA. No need to delete/clear HA data.

Marc-dApice, 2024-05-11:
So I've completely built a new Home Assistant (supervised) server from scratch to try and eliminate any current configuration issues and I am still not able to connect.
I have noticed that the interface on the pocket wifi/inverter has at some stage updated and is now running firmware version.

SolaX Firmware: 3.001.02.
HAOS: 12.3
Core: 2024.5.3

My error is:

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:68
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 16:07:28 (1 occurrences)
Last logged: 16:07:28

SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 68, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 483, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 183, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

ReneMronet, 2024-05-11:
@Marc-dApice
There is a newer firmware for the Pocket Wifi (3.015.02)
And flash the newest for the inverter too.

Codiwr, 2024-05-11:
I do wonder if the error is coming from the Solax end (Modbus failure)
rather than the integration.

Do you still have to email tech support to get inverter updates, crazy!

ReneMronet, 2024-05-11:
Hier gibt es die neusten Firmware Updates für alle SolaX Komponenten.
Aber passet genau auf, welche Hardware ihr habt!

[https://app.box.com/s/3nvo7ic523fhojf8uuto105q9a8dgk9n/folder/79352264617](url)

anton4, 2024-05-11:
It is not SolaX issue as other inverters (Sofar eg.) have the same exact problem.

ReneMronet, 2024-05-11:
I have both SolaX and Sofar inverters with the latest firmware and only had the problems with the SolaX.

Of course, there could be several reasons for this.

Strinzer, 2024-05-12:
updated Home Assistant Core to  2024.5.3 and Solax Inverter Modbus to 2024.05.3, delete Solax Hub and added it again and everythings works fine. 1 device and 221 entities are there and data collected from the Solax system and shown in HA.

Strinzer, 2024-05-12:
problem is there again :-(

purewhitedesign, 2024-05-13:
Same here. It runs for a few days - see https://github.com/wills106/homeassistant-solax-modbus/issues/872#issuecomment-2104997073.
And today it doesn't work again.

wills106, 2024-05-14:
As long as you name the Inverter the same in the Integration it will pick up all of the history.

ernst10, 2024-05-19:
The following System was now running for about 2 weeks
using supervised HA
upgrade from Core 2024.4.4 to core 2024.5.1
Modebus still working
now upgrade modbus solax 2024.4.3 to 2024.5.2

Now it has crashed over night 
Updating HA core and system doesn´t help
Updating SolaxModbus doesn´t work
removing the Integration restart and add again doesn´t work

That´s a real bad situation as i wanted to use the integration to manage power what is just not consumed by the houseload to use an electrical power heater to heat my water boiler

Does anyone know a real working solution

Thanks Ernst10

ernst10, 2024-05-19:
I do have two different HA system running on two Raspi.
the one having the following versioning
using supervised HA
upgrade from Core 2024.4.4 to core 2024.5.1
Modebus still working
now upgrade modbus solax 2024.4.3 to 2024.5.2

this stopped now for about 6 hours and now started working again, without manuell intervention.
The second system has been updated today on all available new versions. Even restart doesn´t worked. 
Interesting situation, as the "old" system started working again, this one with latest updates started working as well round about the same time after a new restart of HA.

I´m wondering what has stopped working and why started again . Could there be any kind of missmatch for some time in the Solax Modbus system, not the integration?
thanks.

Ernst10

SteffenMichalke, 2024-05-21:
I switched back from the latest version to 2024.4.3 and both systems are running. In the latest version my X1 Mini was down, the X1 G4 Hybrid was fine. I have no idea, but I think if the system is good, don't change it. (or: “never touch a running system”)
What is the reason to upgrade if the system is fine?

wills106, 2024-05-21:
@SteffenMichalke for your Mini you will be getting the same error as #907 I think.
Just working on a fix.

SteffenMichalke, 2024-05-21:
THX for your answer. I think, if my system works, then it‘s fine and I don‘t need an update. 
If you solve the problem in a next version, I will read the documentation and decide for an update. 
But - all in all - good work. Thanks for the implementation.

ernst10, 2024-05-21:
@Steffen
Even when the system is running and you don´t do any updates on any integration or OP system, you may end suddenly on a not working modebus integration. you may have a look at my comments above where i described just this kind of situation. You need to go back to this beginning https://github.com/wills106/homeassistant-solax-modbus/issues/872#issuecomment-2088895010 of my comments.

@wills106 
do you work on a global fix for all open issues, as i have the feeling that there are some open "Bugs" with the same root cause perhaps. 
Can you imagine why system get stopped and worked again after several hours of "out of order" as i wrote in my last comment?

thanks
Ernst10

SteffenMichalke, 2024-05-22:
@ernst10 
BTW: I manage my power for the electrical water heater over a Power Genius 3000 and a solid state relais, out of HA via ModBus RTU directly from the inverter. Runs perfect.

ReneMronet, 2024-05-23:
Hi folks,
I wanted to give you some more information.

I have two X3-Hybrid-15 inverters in two different houses.

Yesterday I added more integrations to Home Assistant and configured them.
I also made a few changes to the SolaX Inverter Modbus integration. (Changed the name of the two devices)

I then noticed that Home Connect (Siemens) and the SolaX Inverter Modbus integration are no longer working. (No more data was delivered.

After restarting Home Assistant, Home Connect worked again, but the SolaX Inverter Modbus Integration did not receive any data from either inverter.

Unfortunately, downgrading the Solax Inverter Modbus Integration or reinstalling etc. did not help.

The only thing that helped was unplugging and plugging in the Pocket-Wifi V3.0 or, in the case of the second inverter, unplugging and plugging in the Pocket-Wifi+LAN V3.0.
Both Pocket are integrated into the home network via WLAN.

Data was sent again after approx. 30 seconds.

All SolaX components (inverter, Pocket-Wifi and Pocket-Wifi+LAN) have the latest firmware version.

SolaX Inverter Modbus Version: 2024.05.5
HACS: 1.34.0
Home Assistant:
Core: 2024.5.4
Supervisor: 2024.05.1
Operating System: 12.3
Frontend: 20240501.1

All integrations are currently working normally again.

LG
Rene

mc4wrk, 2024-05-24:
> ### Describe the bug
> After updating to the new software (2024.04.3) I'm no longer able to get any data from my SolaX modbus integration. all sensors are just unavailable. Also I tried to roll back [...]

Same here.
After update all my sensors are gone. :(
I'm very sorry but I only use this integration to check the system "Locally". 
Thank you for your support.

Codiwr, 2024-05-24:
This is still happening to me about every 72 hours, despite all the
suggested solutions.

Temporary solution is straightforward- unbind, repair on router and restart
HA, but can only be done if home with local WIFI connection to the router.

Still hoping on a proper fix

Codiwr, 2024-05-24:
Inverter not router*

wills106, 2024-05-26:
The Wiki does mention [unplugging PocketWiFi](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30) when it stops working. I don't own a PocketWiFi device so I don't know what's wrong with them.

For people having this issue are you on an older [Firmware Version](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#how-to-find-firmware-version) for your PocketWiFi?

Possibly fixed in later Firmware versions.

SteffenMichalke, 2024-05-26:
Pocket WiFi is not the problem. With newest core 2024.5.5 and SolaX 2024.5.5 x1 Mini is fine and x1 hybrid G4 gets no data.  Via a Modbus software datapoints are available. 
I have no idea.

SteffenMichalke, 2024-05-26:
Restart the WiFi dongle - same problem -  login via browser. Everything is ok. Switch my screen to HA. Now it works. 
Why? - I don’t know.

Codiwr, 2024-05-26:
Lasted less than 48hrs before losing entities this time, have upgraded
inverter, pocket wifi, integration and HA to latest versions

ernst10, 2024-05-26:
@wills106 
The pocket WIFI is not generating the problem.
I do have two HA running side by side and both are connecting to the same pocket WIFI. 
Just a few hours ago, one system lost connection and sensors are lost. The second system is just running fine. the system "down" is on latest version for HA and integration and worked fine as well.

thanks Ernst10

SteffenMichalke, 2024-05-26:
I think, one hole is closed, a second is opened. 
My systems are running now - both. We will see. 
It remains exciting.

SteffenMichalke, 2024-05-27:
This morning - same problem, X1 Hybrid G4 was down, X1 Mini was running. Restart the wifi dongle, delete the hub and reinstall - works. But how long?

Codiwr, 2024-05-27:
Unplugging the dongle and reloading the integration without HA restart
worked today!

Just changed DHCP to static ip in case that makes a difference (although
never seen the ip address change previously)

ernst10, 2024-05-28:
Restored backup from second HA on failing system and it´s working now again. Have not changed any TCP parameter or what else

[34 more replies]