wills106/homeassistant-solax-modbus #876: [Bug]: Detected blocking call to import_module inside the event loop by custom integration 'solax_modbus'

wills106, 2024-05-02:
### Describe the bug

Detected blocking call to import_module inside the event loop by custom integration 'solax_modbus'

Relates to this PR https://github.com/home-assistant/core/pull/114488

### Integration Version

Probably all

### Homeassistant core version

2024.5.0

### Inverter brand

All

### Plugin used

plugin_solax.py

### Serial prefix

H1E5

### Inverter firmware versions

ARM 3.28, DSP 3.32

### Connection Method

Gen3 Ethernet

### Dongle firmware

N/A

### Detailed Error Log

`2024-05-02 07:52:18.574 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to import_module inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 143: plugin = importlib.import_module( (offender: /config/custom_components/solax_modbus/__init__.py, line 143: plugin = importlib.import_module(), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues`

### Additional context

Doesn't seem to break anything at the moment and it's just a warning.

Will look into a fix

PatrikTrestik, 2024-05-09:
Will look on it next week.

pickeringar, 2024-05-11:
Also seeing this:
Logger: homeassistant.util.loop
Source: util/loop.py:99
First occurred: 10 May 2024 at 21:49:43 (1 occurrences)
Last logged: 10 May 2024 at 21:49:43
Detected blocking call to import_module inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 142: plugin = importlib.import_module( (offender: /config/custom_components/solax_modbus/__init__.py, line 142: plugin = importlib.import_module(), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

Inverter X1 Hybrid G3
Serial Prefix H1E5
Ethernet connected
Solax ARM 3.35
Solax DSP 3.42

Rpi 4
HA Core 2024.05.3
Supervisor 2024.05.1
Operating system 12.3

PatrikTrestik, 2024-05-11:
Solved by https://github.com/wills106/homeassistant-solax-modbus/pull/890

wills106, 2024-05-13:
Just released [2024.05.4b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.4b1) with the change.

I'll release a full version soon.

wills106, 2024-05-18:
Released [2024.05.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.4)

closed 2024-05-18 by wills106