wills106/homeassistant-solax-modbus #878: [question]: after the update to 2024.5.1, 49 entities remained unknown

matomatusov, 2024-05-03:
### Describe the bug

Hi, after updating to version 2024.5.1, some entities are not displayed. Is it intentional?
[image]

### Integration Version

2024.05.1

### Homeassistant core version

2024.5.0

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM1.39 DSP1.40

### Connection Method

Waveshare adapter

### Dongle firmware

1.002.09

gec75, 2024-05-03:
This is probably related to "internal" entities, they were just shadow of other entities (select, number). If yes, it's safe to delete them.

wills106, 2024-05-03:
The change is Intentional as per gec75's comment relating to "internal" entities.
They might disappear on the next restart. Or you can delete them individually. Or remove the Integration and add it back again, if you use the same name it will carry on as normal.

matomatusov, 2024-05-03:
Thank you for the explanation.

closed 2024-05-03 by matomatusov