wills106/homeassistant-solax-modbus #881: [Bug]: unable to write via modbus (tcp / rtu)

johnnyhero83, 2024-05-04:
### Describe the bug

For some reason I cannot not write into entity via solax modbus (tcp or rtu - doesnt matter). This is very new behaviour, it was normally working for few years now. But few days ago I updated HA core / system plus HACS custom intergation and this is not working now.  
I tried to "unlock inverter" as this is stated in FAQ. The problem is there is no "unlock inverter" button as stated in FAQ. Where do I find this "button". There is only entity in integration that is called "solax lock state". But this entity cannot be changed via modbus if writing is not working. How do I access this entity to change it into unlock not using modbus? And will this help in order to be able able to write into entities after this one is set to "unlock"? Or this is nothing related to this issue?

### Integration Version

2024.04.3

### Homeassistant core version

2024.4.4

### Inverter brand

Solax X3 Hybrid G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

DSP 1.39 ARM 1.38

### Connection Method

Solax dongle 3

### Dongle firmware

V3.001.02

### Detailed Error Log

Logger: homeassistant.components.websocket_api.http.connection
Zdroj: components/websocket_api/commands.py:239
integrace: Home Assistant WebSocket API ([dokumentace](https://www.home-assistant.io/integrations/websocket_api), [problémy](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
První výskyt: 17:42:07 (1 výskyty)
Naposledy logováno: 17:42:07

[547669662016] Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2543, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2580, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 971, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1043, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 112, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/solax_modbus/number.py", line 151, in async_set_native_value
    await self._hub.async_write_register(
  File "/config/custom_components/solax_modbus/__init__.py", line 512, in async_write_register
    return await self.async_lowlevel_write_register(unit, address, payload)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 504, in async_lowlevel_write_register
    resp = await self._client.write_register(address, payload[0], **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 183, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

### Additional context

Thank you

matomatusov, 2024-05-05:
Hi,
@wills106 
For writing, it is necessary to send the value 1AD4(hex) or 6868(dec) to the converter to the write single register 06x0000 right from the start and it will be unlocked.
Is this the case in your integration?

johnnyhero83, 2024-05-05:
> Hi,
> @wills106
> For writing, it is necessary to send the value 1AD4(hex) or 6868(dec) to the converter to the write single register 06x0000 right from the start and it will be unlocked.
> Is this the case in your integration?

Ok, thanks. Can you post some manual for dumnies how to do it? I am using integration but I dont interfere with separate registres nor I dont know where to find them and change them…. So some link to something could help. I tried to search, but not really were sucessful

wills106, 2024-05-11:
The button has changed to a select. As you get feedback if it's locked or not.

I am guessing it's showing as locked on your Inverter?

[image]

wills106, 2024-05-11:
I have updated the Docs in [2024.05.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.3)
Can you close the issue down if that solves the issue.

johnnyhero83, 2024-05-11:
> The button has changed to a select. As you get feedback if it's locked or not.
>
> I am guessing it's showing as locked on your Inverter? [...]

Thanks. The problem was the field was not responding. It was not possible to call the action “Unlocked” 
Not sure if this problem was related to recent issues with integration or not. 
But it was not possible to change that field

perfrank, 2024-08-07:
I have the same issue: All configuration settings are gray and can't be altered.
I can't find the "Lock State" setting anywhere.
Any help will be appreciated.

wills106, 2024-09-11:
@johnnyhero83 Are you still having this issue on a later version such as 2024.09.x? Or was it just the Inverter in Locked mode?

johnnyhero83, 2024-09-15:
Probably fine. I didnt restart since then. It means I got it Unlocked somehow and now it runs OK

wills106, 2024-09-15:
Ok I'll close this down then.

closed 2024-09-15 by wills106