wills106/homeassistant-solax-modbus #882: Export value adds additional 0

SillsHouse, 2024-05-05:
### Describe the bug

Set export value at 9200 it sets 920 in solax modbus. Change this in solax to 9200 it sets 92000 on home assistant
[image]
[image]

### Integration Version

2024.05.2

### Homeassistant core version

2024.5.1

### Inverter brand

Solax 

### Plugin used

plugin_solax.py

### Serial prefix

H4602A

### Inverter firmware versions

Can't find sorry

### Connection Method

Solax pocket wifi

### Dongle firmware

Wifi 3.

### Detailed Error Log

Na

### Additional context

Na

Codiwr, 2024-05-08:
I'd noticed this too. Hopefully a simple one to address!

SillsHouse, 2024-05-08:
There is a change that you can make but you have to add it in to the configuration file and every update you have to add the change.  Its on a lower report about a power factor of 10 error. This has been limiting my export to 920w from 9200w since October. Thought something had failed on a string of my solar panels.  Luckily it's been winter so not much export lost.

wills106, 2024-05-11:
Should be corrected in [2024.05.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.3)

wills106, 2024-05-18:
Closing down as I believe this is corrected now.

Feel free to reopen if needed.

closed 2024-05-18 by wills106

Codiwr, 2024-05-18:
Still showing 60,000W on 2024.5.3 after redownloaded and reconfigured
integration despite 6000W set in Solax app.