wills106/homeassistant-solax-modbus #884: [Bug]:

wartungsmonteur, 2024-05-05:
### Describe the bug

seit Version 2024.05.2 sind viele Entitäten nicht mehr verfügbar, also sind nur noch grau hinterlegt
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Integration Version

2024.05.2

### Homeassistant core version

2024.5.1

### Inverter brand

Solax X3

### Plugin used

plugin_solax.py

### Serial prefix

H34C10

### Inverter firmware versions

2.03

### Connection Method

Modbus TCP

### Dongle firmware

waveshare TCP

wartungsmonteur, 2024-05-05:
[image]
[image]
[image]
[image]
[image]

wills106, 2024-05-11:
This looks to be a duplicate of #878

wills106, 2024-05-18:
Closing down as duplicate.

closed 2024-05-18 by wills106