wills106/homeassistant-solax-modbus #885: [Bug]: can not connect,read entities and serial number

cermis, 2024-05-06:
### Describe the bug

After upgrade HA core to 2024.5.1 and upgrade your addin to 2024.05.2 this addon can not read any entities. Can not connect to dongle via TCP Modbus and read any data. 

### Integration Version

2024.05.2

### Homeassistant core version

2024.5.1

### Inverter brand

Solax X3 hybrid G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

DSP 1.40 ARM 1.39

### Connection Method

Solax pocket LAN+WiFi (new 10s with lan Modbus)

### Dongle firmware

Not know

### Detailed Error Log

[image]
[image]
[image]
[image]
[image]
[image]

Pavelstnl, 2024-05-07:
Same problem, i can't read any entity after update

cermis, 2024-05-07:
I was try to restart router, switch, wifi, dongle, inverter - no effect.
After connection dongle to wifi (was only on LAN), try to connect to wifi IP address - no effect.
downgrade HA, plugin - no effect
remove integration from HA and newly set-up with same name - no effect
upgrade firmware in inverter to 40/41 - on wifi IP address was working
after disable entity "charge EV from battery" it was start working on LAN IP address.

Reason of this was in incompatible inverter firmware (ARM+DSP) OR enabled entity for setting "charge EV from battery" which is not implemented in SolaX Modbus.

purewhitedesign, 2024-05-07:
My simple solution: Restore my backup and now I am waiting for fixed updates ;)

NeilBain, 2024-05-10:
Plus one.... As described above

danny2hats, 2024-05-10:
yeah also me

purewhitedesign, 2024-05-10:
Hi Guys!
I solved this by deleting the integration and installing it again. Now it works.
Solax Version: 2024.05.2
Core: 2024.5.2
Supervisor: 2024.05.1
Operating System: 12.3
Frontend: 20240501.1

wills106, 2024-05-11:
There were internal changes to Integration 2024.05.x

For whatever reason HA doesn't always adapt to the changes and requires removing the Inverter from the Integration page, restarting HA and re-adding it.

There might be a way to force HA to reconfigure the Integration on code change, but I don't know how.

NeilBain, 2024-05-13:
> ... requires removing the Inverter from the Integration page, restarting HA and re-adding it.
Didn't work for me... removed / restarted / confirmed integration not listed / added / restarted 
[image]
:-(

danny2hats, 2024-05-13:
Mine working again. My HACS integration was quite outdated. I upgraded it now it works.

trikidiki, 2024-05-14:
Upgraded and lost entities.
Removed inverter, reboot, re-added. Still the same
Removed. Upgraded to 2024.5.3. Re-added. Still the same.

darksider90, 2024-05-17:
Same pb.
Try to reinstall 2024.5.3 and the last BETA . always the same pb :(

wills106, 2024-05-17:
When you add the [extra logging](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log) does it give any extra info on what it's failing on with 2024.5.x

dasmith83, 2024-05-17:
I’m getting similar issues. 

I’m trying to add a new second inverter. 

Latest Ha version, latest HACS integration version. Solax x1 boost 5.0-T string inverter, on arm 1.42 (above the min 1.37 suggested in the docs), dongle is the new 10s wifi and lan version also on latest fw. Connected to my network via WiFi and dhcp reservation. 

added to the Solax integration as modbus over tcp, modbus address checked in inverter advanced settings - all tally’s up. 

My Ha errors state unrecognised inverter type serial number unknown. 
Solax cannot find serial number even not for mic. 

My other string inverter (Solax x1 mini) via waveshare is working fine still.

NeilBain, 2024-05-20:
In the end I had to completely clean down my docker instance and re-install HA and Solax integration.
This is the nuclear option for those with HA running on a docker container... proceed at your own risk!

`docker stop $(docker ps -a -q)
docker rm $(docker ps -a -q)
docker rmi $(docker images -a -q)
docker network prune
docker system prune -a
sudo apt-get remove docker docker-engine docker.io containerd runc --auto-remove
sudo rm -rf /var/lib/docker
sudo groupdel docker
sudo rm -rf /var/run/docker.sock
sudo rm -rf /usr/local/bin/docker-compose && sudo rm -rf /etc/docker && sudo rm -rf ~/.docker
sudo apt install docker.io`

Then re-install HA and rebuild your integrations.... and create a back-up this time!

wills106, 2024-05-26:
Closing down as I believe cermis has the integration working now.

closed 2024-05-26 by wills106

cermis, 2024-05-28:
Not fixed.
Today, I remove device from integration, then uninstall package from HA,
upgrade HA Core and HA OS, restart HA, newly install Solax package from
repo (clear new install), then try to setup new device, but still same
issue.
custom_components/solax_modbus/__init__.py:487 - Cancel send, because not
connected!
SolaX_stridac: attempt to read serialnumber failed at 0x300 data: None

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line
68, in async_read_serialnr
    inverter_data = await
hub.async_read_holding_registers(unit=hub._modbus_addr,
address=address, count=7)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 487,
in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py",
line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection]
Client is not connected

custom_components/solax_modbus/plugin_solax.py:80

   - SolaX_stridac: reading serial number from address 0x0 failed;
other address may succeed
   - SolaX_stridac: reading serial number from address 0x300 failed;
other address may succeed

custom_components/solax_modbus/plugin_solax.py:6925 - SolaX_stridac:
cannot find serial number, even not for MIC

custom_components/solax_modbus/plugin_solax.py:7002 - unrecognized
inverter type - serial number : unknown

wills106, 2024-05-28:
> I was try to restart router, switch, wifi, dongle, inverter - no effect. After connection dongle to wifi (was only on LAN), try to connect to wifi IP address - no effect. downgrade HA, plugin - no [...]

I thought you had it working here.

Does it sometime work and sometimes doesn't?

What firmware version does it display on the Pocket WiFi (Internal Code) is yours the newer WiFi + LAN
[SolaX FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/)

cermis, 2024-05-28:
Hello,
worked, but when I downgrade month before.
Today, I need to upgrade system and problems are back.

After upgrade, not working definitely. Not on WIFI IP and not on LAN IP.
I am using new wifi+lan dongle, internal code is 01.20 and firmware of
dongle is 1.001.20
--
Martin Čermák
E-mail: ***@***.***

cermis, 2024-05-28:
maybe help more logs.
custom_components/solax_modbus/__init__.py:475 - Unable to connect to
Inverter at 192.168.2.71:502
custom_components/solax_modbus/__init__.py:466 - Failed to connect

2024-05-28 10:34:29.731 DEBUG (MainThread) [custom_components.solax_modbus]
Trying to connect to Inverter at 192.168.2.71:502
2024-05-28 10:34:29.736 INFO (MainThread) [custom_components.solax_modbus]
Inverter connected at 192.168.2.71:502
2024-05-28 10:34:29.737 INFO (MainThread)
[custom_components.solax_modbus.plugin_solax] SolaX_stridac: trying to
determine inverter type
2024-05-28 10:34:34.280 WARNING (MainThread) [pymodbus.logging] Failed to
connect
2024-05-28 10:34:34.281 WARNING (MainThread)
[custom_components.solax_modbus] Unable to connect to Inverter at
192.168.2.71:502
2024-05-28 10:34:34.739 ERROR (MainThread) [pymodbus.logging] Cancel send,
because not connected!
2024-05-28 10:34:34.739 WARNING (MainThread)
[custom_components.solax_modbus.plugin_solax] SolaX_stridac: attempt to
read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
File "/config/custom_components/solax_modbus/plugin_solax.py", line 68, in
async_read_serialnr
inverter_data = await
hub.async_read_holding_registers(unit=hub._modbus_addr, address=address,
count=7)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/__init__.py", line 487, in
async_read_holding_registers
resp = await self._client.read_holding_registers(address, count, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py",
line 175, in async_execute
resp = await asyncio.wait_for(
^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
return await fut
^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client
is not connected
2024-05-28 10:34:34.742 WARNING (MainThread)
[custom_components.solax_modbus.plugin_solax] SolaX_stridac: reading serial
number from address 0x0 failed; other address may succeed
2024-05-28 10:34:34.743 INFO (MainThread)
[custom_components.solax_modbus.plugin_solax] Read SolaX_stridac 0x0 serial
number: None
2024-05-28 10:34:34.743 INFO (MainThread) [custom_components.solax_modbus]
Inverter is not connected, trying to connect
2024-05-28 10:34:34.743 DEBUG (MainThread) [custom_components.solax_modbus]
Trying to connect to Inverter at 192.168.2.71:502
2024-05-28 10:34:35.343 WARNING (MainThread) [pymodbus.logging] Failed to
connect
2024-05-28 10:34:35.832 INFO (MainThread) [custom_components.solax_modbus]
Inverter is not connected, trying to connect
2024-05-28 10:34:35.832 DEBUG (MainThread) [custom_components.solax_modbus]
Trying to connect to Inverter at 192.168.2.71:502
2024-05-28 10:34:39.284 ERROR (MainThread) [pymodbus.logging] Cancel send,
because not connected!
2024-05-28 10:34:39.285 WARNING (MainThread)
[custom_components.solax_modbus.plugin_solax] SolaX_inverter: attempt to
read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
File "/config/custom_components/solax_modbus/plugin_solax.py", line 68, in
async_read_serialnr
inverter_data = await
hub.async_read_holding_registers(unit=hub._modbus_addr, address=address,
count=7)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/__init__.py", line 487, in
async_read_holding_registers
resp = await self._client.read_holding_registers(address, count, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py",
line 175, in async_execute
resp = await asyncio.wait_for(
^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
return await fut
^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client
is not connected
2024-05-28 10:34:39.289 WARNING (MainThread)
[custom_components.solax_modbus.plugin_solax] SolaX_inverter: reading
serial number from address 0x300 failed; other address may succeed
2024-05-28 10:34:39.289 INFO (MainThread)
[custom_components.solax_modbus.plugin_solax] Read SolaX_inverter 0x300
serial number: None
2024-05-28 10:34:39.290 ERROR (MainThread)
[custom_components.solax_modbus.plugin_solax] SolaX_inverter: cannot find
serial number, even not for MIC
2024-05-28 10:34:39.290 ERROR (MainThread)
[custom_components.solax_modbus.plugin_solax] unrecognized inverter type -
serial number : unknown
2024-05-28 10:34:39.290 INFO (MainThread) [custom_components.solax_modbus]
next inverter check in 10sec
--
Martin Čermák
E-mail: ***@***.***

wills106, 2024-05-28:
It just looks like it gives up trying to connect as no response from the Pocket WiFi + LAN.

I think your having the same issue as https://github.com/wills106/homeassistant-solax-modbus/issues/907#issuecomment-2134431640 where the Pocket WiFi + LAN is shutting down the Modbus connection for whatever reason.

purewhitedesign, 2024-05-28:
> Hi Guys! I solved this by deleting the integration and installing it again. Now it works. Solax Version: 2024.05.2 Core: 2024.5.2 Supervisor: 2024.05.1 Operating System: 12.3 Frontend: 20240501.1

Doesnt work, so I restored my backup:
Solax Version: 2024.04.3
Core: 2024.5.1
Frontend: 20240501.0

I also do not update my system to 2024.5.5, because I don't know if the core or the integration is the issue.