wills106/homeassistant-solax-modbus #891: [Bug]: Selector for Charger / Discharger Times

pedromfa, 2024-05-12:
### Describe the bug

Unable to select other than 15 minute intervals to set the charge and discharge times. For example, i would want to allow charging the battery from the grid right until the lower tarif ends and then activate restart the next minute to optimize time. With current selection options i can only stop force charge 15 min before. Is it possible to have these options for every minute. Perhaps instead of a drop down we could have a fiel box to specify the exact time? Thank you so much for this wonderful integration.
[image]
[image]

### Integration Version

2024.05.3

### Homeassistant core version

2024.05.03

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H4372A

### Inverter firmware versions

ARM 2.03 and DSP 2.07

### Connection Method

SolaX Pocket WIFI 3.0

### Dongle firmware

SR7W8UGE2F

wills106, 2024-05-17:
This is actually a duplicate of #432 it's planned. But not had time.

In the short term you can edit the 15 to 5 for example. Then you can select in 5 mins blocks.
You could set it to 1 if needed, but you would have a million or so time slots to pick from the drop down.

https://github.com/wills106/homeassistant-solax-modbus/blob/f27482c8c763bdf42ecf7619bb9554aff00a6bbc/custom_components/solax_modbus/const.py#L315

closed 2024-05-17 by wills106

pedromfa, 2024-05-17:
Thank you @wills106 , this is exactly what i wanted. Ive changed it to every minute and its just perfect. If possible to add this on future releases so i dont have to change it all the time it would be amazing. Thank you for your great project here. ;)