wills106/homeassistant-solax-modbus #892: [Bug]: Missing Battery Entities

MJP-76, 2024-05-13:
### Describe the bug

After updating from 2024.04.03 to 2024.05.03 the following Batterty entities disappear / become unavailable
- Battery Charge Max Current
- Battery Minimum Capacity
- Charger Use Mode
- Lock State

There maybe others but I dont use them so cant advise

If I roll back to 2024.04.3 they start working again

My units SK-SU5000E do not have wifi dongles but built in ethernet and wifi. This is happening on my two units which have a very slight difference

### Integration Version

2024.05.3

### Homeassistant core version

2024.5.3

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

U50ED7047GB036

### Inverter firmware versions

arm 3.31 dsp 3.37

### Connection Method

onboard ethernet

### Dongle firmware

Solax_SI_CH_2nd_20160912_DE02

MJP-76, 2024-05-13:
Not the disabaled by default sensors - Atleast I dont think so as they are sensors you would want

The number and selects seem fine, its the actual sensors that have stopped, did a upgrade to lates and they go unavailable, downgrade and they come back

- sensor.solax1_battery_charge_max_current
- sensor.solax1_battery_minimum_capacity
- sensor.solax1_charger_use_mode
- sensor.solax1_lock_state

The only one I dont really use is the lock_state, but the others are sensors

MJP-76, 2024-05-13:
Well, at least these ones should be enabled by default I think
- sensor.solax1_battery_minimum_capacity
- sensor.solax1_charger_use_mode

I recently had to use this one to fix an inbalnce on my batteries and setting it to 2amp and am sure its a enabled by default
- sensor.solax1_battery_charge_max_current

MJP-76, 2024-07-27:
Updated to 2024.05.7

Still missing the following sensors
sensor.solax_battery_minimum_capacity
sensor.solax_charger_use_mode

These are two important sensors in my automations

wills106, 2024-10-13:
They were both set to internal only.

Are you not able to monitor `number.solax_battery_minimum_capacity` and `select.solax_charger_use_mode` instead?

closed 2025-10-03 by github-actions[bot]