wills106/homeassistant-solax-modbus #900: [Bug]: 'Solax' - 'G4 X1 Hybrid' A number of sensors are showing as unavailable and have error related to blocking call

UK-JustinSmall, 2024-05-18:
### Describe the bug

Log Details (WARNING)
Logger: homeassistant.util.loop
Source: util/loop.py:99
First occurred: 10:35:19 (1 occurrences)
Last logged: 10:35:19

Detected blocking call to import_module inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 142: plugin = importlib.import_module( (offender: /config/custom_components/solax_modbus/__init__.py, line 142: plugin = importlib.import_module(), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

Additionally following entities have become unavailable after upgrading past 2024.04.2 so cannot upgrade past this version number

[image]

### Integration Version

2024.04.2

### Homeassistant core version

2024.5.4

### Inverter brand

Solax

### Plugin used

plugin_growatt.py

### Serial prefix

SOLAX

### Connection Method

Waveshare

### Detailed Error Log

2024-05-18 10:35:19.139 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to import_module inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 142: plugin = importlib.import_module( (offender: /config/custom_components/solax_modbus/__init__.py, line 142: plugin = importlib.import_module(), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

### Additional context

No additional text

wills106, 2024-05-18:
Missing entities is documented #878 
Blocking call #876

closed 2024-05-18 by wills106

UK-JustinSmall, 2024-05-19:
Thanks but it is these internal sensors that are being referenced in jinga2 code thus breaking our application. Can you please reconsider and add these entities back in please.