wills106/homeassistant-solax-modbus #903: [Question]: Does someone know how to configure this inverter so that it works with a Growatt inverter

Sicos1977, 2024-05-19:
### Describe the bug

This is not a bug but a question ... but could not find a place to ask this.

I know that the standard Waveshare RS485 to ethernet module does not work but I have this one --> https://www.waveshare.com/RS232-485-TO-ETH.htm

It supports modbus_tcp 

[image]

I configured SolaxX modbus integration in HA as follow.

[image]

[image]

I see that it starts initalizing...

[image]

And I see in the waveshare module that data is received...

[image]

But nothing comes back...

I check the wiring between the waveshare module and my Growatt MOD 6000TL3-X inverter... I don't have any 120 Ohm resistors on both sides.. I can't find any documentation anywhere if this is needed in this situaties... I'm a noob when it comes to modbus.

### Integration Version

2024.05.3

### Homeassistant core version

2024.5.4

### Inverter brand

Growatt MOD 6000TL3-X

### Plugin used

plugin_growatt.py

### Serial prefix

2024.5.4

### Connection Method

RS232-485-TO-ETH WaveShare

### Additional context

.

001Tycoon, 2024-05-19:
Hello, I have a USR-W610
My settings
USR-W610
Work Mode TCP Transparent (None)

SolaxX modbus
ModBus RTU over TCP

I hope it helps

wills106, 2024-05-26:
Closing down as not an issue. Please use the discussions as mentioned.

closed 2024-05-26 by wills106