wills106/homeassistant-solax-modbus #904: Blocker: Integration crates phantom values when inverter goes offline (during night)

kai-seipp, 2024-05-20:
### Describe the bug

Hi, 

I did see this before with the Feburary versions arising so I switched back to 2024.01.08 (wich works fine). 
But I guess it‘s time to help correcting this… 

Once the inverter goes offline during night the integration after version 2024.01.x creates Phantome readings for that offline inverter proving totally insane numbers. (E.g: Solar production between 22:40-22:45h is shown with 22.4 **m**Wh. 

[image]

This continues trough out the night until the inverter comes back online once the sun comes up. 
This messes up all data and statistics. 

[image]
[image]
[image]

Let me know what I can do to resolve this as quickly as possible. 
It‘s important to me to have and keep correct historic data of my solar production in HA. 
Thanks for your support! 

### Integration Version

2024.05.04

### Homeassistant core version

2024.05.04

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

110ca2

### Connection Method

Wave Share

### Detailed Error Log

2024-05-19 22:41:06.268 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x810c: power_generation_yesterday
2024-05-19 22:41:06.268 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x810d: power_generation_this_year
2024-05-19 22:41:06.268 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x810f: power_generation_last_year
2024-05-19 22:41:06.269 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8119: pv_voltage_1
2024-05-19 22:41:06.269 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x811a: pv_current_1
2024-05-19 22:41:06.269 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x811b: pv_voltage_2
2024-05-19 22:41:06.269 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x811c: pv_current_2
2024-05-19 22:41:06.269 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8121: pv_total_power
2024-05-19 22:41:11.340 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8133: grid_voltage_l3
2024-05-19 22:41:11.341 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8134: grid_current_l1
2024-05-19 22:41:11.341 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8135: grid_current_l2
2024-05-19 22:41:11.342 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8136: grid_current_l3
2024-05-19 22:41:11.342 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8137: active_power
2024-05-19 22:41:11.343 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8139: reactive_power
2024-05-19 22:41:11.343 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x813b: apparent_power
2024-05-19 22:41:11.343 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8145: inverter_temperature
2024-05-19 22:41:11.344 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8146: grid_frequency
2024-05-19 22:41:11.344 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x8147: inverter_status
2024-05-19 22:41:16.523 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x818b: battery_charge_today
2024-05-19 22:41:16.524 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x818c: battery_charge_yesterday
2024-05-19 22:41:16.524 WARNING (MainThread) [custom_components.solax_modbus] solis-sued: read failed at 0x818d: total_battery_discharge
2024-05-19 22:41:26.830 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 614, in async_read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 793, in async_read_modbus_registers_all
File "/config/custom_components/solax_modbus/plugin_solis.py", line 134, in value_function_pv1_power

Repeats every 1-2 Minutes.

### Additional context

I have two inverters in my setup. 
Solis South and Solis North. 
There is only one Battery included on Solis south. The effect is that the north inverter totally shuts down at night. It has no power supply at all. (This just seams to be the behaviour of Solis inverters.) The south inverter has the battery attached to it and is always online.

kai-seipp, 2024-05-20:
It was just to broken. I switched back to 2023.1.8. Let me know if you need me to try something! Or if you miss any details. 
Strangely I got all wrong values all over the day now on both inverters - but much more often on the one that goes offline during night (Solis North).

wills106, 2024-05-21:
Just released [2024.05.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.5) which alters the behaviour when the Inverter goes off line / you restart HA when the Inverter is offline.

Just for info my Inverter never goes offline so I am reliant on others for solutions as in the case for [2024.05.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.5)

kai-seipp, 2024-05-22:
Thank you very much! 
I have just updated to 2024.05.05 and will let you know how it does. I did have one (few off values this night with 2024.01.8) as well. But only one at midnight.

So far communication works and both inverters are online. 🤞

kai-seipp, 2024-05-22:
Did have one more glitch today 

Log: 

```
2024-05-22 19:20:56.971 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.solis_sud_meter_activepower_total (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 1.731 seconds. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-05-22 19:30:29.270 WARNING (MainThread) [custom_components.solax_modbus] Solis-Nord: read failed at 0x810f: power_generation_last_year
2024-05-22 19:30:29.270 WARNING (MainThread) [custom_components.solax_modbus] Solis-Nord: read failed at 0x8119: pv_voltage_1
2024-05-22 19:30:29.270 WARNING (MainThread) [custom_components.solax_modbus] Solis-Nord: read failed at 0x811a: pv_current_1
2024-05-22 19:30:29.271 WARNING (MainThread) [custom_components.solax_modbus] Solis-Nord: read failed at 0x811b: pv_voltage_2
2024-05-22 19:30:29.271 WARNING (MainThread) [custom_components.solax_modbus] Solis-Nord: read failed at 0x811c: pv_current_2
2024-05-22 19:30:29.271 WARNING (MainThread) [custom_components.solax_modbus] Solis-Nord: read failed at 0x8121: pv_total_power
2024-05-22 19:30:29.900 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 619, in async_read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 798, in async_read_modbus_registers_all
File "/config/custom_components/solax_modbus/plugin_solis.py", line 134, in value_function_pv1_power
2024-05-22 20:01:54.767 WARNING (MainThread) [custom_components.solax_modbus] Solis-Nord: read failed at 0xa810: backup_mode_soc
```

[image]

kai-seipp, 2024-05-22:
[image]

Doesn't look too good. I fear it's not yet working as expected.

kai-seipp, 2024-05-24:
I switched back to 2024.1.8 as it is just a lot of work to correct all these false values of the night every day. 
I also think that the behavior of that version is better as it keeps my available, just 0. This way combined template sensors will keep working.

pulss, 2024-06-04:
Good day,

I have the same problem, I correct the values ​​in the statistics daily, because the energy panel throws me off. I'm already at a loss.

How can I revert to version 2024.1.8 please? Through HACS, it offers me the oldest version 2024.5.4.

Thank you for your help.

Pavel

kai-seipp, 2024-06-04:
Hi Pavel, 

I feel you! That's really annoying. I even get this like every other day with the 2024.1.8. But than usually only one wrong value a day. So it's a lot better! I hope this issue can be solved in the future so users like us can continue using the latest stable. 

To install a far older version you need to do that manually. Just go to the git hub repository. Go to releases (https://github.com/wills106/homeassistant-solax-modbus/releases) and look up the one you want to install. If you expand "Assets" under the desired release you will find the a .zip and a .tar.gz version of the release. Download the format you prefere and replace the current integration folder with the folder within this release file. Restart home assistant and you're done. 

The 2024.1.8 file is to find here: https://github.com/wills106/homeassistant-solax-modbus/archive/refs/tags/2024.01.8.zip

The install directory (of your current installation will probably be ../custom_components/solax_modbus/.

Hope this helps! 
Best of luck, 
Kai

> Good day,
> 
> 
> 
> I have the same problem, I correct the values ​​in the statistics daily, because the energy panel throws me off. I'm already at a loss.
> 
> 
> 
> How can I revert to version 2024.1.8 please? Through HACS, it offers me the oldest version 2024.5.4.
> 
> 
> 
> Thank you for your help.
> 
> 
> 
> Pavel

wills106, 2024-06-04:
There is an easier option https://hacs.xyz/docs/faq/select_version

wills106, 2024-06-04:
Sorry that links missing the select a different number of older versions.

wills106, 2024-06-04:
There used to be an option to override the pervious 5 or so versions. But can't find that in the docs now...