wills106/homeassistant-solax-modbus #907: [Bug]: Solax X1 Boost not working via modbus with latest 2024.05.4

mapotn, 2024-05-21:
### Describe the bug

Hi @wills106 , integration is not working with my Solax X1 boost inverter after installing latest version 2024.05.4
I have tried remove integration completely and add it again but I have no entities now.
I was working well via modbus & usb until now. 

### Integration Version

2024.05.4

### Homeassistant core version

2024.5.4

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

XB33

### Inverter firmware versions

ARM 1.50 DSP 2.32

### Connection Method

usb modbus adaptor

### Dongle firmware

3.001.02

### Detailed Error Log

Logger: homeassistant.config_entries
Zdroj: config_entries.py:772
Prvýkrát sa vyskytol: 08:50:18 (2 výskytov)
Naposledy prihlásený: 08:50:18

Error unloading entry SolaX for button
Error unloading entry SolaX for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 772, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 75, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 203, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!

Táto chyba pochádza z vlastnej integrácie.

Logger: homeassistant
Zdroj: custom_components/solax_modbus/plugin_solax.py:7014
Integrácia: SolaX Inverter Modbus (dokumentácia, problémy)
Prvýkrát sa vyskytol: 08:47:58 (1 výskytov)
Naposledy prihlásený: 08:47:58

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 302, in async_init
    self._invertertype = await self.plugin.async_determineInverterType(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 7014, in async_determineInverterType
    model_power = await _read_model_power(hub) /1000
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'

mapotn, 2024-05-21:
update: 2024.05.3 works well

MrEd350, 2024-05-21:
Updated to 2024.05.4  Connection lost/no data updates.

Removed the Integration.

Restarted HA

No data.

Integration reports No Devices and No entities.

wills106, 2024-05-21:
Yea the "Device Info" bit of 2024.05.4 has broken PV only SolaX Inverters.
Stick to 2024.05.3 till the next release comes outs. I'm working on fixing it for X1 Air, Boost, Mini & X3 MIC's.

MrEd350, 2024-05-21:
> Yea the "Device Info" bit of 2024.05.4 has broken PV only SolaX Inverters. Stick to 2024.05.3 till the next release comes outs. I'm working on fixing it for X1 Air, Boost, Mini & X3 MIC's.

Yep.  I did wonder that.  I'm a newbie so need to figure out how to roll back to the previous version....

wills106, 2024-05-21:
https://hacs.xyz/docs/faq/select_version

MrEd350, 2024-05-21:
> https://hacs.xyz/docs/faq/select_version

Thanks for your guidance.  HACS is installed and downloaded the version that worked so assuming HACS already managing the repository.

Not sure where to find the relevant card in HA on my Green RPI set up.  Clicking on the Solax integration 3 dots does not bring up any options as mention on https://hacs.xyz/docs/faq/select_version   

I must be looking in the wrong place in my ignorance/inexperience.

wills106, 2024-05-21:
Just released [2024.05.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.5) can take 15mins or more to show up in HACS

MrEd350, 2024-05-21:
> Just released [2024.05.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.5) can take 15mins or more to show up in HACS

Installed 2024.05.5

Rebooted RPi and also removed integration card.  After configuring the Integration it gets stuck on "initialising".

I know the inverter and network are still sending data as I have another instance of HA with the SOlax Modbus installed and running OK with updating data.  Both HA instances ran concurrently for two weeks and the RPI Green only broke when the update was installed earlier today.

MrEd350, 2024-05-21:
> It should correct in the morning when enough sunlight.
> 
> Your PV only inverter goes to sleep at night.

Noted.  Inverter was just about generating and not asleep.  NAS Instance of HA/Solax still updating.

I'll check in the morning.

wills106, 2024-05-21:
My Inverter hasn't been offline for about 3.5 ~ 4 years. So reliant on PR's such as #905 to fix these issues, which is in [2024.05.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.5)

MrEd350, 2024-05-22:
> > It should correct in the morning when enough sunlight.
> > Your PV only inverter goes to sleep at night.
> 
> Noted. Inverter was just about generating and not asleep. NAS Instance of HA/Solax still updating.
> 
> I'll check in the morning.

No data coming in this morning.  Generation is low due to very overcast skies but the inverter is definitely awake.

HA restarted this morning whilst the inverter was awake.  Integration re configured with same settings as before update and the same with HA running on my NAS.

Perhaps I am doing something "wrong" in the restart sequence.

Any suggestions?

Kars-de-Jong, 2024-05-23:
Strange, I have a SolaX X3-MIC (PV only) and it doesn't go offline at night, and I also don't have any issues with 2024.05.4.

boarderhype, 2024-05-25:
The integration did run with version 2024.05.3 and Core 2024.5.4. and a Solax X3-Hybrid-G4. When updating to integration 2024.05.5 and Core 2024.5.5 it got stuck initializing. On the Solax side nothing changed.  

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:80
integration: SolaX Inverter Modbus ([documentation](https://homeassistant-solax-modbus.readthedocs.io/), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 12:18:53 (29 occurrences)
Last logged: 12:27:19

SolaX: reading serial number from address 0x0 failed; other address may succeed
SolaX: reading serial number from address 0x300 failed; other address may succeed

boarderhype, 2024-05-26:
> The integration did run with version 2024.05.3 and Core 2024.5.4. and a Solax X3-Hybrid-G4. When updating to integration 2024.05.5 and Core 2024.5.5 it got stuck initializing. On the Solax side [...]

boarderhype, 2024-05-26:
After disconnecting the Solax Pocket V3.0 WiFi & LAN Dongle and connecting it again, everything works again with the latest software versions.

wills106, 2024-05-26:
The Wiki does mention [unplugging PocketWiFi](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30) when it stops working.

Are you on an older [Firmware Version](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#how-to-find-firmware-version) for your PocketWiFi?

MrEd350, 2024-05-26:
Update.

I completely removed the integration and re installed.

It connected ok but had lost the historical data except for the first few days after I originally installed it. I think this may be due to the integration using a file that was part of a back up on the same date that the data is still available.

Ironically a day later my NAS instance stopped working.

wills106, 2024-05-26:
The Historical Data is stored in the HA Database. If you named the Inverter the same as before it should carry on displaying all of the Data. Was the Database deleted?

MrEd350, 2024-05-26:
> The Historical Data is stored in the HA Database. If you named the Inverter the same as before it should carry on displaying all of the Data. Was the Database deleted?

Ah. May have give it a slightly different name.  I did look for the data failed but was not sure what it was called or where it resides.  Only one file was not deleted. From memory it was the Solax_data.json file in config directory

miezie, 2024-05-28:
> After disconnecting the Solax Pocket V3.0 WiFi & LAN Dongle and connecting it again, everything works again with the latest software versions.

Is this now the daily fix? My X1 drops daily, the X3 doesn't...

What do we need to analyse this in sense of logging and how do I get it? Then I can provide it.

wills106, 2024-05-28:
What firmware version does it display on the Pocket WiFi + LAN (Internal Code)
[SolaX FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/)

Sounds like the Pocket WiFi + LAN could be having similar issues to early PocketWiFi 3.0 where they would internally shutdown the Modbus connection. This was resolved in a later Firmware update for them.

If that's the case then you'll need to contact SolaX to ask if there is a firmware update for it.

Extra logging can be enabled [detailed-error-log](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log)

miezie, 2024-05-28:
I just requested an update and have the latest firmwares now. I willl keep you posted when issues re-occur.

miezie, 2024-05-28:
Yes, I was on 3.010.16 and I am now on 3.015.02

MrEd350, 2024-05-28:
Just an update for completeness.

A few days ago I did a full reinstall of the Integration and the data started coming in again.

A day later my HA instance on my NAS stopped seeing the data...it had been running for about 2 weeks faultlessly.

I then saw a comment to unplug the dongle on the inverter and wait 30 seconds.  I did this and everything is back to normal.

The lesson....turn it off and on again buy pulling the Dongle to save days of head scratching ;-)

Schafy, 2024-05-30:
Hi guys,

yesterday I updated the Integration to 2024.05.7. After HA restart, no data from inverter. Had to reseat (pull and replug) the wifi dongle. Than everything worked fine again. No big deal when you know what to do. Thanks to this thread ❤️

miezie, 2024-06-04:
> Do you know what version you were on and what your on now?
2024-06-04 15:27:12.533 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2024-06-04 15:27:12.533 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

I get this one on the X1.. X3 works fine... :)

wills106, 2024-08-01:
@mapotn can this be closed down now?

mapotn, 2024-08-01:
@wills106 yes, thank you

closed 2024-08-01 by wills106