wills106/homeassistant-solax-modbus #909: [Bug]: firmware sensors

pchaves1, 2024-05-22:
### Describe the bug

firmware sensors are not available anymore!?

### Integration Version

2024.05.5

### Homeassistant core version

2024.5.4

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H4752

### Inverter firmware versions

DSP v1.36 ARM v1.39

### Connection Method

Waveshare

### Dongle firmware

could not find

pchaves1, 2024-05-22:
[image]

wills106, 2024-05-22:
Aren't they displayed in the "Device info" panel as per [2024.05.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.4)?

[image]

pchaves1, 2024-05-22:
on "Device info" panel, yes.
Were the sensors removed? 
If yes, is there a way to create a sensor just with the firmware, like the one existing on the panel?

wills106, 2024-05-22:
Sensors are removed as per the release notes. The "Device info" gets updated when reloading the Integration and HA.

I'm trying to reduce the amount of registers constantly being polled.

Do you really need to check them that often?

pchaves1, 2024-05-22:
I see, so it's not a bug then. My bad.
I was using it to check when a request firmware upgrade was completed (on the history), but that's ok, I will start checking it on the panel.

closed 2024-05-26 by wills106

wills106, 2024-07-03:
Device info has live sensors in [2024.07.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b1)