wills106/homeassistant-solax-modbus #913: [Bug]: Missing firmware sensors after 2024.05.7

ivankarpov, 2024-05-28:
### Describe the bug

the following sensors are missing:

sensor.solax_bootloader_version
sensor.solax_firmware_version_modbus_tcp_major
sensor.solax_firmware_version_modbus_tcp_minor
sensor.solax_inverter_arm_firmware_major_version
sensor.solax_inverter_arm_firmware_minor_version
sensor.solax_inverter_dsp_firmware_major_version
sensor.solax_inverter_dsp_firmware_minor_version
sensor.solax_inverter_dsp_hardware_version

### Integration Version

2024.05.7

### Homeassistant core version

2024.5.5

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H4752

### Inverter firmware versions

ARM 1.40 DSP 1.43

### Connection Method

Solax Pocket wifi 3

### Dongle firmware

n/a

wills106, 2024-05-28:
The important ones are displayed in the "Device info" panel as per [2024.05.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.4)

Duplicate of #909

closed 2024-05-28 by wills106

wills106, 2024-07-03:
Device info has live sensors in [2024.07.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b1)