wills106/homeassistant-solax-modbus #916: update 2024.05.7 problem

kajik2, 2024-06-06:
### Describe the bug

Hello,

after installing the update 2024.05.7 yesterday, downloading data from the inverter does not work. :-( Home assistant Core 2024.6.0, Supervisor 2024.06.0 . Solax X3-Hybrid-G4, SRPJT5DVS8.
I updated home assistant and then modbus and since then it doesn't work

Thank you

Karel

### Integration Version

2024.05.7

### Homeassistant core version

2024.6.0

### Inverter brand

Solax X3-Hybrid-G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10IC565252

### Inverter firmware versions

Solax X3-Hybrid-G4

### Connection Method

wifi dongle ?? 2022.10.26A

### Dongle firmware

3.001.02

### Detailed Error Log

2024-06-06 00:17:53.052 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-06-06 00:17:53.078 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-06-06 00:18:03.431 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-06-06 00:18:04.941 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform tuya does not generate unique IDs. ID tuya.62450715a4e57c10c1d4smoke_sensor_status already exists - ignoring binary_sensor.solight_detektor_koure_kour
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.select] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqrelay_status already exists - ignoring select.spinac_nasi_boiler_chovani_pri_zapnuti
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.62450715a4e57c10c1d4battery_state already exists - ignoring sensor.solight_detektor_koure_stav_baterie
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqcur_current already exists - ignoring sensor.spinac_nasi_boiler_proud
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqcur_power already exists - ignoring sensor.spinac_nasi_boiler_napajeni
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqcur_voltage already exists - ignoring sensor.spinac_nasi_boiler_napeti
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bfd24a379526c1adde1t7wtemp_current already exists - ignoring sensor.bojler_1_my_teplota
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bfb262e0599504cfe3qwrttemp_current already exists - ignoring sensor.bojler_nasi_teplota
2024-06-06 00:18:04.943 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf9a5bb46df2a13b28jfsrtemp_current already exists - ignoring sensor.bojler_2_my_teplota
2024-06-06 00:18:04.944 ERROR (MainThread) [homeassistant.components.switch] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqswitch_1 already exists - ignoring switch.spinac_nasi_boiler_prepinac_1
2024-06-06 00:18:04.944 ERROR (MainThread) [homeassistant.components.switch] Platform tuya does not generate unique IDs. ID tuya.bfd24a379526c1adde1t7wswitch_1 already exists - ignoring switch.bojler_1_my_prepinac_1
2024-06-06 00:18:04.944 ERROR (MainThread) [homeassistant.components.switch] Platform tuya does not generate unique IDs. ID tuya.bfb262e0599504cfe3qwrtswitch_1 already exists - ignoring switch.bojler_nasi_prepinac_1
2024-06-06 00:18:04.944 ERROR (MainThread) [homeassistant.components.switch] Platform tuya does not generate unique IDs. ID tuya.bf9a5bb46df2a13b28jfsrswitch_1 already exists - ignoring switch.bojler_2_my_prepinac_1
2024-06-06 00:18:04.944 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform tuya does not generate unique IDs. ID tuya.62450715a4e57c10c1d4smoke_sensor_status already exists - ignoring binary_sensor.solight_detektor_koure_kour
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.select] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqrelay_status already exists - ignoring select.spinac_nasi_boiler_chovani_pri_zapnuti
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.62450715a4e57c10c1d4battery_state already exists - ignoring sensor.solight_detektor_koure_stav_baterie
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqcur_current already exists - ignoring sensor.spinac_nasi_boiler_proud
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqcur_power already exists - ignoring sensor.spinac_nasi_boiler_napajeni
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf19ba3146bdfb91b5x2qqcur_voltage already exists - ignoring sensor.spinac_nasi_boiler_napeti
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bfd24a379526c1adde1t7wtemp_current already exists - ignoring sensor.bojler_1_my_teplota
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bfb262e0599504cfe3qwrttemp_current already exists - ignoring sensor.bojler_nasi_teplota
2024-06-06 00:18:04.945 ERROR (MainThread) [homeassistant.components.sensor] Platform tuya does not generate unique IDs. ID tuya.bf9a5bb46df2a13b28jfsrtemp_current already exists - ignoring sensor.bojler_2_my_teplota
[... 507 lines cut ...]
2024-06-06 00:23:06.479 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-06-06 00:23:11.531 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-06-06 00:23:11.532 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 68, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 487, in async_read_holding_registers
    resp = await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-06-06 00:23:11.532 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-06-06 00:23:11.532 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2024-06-06 00:23:11.532 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-06-06 00:23:26.578 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-06-06 00:23:26.578 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 68, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)

wills106, 2024-06-06:
Have a look at [SolaX - FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)

Also have a look at https://github.com/wills106/homeassistant-solax-modbus/issues/872#issuecomment-2152990835

kajik2, 2024-06-06:
> Have a look at [SolaX - FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30)
> 
> Also have a look at [#872 (comment)](https://github.com/wills106/homeassistant-solax-modbus/issues/872#issuecomment-2152990835)

Hello,

So that I don't have to disconnect the wifi dongle, I first tried restarting the wifi router. Then I restarted home assistant and it really started working. But it worked for about 2 months without any problems until that update. Many thanks for the advice.

Karel

closed 2024-06-22 by wills106