wills106/homeassistant-solax-modbus #917: [Bug]: Solax 3X 10.0 kWp - G2: Modbus Error when trying to modify "Change Use Mode) Since Update to 2024.05.7

Alfred99-HA, 2024-06-07:
### Describe the bug

After Update to 2024.05.7 changing the entity "Change Use Mode" dos not work, but produces an error message
"Failed to call service select/select_option. Modbus Error: [Conntection] Not connected AsyncModbusTcpClietn 192.168.0228:508"
Displying works for all my entities, I have not checked whether other entities have the same problem when writing back a value.

https://github.com/wills106/homeassistant-solax-modbus/issues/916
[image]

### Integration Version

2024.05.7

### Homeassistant core version

20240601

### Inverter brand

Solax X3 Hybrid 10 kWp G2

### Plugin used

plugin_solis.py

### Serial prefix

H3UE10

### Connection Method

TCP / Ethernet

### Detailed Error Log

2024-06-07 08:25:26.295 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-06-07 08:26:19.373 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ '%0.1f' | format(states('sensor.solax_grid_import') | float +
states('sensor.solax_pv_power_total') | float -
states('sensor.solax_house_load') | float) }}) renders=2>
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ '%0.1f' | format(states('sensor.solax_grid_import') | float +
states('sensor.solax_pv_power_total') | float -
states('sensor.solax_house_load') | float) }}' but no default was specified
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ '%0.1f' | format(states('sensor.solax_grid_import') | float +
states('sensor.solax_pv_power_total') | float -
states('sensor.solax_house_load') | float) }}' but no default was specified
2024-06-07 08:26:19.402 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ '%0.1f' | format(states('sensor.solax_grid_import') | float +
states('sensor.solax_pv_power_total') | float -
states('sensor.solax_house_load') | float) }}' but no default was specified') while processing template 'Template<template=({{ '%0.1f' | format(states('sensor.solax_grid_import') | float +
states('sensor.solax_pv_power_total') | float -
states('sensor.solax_house_load') | float) }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.solax_battery_import_from_grid'
2024-06-07 08:26:19.404 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ '%0.1f' | format(states('sensor.solax_house_load') | float -
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ '%0.1f' | format(states('sensor.solax_house_load') | float -
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ '%0.1f' | format(states('sensor.solax_house_load') | float -
2024-06-07 08:26:19.409 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{ '%0.1f' | format(states('sensor.solax_house_load') | float -
states('sensor.heatpump_total_active_power') | float) }}' but no default was specified') while processing template 'Template<template=({{ '%0.1f' | format(states('sensor.solax_house_load') | float -
2024-06-07 08:26:19.477 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ ((states("sensor.solax_today_s_solar_energy")|int) * 100 / states("sensor.energy_production_today")|int) | round(2)}}) renders=2>
ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ ((states("sensor.solax_today_s_solar_energy")|int) * 100 / states("sensor.energy_production_today")|int) | round(2)}}' but no default was specified
homeassistant.exceptions.TemplateError: ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ ((states("sensor.solax_today_s_solar_energy")|int) * 100 / states("sensor.energy_production_today")|int) | round(2)}}' but no default was specified
2024-06-07 08:26:19.481 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ ((states("sensor.solax_today_s_solar_energy")|int) * 100 / states("sensor.energy_production_today")|int) | round(2)}}' but no default was specified') while processing template 'Template<template=({{ ((states("sensor.solax_today_s_solar_energy")|int) * 100 / states("sensor.energy_production_today")|int) | round(2)}}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.actual_vs_forecast_production_pf'
2024-06-07 08:26:42.774 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json")) (offender: /config/custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json"))), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
File "/config/custom_components/solax_modbus/__init__.py", line 377, in _refresh
File "/config/custom_components/solax_modbus/__init__.py", line 403, in async_refresh_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 619, in async_read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 795, in async_read_modbus_registers_all
File "/config/custom_components/solax_modbus/__init__.py", line 332, in loadLocalData
2024-06-07 08:26:57.780 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json")) (offender: /config/custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json"))), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
File "/config/custom_components/solax_modbus/__init__.py", line 377, in _refresh
File "/config/custom_components/solax_modbus/__init__.py", line 403, in async_refresh_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 619, in async_read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 795, in async_read_modbus_registers_all
[... 19 lines cut ...]
File "/config/custom_components/solax_modbus/__init__.py", line 619, in async_read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 795, in async_read_modbus_registers_all
File "/config/custom_components/solax_modbus/__init__.py", line 332, in loadLocalData
2024-06-07 08:29:02.757 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.0.228:502
File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option
File "/config/custom_components/solax_modbus/__init__.py", line 517, in async_write_register
File "/config/custom_components/solax_modbus/__init__.py", line 509, in async_lowlevel_write_register
File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option
File "/config/custom_components/solax_modbus/__init__.py", line 517, in async_write_register
File "/config/custom_components/solax_modbus/__init__.py", line 509, in async_lowlevel_write_register
File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option
File "/config/custom_components/solax_modbus/__init__.py", line 517, in async_write_register
File "/config/custom_components/solax_modbus/__init__.py", line 509, in async_lowlevel_write_register
2024-06-07 09:12:18.064 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.0.228:502
File "/config/custom_components/solax_modbus/select.py", line 103, in async_select_option
File "/config/custom_components/solax_modbus/__init__.py", line 517, in async_write_register
[... cut ...]
2024-06-07 09:12:18.116 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.0.228:502
2024-06-07 09:12:18.129 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.0.228:502
2024-06-07 09:12:18.146 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.0.228:502
2024-06-07 09:12:18.155 WARNING (MainThread) [custom_components.solax_modbus] Unable to connect to Inverter at 192.168.0.228:502

### Additional context

Writing values via Solax cloud works still fine.

wills106, 2024-06-22:
Have you checked the [SolaX - FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#unable-to-change-values-read-only) or can you change other parameters ok?

Alfred99-HA, 2024-08-11:
yes I did read the FAQ but could not find any hint to solve the problem.
Furthermore I updraded to version 2024.08.08 by deleting exisitng version first and installing completely new. The behaviour ist still the same. However in addition to the same problem now I lost the price information in the energy dashboard for the time before the new installation.

Looks like same as  #992 and #993
Hope for a fix and any hint for reestablishing price for historic energy values

fastsonic, 2024-08-13:
i also can´t change my Storage control switch. I do my last upgrade some days ago and a automation that switch from Selfuse to grid and Back  hangs with error on Modbus. Also my Modbus Device EW11 Elfin freeze in some Szenarios and i have to reboot the Efin.   Will check logs and jump back to an older Version.

Alfred99-HA, 2024-08-21:
I was hoping the issue has been solved in 2024.08.09, in particular since retries for modbus have been increased. However problem is still there and others have the same problem as can be seen in contributions to this issue and also issue
https://github.com/wills106/homeassistant-solax-modbus/issues/998

as far as lost energy prices are concerned I got a hint to reestablish this with https://community.home-assistant.io/t/custom-integration-to-import-long-term-statistics-from-a-file-like-csv-or-tsv/689793 . I will do this after the problem is fixed in case another delete and new install is necessary and data would be lost again.

Alfred99-HA, 2024-08-24:
I am using now version 2024.08.11 and the issue described is gone. Controlling values is again possible. Thus I will close the isseu. Others who might still have the problem - there are similar issues or need to open a new for the version 202.08.11

closed 2024-08-24 by Alfred99-HA