wills106/homeassistant-solax-modbus #918: Write DOD

jolly12f, 2024-06-08:
### Description

I would like to know if it is possible to add a box to set the DOD

### Suggested Solution

Set a checkbox to enter DOD percentage in sofar inverter

### Alternatives

No

### Additional Context

No

cschlipf, 2024-08-07:
Duplicate of #703

In theory yes, in practice no. The problem is that Sofar often requires to write a lot of registers in one chunk. In case of the DoD it's a very large chunk that includes a lot of very critical battery parameters.

Now we could read these parameters and set them again - however if there is any read error on one of the critical parameters, we are potentially setting very harmful values. This is too risky.

See also the discussion on #703

wills106, 2024-08-07:
Closing due to being a duplicate Issue

closed 2024-08-07 by wills106