wills106/homeassistant-solax-modbus #920: HA 2024.6.0 Detected blocking call to open inside the event loop by custom integration 'solax_modbus'

litinoveweedle, 2024-06-10:
### Describe the bug

After upgrade to HA 2024.6.0 during the startup new warning appear during solax_modbus integration init. Those are linked to the additional HA checks for identifying integration blocking calls in the HA event loop.

```
Log details (WARNING)
Logger: homeassistant.util.loop
Source: util/loop.py:84
First occurred: 5:45:10 PM (6 occurrences)
Last logged: 5:46:25 PM

Detected blocking call to open inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json")) (offender:
```

see log bellow.

### Integration Version

 2024.05.7

### Homeassistant core version

2024.6.0

### Inverter brand

doesn't matter

### Plugin used

plugin_growatt.py

### Serial prefix

doesn't matter

### Connection Method

doesn't matter

### Detailed Error Log

2024-06-10 17:45:25.030 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json")) (offender: /config/custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json"))), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
Traceback (most recent call last):
File "<frozen runpy>", line 198, in _run_module_as_main
File "<frozen runpy>", line 88, in _run_code
File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
sys.exit(main())
File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
exit_code = runner.run(runtime_conf)
File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
return loop.run_until_complete(setup_and_run_hass(runtime_config))
File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
self.run_forever()
File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
self._run_once()
File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
handle._run()
File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
self._context.run(self._callback, *self._args)
File "/config/custom_components/solax_modbus/__init__.py", line 377, in _refresh
await self.async_refresh_modbus_data(grp, _now)
File "/config/custom_components/solax_modbus/__init__.py", line 403, in async_refresh_modbus_data
update_result = await self.async_read_modbus_data(group)
File "/config/custom_components/solax_modbus/__init__.py", line 619, in async_read_modbus_data
res = await self.async_read_modbus_registers_all(group)
File "/config/custom_components/solax_modbus/__init__.py", line 795, in async_read_modbus_registers_all
self.loadLocalData()
File "/config/custom_components/solax_modbus/__init__.py", line 332, in loadLocalData
fp = open(self._hass.config.path(f"{self.name}_data.json"))

prehor, 2024-06-11:
Same with me. Just installed of HA 2024.6.1 and solax-modbus 2024.5.7

jamespreedy, 2024-06-12:
Same message in the logs with me since 2024.6.0.
It's remained on HASS 2024.6.1 and 2024.6.2.
Running SolaX 2024.5.7.
SolaX X3-Hybrid-G4 too.

Thanks! 🍺

litinoveweedle, 2024-06-12:
Guys you do not need to report, that this affect you setup or new versions of the HA. It affects all.

- This is new check in HA, which will be there from now forever
- This affects all and any used converter, the issue is in the using blocking call in generic part:

`fp = open(self._hass.config.path(f"{self.name}_data.json"))`

**From now th HA rule is 'No blocking calls in the event loop'**

litinoveweedle, 2024-06-12:
The potential fix is to implement fix based on HA [suggested approach](https://developers.home-assistant.io/docs/asyncio_working_with_async/):

**Calling sync functions from async**
If you are running inside an async context, it might sometimes be necessary to call a sync function. Do this like this:

```
#hub.update() is a sync function.
result = await hass.async_add_executor_job(hub.update)
```
**Starting independent task from async**
If you want to spawn a task that will not block the current async context, you can choose to create it as a task on the event loop. It will then be executed in parallel.

`hass.async_create_task(async_say_hello(hass, target))`

cfarm31, 2024-06-16:
Hi all 

I'm getting the same issue with the integration sorry if I'm ABIT slow but any ideas how to fix it?

litinoveweedle, 2024-06-16:
oh gosh... Author of the integration need to change it by fixing the code. Specifically file open needs to be executed in the async and not in the event loop. I think I documented it rather well just one post above. Did you read it? Is there anything you did not understand specifically?

@all: Please do not comment more on this issue, especially if you do not understand it. There is nothing you can do about except fix in the code. Thank you.

misakm, 2024-06-25:
Hi All,

Do I understood well that this is only warning and not affecting solax modbus integration functionalities?

Is it safe to use with HA4 2024.6.4?

Thank you, very much, Michal

wills106, 2024-06-30:
It's only a warning, I have tried to fix it. But don't understand how.

litinoveweedle, 2024-06-30:
Hello @wills106, I may try to prepare PR for you if you want. 😉 Give me please day or two.

wills106, 2024-06-30:
If you could that would be great as I really can't get my head around the issue.

I don't know if `def saveLocalData(self):` will also need turning async?

It probably doesn't help that I didn't write any of the `loadLocalData` code, so not sure 100% how to test it all.

cfarm31, 2024-06-30:
Although it is only a warning it causes the integration to stop consuming data and you have to restart HA to get it working again

wills106, 2024-06-30:
> Although it is only a warning it causes the integration to stop consuming data and you have to restart HA to get it working again

I haven't come across that issue either on my main HA or test HA.

cfarm31, 2024-06-30:
Yer it happens a few times a day to me and I have a automation in place whereby it the data is stale for 5 minutes it restarts HA 

I'm using a Solis inverter

litinoveweedle, 2024-06-30:
Same for me, to my knowledge this is only warning over potential blocking call in the main event loop. I got it in my other integrations and the behaviour was same.

litinoveweedle, 2024-06-30:
> Yer it happens a few times a day to me and I have a automation in place whereby it the data is stale for 5 minutes it restarts HA 
> 
> I'm using a Solis inverter 

I am afraid your issue is unrelated. But let's see after this warning is fixed.

cfarm31, 2024-06-30:
> > Yer it happens a few times a day to me and I have a automation in place whereby it the data is stale for 5 minutes it restarts HA
> > I'm using a Solis inverter
> 
> I am afraid your issue is unrelated. But let's see after this warning is fixed.

Weird because it started after the recent upgrades to HA core 

@wills106 what would help diagnose my issue?

litinoveweedle, 2024-06-30:
As I said, would you please wait until the warning is fixed? Btw: it is not that strange, that two unrelated issues are introduced by HA update, especially if the one is explicitly mentioned and not an issue. 🤣

litinoveweedle, 2024-06-30:
@wills106 I just checkout your repo, it seems, that you were already changing some code around line 332 in the __init__.py Is there any branch I should start with?

wills106, 2024-06-30:
I would wait till I have got 2024.07.0 Beta 1 / 2024.07.1 working.

I thought I was ready to release it following the changes to #926 but I have come across another issue with SolaX Gen4 Hybrids.

litinoveweedle, 2024-06-30:
Ok, I would wait, as I am having exactly that one myself.... :-) Please let me know when and which branch I should use.

litinoveweedle, 2024-06-30:
@wills106 

here you go. I was able to test it on my HA. Inverter is accessible and reports data. ;-) Was really easy at the end.

wills106, 2024-07-03:
This is implemented in [2024.07.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b1)

wills106, 2024-08-03:
Closing down as released [2024.08.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.1)

closed 2024-08-03 by wills106