wills106/homeassistant-solax-modbus #922: Export Control is 10x higher than configured.

Maergsche, 2024-06-12:
### Describe the bug

Hello, I have a Solax Gen3 Hybrid. The Sliders of the Export Control shows 1448w after I configured the parameter over the Solax App to 14480. 

The Value is 1/10 . 
In the Solax App Values can be set in 20w steps.

### Integration Version

2024.05.7

### Homeassistant core version

2024.6.1

### Inverter brand

Solax X3 Hybrid 8.0-D-E

### Plugin used

plugin_solax.py

### Serial prefix

H3UE08

### Inverter firmware versions

DSP v3.63 ARM v3.53

### Connection Method

Default inten

### Dongle firmware

None

Codiwr, 2024-06-15:
Another thread with this issue was closed 2 weeks ago on the assumption it had been solved - but it's still an issue.

wills106, 2024-07-03:
This is implemented in [2024.07.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b1)

Xeudoron, 2024-07-15:
Hi,

I'd like to report same behavior on inverters:

X3-Hybrid-10.0
Firmware: DSP v1.35 ARM v1.34
Hardware: Gen4
SN prefix: H34A10

X3-Hybrid-15.0
Firmware: DSP v1.42 ARM v1.41
Hardware: Gen4
SN prefix: H34A15

Integration version: 2024.05.7
HA Core version: 2024.6.4

When `export control` is set through solaxcloud to any value, then the HA modbus integration shows 10x that value for number.solax_export_control_user_limit
[image]

same on the other inverter
[image]

KOPRajs, 2024-08-09:
I'm having the same issue (export control is scaled by 0.1) with the current 2024.08.8 version.

HA: 2024.8.0
SolaX: 2024.08.8
Model: X3-Ultra
Firmware: DSP 80 ARM 14.8
Hardware: Gen5
Serial number: H3BC30JC18xxxx

Codiwr, 2024-09-20:
This has continued to be incorrect and is still the case on 2024.09.4. Anyone else still finding this?

Xeudoron, 2024-12-04:
At least for my setup, I found that I can simply set `number.solax_config_export_control_limit_readscale` to 10. So, I created a YAML file that, when needed, sets the required hard-configured values inside the inverter. **Beware of frequent writes to the inverter's Modbus memory**.

```
##=========================================================================##
## COUNTER

counter:
  custom_solax_modbus_counter:
    name: "Inverter Flash Memory Write Counter"
    step: 1
    maximum: 30
    restore: true
    icon: mdi:lifebuoy

##=========================================================================##
## TEMPLATE

template:
  ## Sensor to check if it's possible to write to inverter flash memory (condition for automation)
  - binary_sensor:
      - name: custom_solax_modbus_writes
        state: >
          {% set flash_writes_daily_limit = 20 %}
          {% set flash_writes_counter = states('counter.custom_solax_modbus_counter') | int(default=flash_writes_daily_limit) %}

          {{ flash_writes_counter < flash_writes_daily_limit }}
        availability: >
          {% from 'tools.jinja' import is_valid %}
          {{ 
            is_valid('counter.custom_solax_modbus_counter') | bool
          }}

##=========================================================================##
## AUTOMATION

automation:
  - id: custom_solax_check_config_export_control_limit_readscale
    alias: "Check and Set Export Control Limit Readscale"
    trigger:
      - platform: state
        entity_id: number.solax_config_export_control_limit_readscale
    condition:
      - condition: template
        value_template: "{{ states('number.solax_config_export_control_limit_readscale') in ['unavailable', 'unknown', 'none', None, ''] }}"
      - condition: template
        value_template: "{{ states('number.solax_config_export_control_limit_readscale') != 10 }}"
    action:
      - service: script.custom_solax_set_solax_config_export_control_limit_readscale

  - id: custom_solax_check_config_max_export
    alias: "Check and Set Max Export"
    trigger:
      - platform: state
        entity_id: number.solax_config_max_export
    condition:
      - condition: template
        value_template: "{{ states('number.solax_config_max_export') in ['unavailable', 'unknown', 'none', None, ''] }}"
      - condition: template
        value_template: "{{ states('number.solax_config_max_export') != 30000 }}"
    action:
      - service: script.custom_solax_set_solax_config_max_export

  - id: custom_solax_check_device_lock
    alias: "Check and Set Device Unlock State"
    trigger:
      - platform: state
        entity_id: select.solax_device_lock
    condition:
      - condition: template
        value_template: "{{ states('select.solax_device_lock') in ['unavailable', 'unknown', 'none', None, ''] }}"
[... 102 lines cut ...]
          option: "Unlock"
      - condition: state
        entity_id: binary_sensor.custom_solax_modbus_writes
        state: "on"
      - service: script.custom_solax_modbus_counter_increment
    mode: single

  ## Unlock the inverter
  custom_solax_set_solax_lock_state:
    alias: "Set lock_state"
    sequence:
      - condition: template
        value_template: "{{ states('select.solax_lock_state') != 'Unlocked - Advanced' }}"
      - service: select.select_option
        target:
          entity_id: select.solax_lock_state
        data:
          option: "Unlocked - Advanced"
      - condition: state
        entity_id: binary_sensor.custom_solax_modbus_writes
        state: "on"
      - service: script.custom_solax_modbus_counter_increment
    mode: single

```

closed 2025-10-03 by github-actions[bot]