wills106/homeassistant-solax-modbus #924: [Bug]: Battery to EV Charger

matomatusov, 2024-06-13:
### Describe the bug

Hi @PatrikTrestik ,
the problem is that when I want to set the EV Charger to Enabled, it switches to Disabled.
[image]

### Integration Version

2024.05.7

### Homeassistant core version

2024.6.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM1.40 DSP1.41

### Connection Method

Packet Wifi + LAN 3.0

### Dongle firmware

1.002.09

PatrikTrestik, 2024-06-13:
Working on it.

Henry1-github, 2024-08-09:
I have the same problem. :-(

PatrikTrestik, 2024-08-10:
Please check what is status of this setting in solax claud app, when you experience this bug in HA.

Does it change or not at all?

matomatusov, 2024-08-18:
If I change it, this message pops up.

Nepodařilo se provést akci select/select_option. Modbus Error: [Input/Output] ERROR: No response received after 3 retries

Sorry for the late response.

PatrikTrestik, 2024-08-19:
What does SolaX claud app? Will it change?
Can you set it through app?

matomatusov, 2024-08-19:
yes, it can be set through the application, and when I change it in HA, it is the same in the application. It doesn't change.

Henry1-github, 2024-08-19:
Yes. It is the same by me.

matomatusov, 2024-08-20:
@PatrikTrestik 
So I wondered if there would be a problem in the Modbus Register(?).
This item has been moved under modbus in the Solax application.
It was definitely not there.

PatrikTrestik, 2024-08-20:
> @PatrikTrestik So I wondered if there would be a problem in the Modbus Register(?). This item has been moved under modbus in the Solax application. It was definitely not there.

I think the same. 
Next week, when I'm home, I'll go through it.
I have latest (i hope) modbus solax doc.

Lake292, 2024-09-01:
Hi, I came across this bug as well today. One more observation, is that the value shown in drop down is actually opposite to what is set in the Solax Cloud. Not sure whether that happens to you as well. 

[image]
[image]

I can confirm as well that setting the value triggers the error in modbus mentioned already and nothing changes in Solax Cloud. My inverter is G4 10kw running 1.41/1.40.

Thank you for looking into this!

Henry1-github, 2024-09-01:
Yes. It is exactly what you write. The opposite in SC and HA.  1.40/1.39

PatrikTrestik, 2024-09-02:
There is something happening on SolaX side. I have older firmware (DSP v1.33 ARM v1.31) and this setting completely disapeared from SolaX cloud settings for me.

I'm communicating with SolaX. I ask for firmware upgrade and also for latest Modbus document.
It takes some time.

matomatusov, 2024-09-02:
That setting was moved under the modbus section.

PatrikTrestik, 2024-09-02:
I have solution, going to code.

TLDR: Write register (0x0E1) is different than read register (0x10F). Happy SolaXing 😄

PatrikTrestik, 2024-09-02:
I hope that it is fixed in  #1033 

My SolaX app does not show the option so I can only test HA entity. Please check if it is aligned with App.

Henry1-github, 2024-09-02:
Nothing has changed in my case. :-(

PatrikTrestik, 2024-09-02:
> Nothing has changed in my case. :-(

You installed from PR mamually?

Henry1-github, 2024-09-02:
> You installed from PR mamually?

What do you mean by PR?

PatrikTrestik, 2024-09-02:
> > You installed from PR mamually?
> 
> What do you mean by PR?

This is Github. It works on principles of Git, which uses Pull Requests for code changes. 
For more info see help pages.

In my comment you see that it is fixed in PR #1033.
You have to manually get that code and install it if you want to test not yet published code.

This repo owner has to prepare new release version which will contain this change.
If you are not developer, you have to wait.

Lake292, 2024-09-02:
I can confirm that this fixes it. Reading and writing is properly reflected in Solax Cloud.
Thanks again for looking into this... simple fixes are best :)

PatrikTrestik, 2024-09-02:
@wills106 please merge PR. It was tested. @Lake292  thank you for test.

wills106, 2024-09-06:
I have released [2024.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.09.1) with this fix.

matomatusov, 2024-09-06:
Hi,
thanks for fixing the problem.

closed 2024-09-06 by matomatusov