wills106/homeassistant-solax-modbus #928: [FR]: 'Add option to control the max output for a Growatt inverter without using a smart meter'

Sicos1977, 2024-06-16:
### Description

I have a smart gridmeter that I can read with Home Assistant so I was trying to find a way to control the max output of the Growatt solar inverter without buying this device --> https://www.amazon.de/-/en/EASTRON-SDM630-MODBUS-MID-Power-Meter/dp/B08X911LWF/ref=pd_day0fbt_thbs_d_sccl_1/261-4875274-2232134?pd_rd_w=SJByr&content-id=amzn1.sym.a448a60d-171e-41b3-a693-2dc17f9728e0&pf_rd_p=a448a60d-171e-41b3-a693-2dc17f9728e0&pf_rd_r=5P9XSSMQEE2Z9HMYCH5A&pd_rd_wg=CJ3dh&pd_rd_r=8e0001cb-e507-4596-ac45-6534e7bcfd5b&pd_rd_i=B08X911LWF&psc=1

This seems possible the following way if you have hooked up your Growatt inverter through modbus.

### Suggested Solution

So to control the output limit without a Growatt Smart meter you set the following modbus values:

122 to 1 = Meter 1 ... set it back to 0 to disable
123 to 1000 = Set output limit to 100%
And then set the Default Power After Exportlimit failure to the following percentage

The max output of your PV inverter (in my case 6000) * 10% = limit the power output to 600 watt

If you also use the Shine App you probably get warnings there that the Smart Meter has invalid values... just ignore it.

Set register 113 to the percentage you calculated. After setting you can read back this value and it will give the current export limit so when you for example write 50 to this register and you solar panels are not giving the max output this will give a different value ... for example 25 if the max output is half of what you requested.

So it would be nice if there is an extra field in the addon to set these output values. For example an edit box to set my max PV output to 6000 and an edit box to set the percentage I need to limit my grid export to 0.

Sicos1977, 2024-06-16:
Seems I was wrong... register 113 is just the output power in % of the max of your pv inverter. Still need to figure out how to set this max output value through modbus

gekkehenkie11, 2024-06-21:
Wouldn't that cause a lot of EEPROM writes to the growatt inverter, shortening its lifetime?  I don't think this is the way to do it. I think the correct way is to just supply the measure consumption to the inverter, like explained in this topic (regarding a solax inverter) -> https://github.com/syssi/esphome-solax-x1-mini/discussions/59

BTW I'm interested in doing the same thing as you're trying to achieve.

Sicos1977, 2024-06-21:
The Growatt inverter has to 2 writes modes. One that is stored in memory that is gone after the inverter switches of and the other mode is to EEPROM... but that is something that I never am going to use. So I don't see any problems that will give negative impact on the inverter.

FlexxFR, 2025-01-05:
Also interested, any progress with this?

flopp999, 2025-04-02:
@Sicos1977 did you try to config your Growatt through the web and did it worked?

closed 2025-10-03 by github-actions[bot]