wills106/homeassistant-solax-modbus #934: [Bug]: Multiple Solax Inverters

Bl00d-B0b, 2024-06-20:
### Describe the bug

Hello,
I use several SolaS Inverters (G2 & G4) and EV charger via Modbus, As I can see, when You add multiple inverters, they report wrong information:

[image]

My solax 8K type is X3-MIC-8K-G2, when I add it alone, then information is correct, but when I add two inverters both shows last added model and firmware inforamtion:

[image]
and 
[image]

EV Charger is showing unknow !!!

### Integration Version

2024.6.3

### Homeassistant core version

2024.6.3

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

MC208T for MIK-8K

### Inverter firmware versions

latest

### Connection Method

Pocket Wifi 3.0 to 8K, and modbus to 15K

### Dongle firmware

latest

wills106, 2024-07-03:
This should be corrected in [2024.07.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b1)

wills106, 2024-08-01:
Did you check one of the latest Beta's such as [2024.07.0b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b4)

wills106, 2024-08-03:
Closing down as released [2024.08.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.1)

closed 2024-08-03 by wills106