wills106/homeassistant-solax-modbus #940: [Bug]: 'Growatt' - 'MOD 10KTL3-XH' Priority not working

petper, 2024-06-26:
### Describe the bug

Hello,
I have a Growatt MOD 10KTL3-XH with batteries, and everything works well.
I have an issue that the priority isnt working.
As soon as i change these to anything, it takes ~10 seconds and it changes back to "Load first"
[image]

[image]

I think this is due to a registry error, so it might be needed for adjusting.
I am not really familiar with debugging for getting correct data, so I would like some help with it, so I can provide more data.

### Integration Version

2024.05.7

### Homeassistant core version

2024.6.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

015100

### Connection Method

Waveshare RS485 TO ETH (B)

### Detailed Error Log

Detected blocking call to open inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json")) (offender: /config/custom_components/solax_modbus/__init__.py, line 332: fp = open(self._hass.config.path(f"{self.name}_data.json"))), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/config/custom_components/solax_modbus/__init__.py", line 377, in _refresh await self.async_refresh_modbus_data(grp, _now) File "/config/custom_components/solax_modbus/__init__.py", line 403, in async_refresh_modbus_data update_result = await self.async_read_modbus_data(group) File "/config/custom_components/solax_modbus/__init__.py", line 619, in async_read_modbus_data res = await self.async_read_modbus_registers_all(group) File "/config/custom_components/solax_modbus/__init__.py", line 795, in async_read_modbus_registers_all self.loadLocalData() File "/config/custom_components/solax_modbus/__init__.py", line 332, in loadLocalData fp = open(self._hass.config.path(f"{self.name}_data.json"))

### Additional context

Serial prefix unable to get from HA, When enabling the entities all sensors goes offline.
Information got from Growatt site

ghost2289, 2024-07-07:
same issue with SPH inverter.
i believe those registers need to be set in groups and cannot be set individually.
annoyingly i can't seem to find details on the groupings. but i assume they are the same as they are presented on the growatt website when you are using it to apply the settings.
[image]
[image]

maybe something like the Sofar inverters batch register writes could be implemented for growatt?

ABRUUS, 2024-08-27:
I have the exact same issue to change priority on Growatt MID 20KTL3-XH (BP). Did you find a solution?

flopp999, 2024-09-02:
@wills106 
I have KTL3-XH and I know how to change the priority. What I found is that you need to set two registers then the priority can be changed. I do not know if there is another way to just change one register to set priority.
How can I help you to implement this?

petrgru, 2024-09-11:
Pleas send more info in this issues , what address changed

wills106, 2024-09-12:
> i believe those registers need to be set in groups and cannot be set individually.

@ghost2289 That's probably correct as other Inverters such as Solis need to write in groups:

https://github.com/wills106/homeassistant-solax-modbus/blob/5139f2fc8c6f0becbc3444b5e01dd01a39b97225/custom_components/solax_modbus/plugin_solis.py#L99-L108

https://github.com/wills106/homeassistant-solax-modbus/blob/5139f2fc8c6f0becbc3444b5e01dd01a39b97225/custom_components/solax_modbus/plugin_solis.py#L157-L165

> I have KTL3-XH and I know how to change the priority. What I found is that you need to set two registers then the priority can be changed.

@flopp999 Is this similar to what you have done?

People may notice Growatt contains the following, but it's not called by anything:

https://github.com/wills106/homeassistant-solax-modbus/blob/5139f2fc8c6f0becbc3444b5e01dd01a39b97225/custom_components/solax_modbus/plugin_growatt.py#L99-L108

flopp999, 2024-09-12:
> @ghost2289 That's probably correct as other Inverters such as Solis need to write in groups: [...]

Yes, that's what I do. Write to two registers otherwise it responds error

wills106, 2024-09-12:
Do you want to submit a PR with the changes? Then others can test it as well.

TonyMathiesen, 2024-10-21:
Support for controlling priority settings (time slot 1, 2 & 3) is included in version 2024.10.4. 

Time Slot 1, 2, 3 is possible to set through the integration now. With settings for active, time begin, time end and time mode for each of the 3 time slots. Similar as Growatt WebUI and App. 
[image]
[image]

There is a button under controls to send the values to the inverter:
[image]
And also one to clear all three settings (Clear Time Slots). 

Overlapping time slots will be rejected by the inverter with no update to the values. 
Under diagnostic the values will be read out on next read cycle, can be used for verification:
[image]

Note that time slot 2 & 3 is hidden by default.

petper, 2024-10-24:
> Support for controlling priority settings (time slot 1, 2 & 3) for TL3-XH (MOD) is included in version 2024.10.4.
> 
> Time Slot 1, 2, 3 is possible to set through the integration now. With settings for active, time begin, time end and time mode for each of the 3 time slots. Similar as Growatt WebUI and App. ![image](https://private-user-images.githubusercontent.com/95135900/378377549-9e8b72a0-9004-4b09-80ff-43b9bba73f79.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3ODkzNzIsIm5iZiI6MTcyOTc4OTA3MiwicGF0aCI6Ii85NTEzNTkwMC8zNzgzNzc1NDktOWU4YjcyYTAtOTAwNC00YjA5LTgwZmYtNDNiOWJiYTczZjc5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI0VDE2NTc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzOTQ5OGIwNTNkYzU1Y2EwMmU1ZmIwNTZhNzliMTk1ZWMzOTM0MTQwMGQ4MTBjMGJiMDAzMGRiMmE5YzIwNjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CuyD6S5YIzUa7-T9N9FErjtXtrQutX3XyX2DM3XvbwY) ![image](https://private-user-images.githubusercontent.com/95135900/378377105-a1991181-ae42-4445-a64a-f1867e16a7e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3ODkzNzIsIm5iZiI6MTcyOTc4OTA3MiwicGF0aCI6Ii85NTEzNTkwMC8zNzgzNzcxMDUtYTE5OTExODEtYWU0Mi00NDQ1LWE2NGEtZjE4NjdlMTZhN2UzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI0VDE2NTc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyNGQ0ODllYTc1YzA4YzIzNmI0MzJmZTI0ZWZmZTg2MzhmNDQ3NzdlNDFmNzdjMzBkOGZkYzZmMjMxNzNjZDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.qfqSb1vAXPaOtvEOoLKkkMzSHxrpzTqtuH16xyDVEt0)
> 
> There is a button under controls to send the values to the inverter: ![image](https://private-user-images.githubusercontent.com/95135900/378378084-8616fdfa-4ba2-4200-ab90-d0fbd7de521c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3ODkzNzIsIm5iZiI6MTcyOTc4OTA3MiwicGF0aCI6Ii85NTEzNTkwMC8zNzgzNzgwODQtODYxNmZkZmEtNGJhMi00MjAwLWFiOTAtZDBmYmQ3ZGU1MjFjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI0VDE2NTc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQwYWNhYjMxMmZjMjc0NWM3Yzg5YjgwMDYxMmJmOTI3MmJmZTkwZDRiZDAyNTBkOTRlNGUxZDRkM2QwZTkzMGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lbuMCaruJTPrevRLlz0rRuTNu5YeuwZXCkOMyQrEqNo) And also one to clear all three settings (Clear Time Slots).
> 
> Overlapping time slots will be rejected by the inverter with no update to the values. Under diagnostic the values will be read out on next read cycle, can be used for verification: ![image](https://private-user-images.githubusercontent.com/95135900/378378620-85d043e0-6f8f-4582-8000-de62ee9ac504.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3ODkzNzIsIm5iZiI6MTcyOTc4OTA3MiwicGF0aCI6Ii85NTEzNTkwMC8zNzgzNzg2MjAtODVkMDQzZTAtNmY4Zi00NTgyLTgwMDAtZGU2MmVlOWFjNTA0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI0VDE2NTc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjYmU5YzJjNmI2NmJmOTIzZDU5ZTMzZjU1ZTc2MWMwOGEzYTZmMmRmMDk4ZDM2NGViMTAyNTQ5YjFkNzdjNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.WuCmlcFgHa7Db2xWmlHfGFG54nf-d6K_kxx6TWEMemI)
> 
> Note that time slot 2 & 3 is hidden by default.

Wonderful development!

I just have some troubles.
[image]
[image]
I get the error:
`Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 6:47:46 PM (5 occurrences)
Last logged: 6:55:35 PM

[140116964105312] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 132, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 88, in async_press
    res = self.button_info.value_function(0, self.button_info, self._hub.data )
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 122, in value_function_time_slot_1
    time_2_begin = time_to_int(time_2_begin)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 101, in time_to_int
    hours, minutes = map(int, time_str.split(':'))
                              ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'split' `

Am i missing something? Do anyone else have this problem?

TonyMathiesen, 2024-10-24:
Hi Peter @petper,
Could you try activating the four time_2_xxx and four time 3_xxx hidden under configurations? Suspecting that if a "clean" install, then these are not known and I then have to adjust the logic a bit to handle this.

petper, 2024-10-24:
> Hi @petper, Could you try activating the four time_2_xxx and four time 3_xxx hidden under configurations? Suspecting that if a "clean" install, then these are not known and I then have to adjust the logic a bit to handle this.

You are correct!
With all time fields set ut works.

TonyMathiesen, 2024-10-24:
Thanks for confirming. I will have a look at the logic to avoid this.

JeroenZwartAssen, 2024-11-06:
[plugin_growatt.txt](https://github.com/user-attachments/files/17645763/plugin_growatt.txt)

Hello all, I wanted to add the extra time slots 4 - 9.
But when I extend the codes with extra logic, only time slot 1 - 4 is working.
Does somebody know, what I'm doing wrong?
I don't get any error message, but time slot 5-9 is not reacting and read back.