wills106/homeassistant-solax-modbus #942: [Bug]: changeless value 'SolaX Battery Charge Max Current'

t3lmoazevedo, 2024-06-28:
### Describe the bug

The register for 'SolaX Battery Charge Max Current' can't be changed now, returning always for the value manually changed on the remote settings app.
Everything else changes correctly, and the inverter is on 'Unlocked' state, just not this value anymore.

[image]

Thanks for all the suport, and I hope its not something else related, and I could be missing it.

### Integration Version

2024.05.7

### Homeassistant core version

2024.6.4

### Inverter brand

Solax X1 gen 3 hybrid

### Plugin used

plugin_solax.py

### Serial prefix

 H1E502

### Inverter firmware versions

Not available

### Connection Method

LAN

### Dongle firmware

LAN

t3lmoazevedo, 2024-06-29:
Found a usage pattern for the problem.
When the value is changed, it returns to the original one, and then jumps automaticly to the new value, exact 5 minutos after.

Example: it was 13.6A, I select 7A, and it gets back to 13.6A. After 5 minutes goes to the request 7A.
Tested several times, same behaviour with the 5 minutes interval.

[image]

wills106, 2024-06-30:
Mine exactly the same on DSP v3.32 ARM v3.28
Never noticed before as I don't alter it.

I'm guessing it's a quirk of the Inverter?

t3lmoazevedo, 2024-06-30:
It was working perfectly in sync for the last 3 years, like the other registers also.
(I charge the batteries with cheap energy at a low amperage during the night, to use the next rainy day with low solar)
Maybe collateral efect from an inverter firmware update :(

wills106, 2024-06-30:
If it was an Integration issue I would expect Battery Discharge to function the same as they work in the same manner.

Did you know there is another way to limit import / export.

https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-G23-operation-modes/#grid-export-gen-3-only

If you use a positive value you can charge that way as well. Say 2000w

Is there a reason you want to limit charging from the grid? Do you get charged more if you import above a certain value?

t3lmoazevedo, 2024-06-30:
I will try the export procedure with a positive value. Thanks for the tip.

In the last years, and having only 3.4 kW power contract from the grid, I set the inverter Charge Use Mode 'Force Time Use' during the night to use the grid to charge the battery with a limited power, since I also need more power for the home consuption that includes charging an electric vehicle.
It's also useful to use cheap tariff during the night to use that energy during the rainy days (with high price tariff).

[image]

wills106, 2024-08-03:
Closing down as not an Integration issue.

closed 2024-08-03 by wills106