wills106/homeassistant-solax-modbus #944: [FR]: Add support for X3 MEGA Inverters

jiriknez, 2024-06-30:
### Description

[Max&Mega K-TL  Modbus Protocol V4.16.pdf](https://github.com/user-attachments/files/16043599/Max.Mega.K-TL.Modbus.Protocol.V4.16.pdf)

### Suggested Solution

Add new set of modbus register for X3 MEGA inverters.

wills106, 2024-07-03:
I have added in initial support in [2024.07.0b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b1) but is your Mega a newer G2 version? As it might use different registers compared to that doc. That doc is quite old.

jiriknez, 2024-07-09:
You're right, G2 version use different registers. I'm sending new document for Mega and Forth inverters. Please could you add these registers support? 

[Solax Power X3-MEGA G2_FORTH Modbus-RTU comms protocol (2023.5.16)-1.pdf](https://github.com/user-attachments/files/16138458/Solax.Power.X3-MEGA.G2_FORTH.Modbus-RTU.comms.protocol.2023.5.16.-1.pdf)

wills106, 2024-08-02:
Just released [2024.07.0b5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b5) does it return a serial number?

If it does then replace `MAXMEG_G2` with the first 4 or 6 digits of the serial number it returns.

https://github.com/wills106/homeassistant-solax-modbus/blob/2666d6320801bfeecc24bf2c28902113d4f5e525/custom_components/solax_modbus/plugin_solax.py#L7884-L7886

jiriknez, 2024-08-02:
No, it doesn't read SN. 

custom_components/solax_modbus/plugin_solax.py:81
reading serial number from address 0x0 failed; other address may succeed
reading serial number from address 0x300 failed; other address may succeed

jiriknez, 2024-08-02:
It seems that reading stops on address 0x1a10 and return blank value: 
Read SolaxMAX 0x1a10 serial number:     
unrecognized inverter type - serial number :

jiriknez, 2024-08-02:
[home-assistant_2024-08-02T15-56-51.588Z.log](https://github.com/user-attachments/files/16472670/home-assistant_2024-08-02T15-56-51.588Z.log)

wills106, 2024-08-03:
I have released [2024.08.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.1) where the Mega / Forth has it's own plugin. Can you try that and select solx_mega_forth from the dropdown.

jiriknez, 2024-08-04:
[plugin_solax_mega_forth.zip](https://github.com/user-attachments/files/16487076/plugin_solax_mega_forth.zip)

Now it works, but I have problem with register with data type uint32 and int32. The values are too big. I also made same modifications in the plugin. Do you have idea why are values from 2bytes regiters read wrong?

wills106, 2024-08-04:
Looks like I forgot to delete "Measure Power L1" I copied it for the unit_of_measurement and class
I wasn't sure if the docs were correct on the int32's

> Do you have idea why are values from 2bytes regiters read wrong?

If it's a register showing as int16 us REGISTER_S16

jiriknez, 2024-08-04:
[image]

If I use int16 insted of int32, value is 0 as you can see on Active Power (register 183). Second line is read as REGISTER_S32 and value is nonsense (register 185). Values (Voltage and Current) with 1 bytes length are ok.

wills106, 2024-08-04:
I meant the PV ones that are showing as int16 in the docs, but are uint16 in the Integration.

I have just pushed out [2024.08.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.3) with corrections to match the docs.

Just so you know I have renamed "Today Energy" to "Today's Energy" to match the other Inverters.

jiriknez, 2024-08-04:
I don't know what's wrong now, because i see every sencor as Unknown, but it doesn't report any error. Everything seems to be ok in log, values are read.  

Please correct:
register 0x28 is used 2x
name and key reactive_power is also 2x

wills106, 2024-08-04:
I take it you meant 0x28F?

Just released [2024.08.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.4)

Had to change names on the active / reactive control.

wills106, 2024-08-04:
> no local data file found after 5 tries

That can be ignored. It's only used on the Gen4 & 5 SolaX Hybrids.

> And pls add "register_type = REG_INPUT" to Model Type and Software Version.

Have you corrected that locally? If so what do they return?

jiriknez, 2024-08-04:
I corrected it locally, but I still see Unknown and I'm not able to figure out, why. 
[image]

wills106, 2024-08-04:
If you go to the detail error log does it say it's failing on one of the registers?

If it doesn't try deleting the whole block for "Model Type", if it still doesn't work try deleting the "Software Version" as well.

"Model Type" is Red in the Modbus docs, so might not be there on all versions?

jiriknez, 2024-08-04:
I found it. Register 0x18F was set as holding. I'm sending corrected plugin. 
[plugin_solax_mega_forth(1).zip](https://github.com/user-attachments/files/16487613/plugin_solax_mega_forth.1.zip)

But, back to the start. Values from 2bytes registers are still wrong. Do you have any idea what to do with it.
[image]

wills106, 2024-08-04:
Try changing `LITTLE` to` BIG`

https://github.com/wills106/homeassistant-solax-modbus/blob/2389c7cb058b5eae527e0a4b2cc321bca7f5b490/custom_components/solax_modbus/plugin_solax_mega_forth.py#L674

jiriknez, 2024-08-04:
Yes, that's it. Thanks a lot. Now is everything works perfectly.

wills106, 2024-08-04:
Does "Model Type" & "Software Version" return anything useful?

If they do then I can add them to the Device Info panel.

Is your Inverter a 40kW version with 4 MPPT's or a larger Inverter?

jiriknez, 2024-08-04:
Yes, it does.
[image]

This tested inverter is X3-40 with 4 MPPT's. But we have also installed a lot of X3-50, X3-60 and X3-100 (110). I will integrate all of them to HA. If I will add sensors in plugin, I'll send it to you.

wills106, 2024-08-04:
I'm in the progress of adding in the other MPPT's

Just need to find the first 6 or so digits of each power type. So I can show the extra MPPT's only on the larger systems.

jiriknez, 2024-08-04:
X3G060 - X3-60 - 6 MPPT
X3G050 - X3-50 - 5 MPPT

X3-FORTH-80-110K - 9 MPPT, but now I don't know what SN starts with.

wills106, 2024-08-04:
I thought the 50 & 60 would be like that.

75 onwards could start using X3G075 or X3G75 hard to tell till we get reports of others.

jiriknez, 2024-08-04:
I think that Solax have only X3-MGA-(40-60) and X3-FTH-(80-125), none 75kW.

wills106, 2024-08-04:
It's mentioned in the Datasheet. But maybe no one sells them as so close to the 80kw?

wills106, 2024-08-04:
Don't suppose you have access to the Modbus docs for the new X1-SMART-G2?

https://github.com/wills106/homeassistant-solax-modbus/discussions/520#discussioncomment-10110669

jiriknez, 2024-08-04:
I can ask for it, it's not problem. I can send it to you in two days.

wills106, 2024-08-04:
[2024.08.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.5) contains the extra MPPT's.
It will always return the set amount for Inverter type even if some MPPT's are not connected.

I don't know if input register 0x280 returns a dynamic count, or if that's always fixed as well.

wills106, 2024-08-04:
Just realised there is an error in the setup. On a 60kW it will only display MPPT 1-4 & 6.
I need to correct it so it displays 5 as well.

Same on a 150kW. Will display MPPT 1-4 & 10-12

wills106, 2024-08-04:
Corrected in [2024.08.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.6)

jiriknez, 2024-08-04:
> I don't know if input register 0x280 returns a dynamic count, or if that's always fixed as well. 

I don't know either. After I will connect next inverters, I'll let you know.

After update, it works good.

wills106, 2024-09-07:
How complete do you think the support is for X3 Mega Inverters? Can this be closed down now?

jiriknez, 2024-09-13:
Yes, it can be closed.

closed 2024-09-13 by wills106

MultissCZ, 2025-03-06:
Hello,

I would like to ask about the status of the functionality for the Solax X3 Forth 120kW.

I need to not only read data from it but also write data. Reading works, but writing does not. Is there any issue with this?

Thank you for your response.