wills106/homeassistant-solax-modbus #948: [Bug]: HA 2024.7.x Detected code that calls async_forward_entry_setup

wills106, 2024-07-03:
### Describe the bug

Detected code that calls async_forward_entry_setup for integration solax_modbus with title: SolaX and entry_id: xyz123, during setup without awaiting async_forward_entry_setup, which can cause the setup lock to be released before the setup is done. This will stop working in Home Assistant 2025.1. Please report this issue.

### Integration Version

2024.07.0b1

### Homeassistant core version

2024.7.0b10

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H1E5

### Inverter firmware versions

ARM 3.28, DSP 3.32

### Connection Method

Gen3 Ethernet

### Dongle firmware

N/A

### Detailed Error Log

`2024-07-03 09:43:38.436 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'solax_modbus' calls async_forward_entry_setup for integration, solax_modbus with title: SolaX and entry_id: xyz123, which is deprecated and will stop working in Home Assistant 2025.6, await async_forward_entry_setups instead at custom_components/solax_modbus/__init__.py, line 339: self._hass.async_create_task(, please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues`

wills106, 2024-07-03:
There is no need for additional reports of this, or for people to also confirm they get this. Everyone on HA 2024.7 will get this.
It's the same with #920 on HA 2024.6, everyone will get the error until it's fixed.

It's just a single warning at the start of HA per Inverter configured.

wills106, 2024-07-03:
Just released [2024.07.0b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b2)

wills106, 2024-07-03:
Reopened as not fully resolved.

litinoveweedle, 2024-07-04:
> It's the same with #920 on HA 2024.6, everyone will get the error until it's fixed.

Hi, not it is not the same, this one is a new warning, but for similar topic. I can prepare fix for this one as well.

wills106, 2024-07-04:
> Hi, not it is not the same, this one is a new warning

I know it's different, I just meant everyone will get this error like everyone got the last error. So no need for people to keep reporting it.

PatrikTrestik, 2024-07-06:
> > It's the same with #920 on HA 2024.6, everyone will get the error until it's fixed.
> 
> Hi, not it is not the same, this one is a new warning, but for similar topic. I can prepare fix for this one as well.

@litinoveweedle are you working on it? I can do it, but I don't want to duplicate work.

litinoveweedle, 2024-07-06:
@PatrikTrestik sorry not yet on the task, busy this weekend with unrelated stuff, Johny Goose day you know... 🤣 I got time only for quick try to put await for the reported coroutine, but it was not enough to fix it. So if you have solution please feel free to PR.

PatrikTrestik, 2024-07-06:
> @PatrikTrestik sorry not yet on the task, busy this weekend with unrelated stuff, Johny Goose day you know... 🤣 I got time only for quick try to put await for the reported coroutine, but it was not enough to fix it. So if you have solution please feel free to PR.

Will work on it tomorrow. I'll post PR in case I have success :)

PatrikTrestik, 2024-07-06:
> Will work on it tomorrow. I'll post PR in case I have success :)

@wills106 please consider allowing contributors in this reporsitory. We can than asign ourselves to tasks to indicate workload.
It will be still you who accepts PRs, but our coordination will be easier.

litinoveweedle, 2024-07-07:
related [HA docs](https://developers.home-assistant.io/docs/config_entries_index#for-platforms)

seems `config_entries.async_forward_entry_setups` is required to be awaited:

```
        for component in PLATFORMS:
            self._hass.async_create_task(
                self._hass.config_entries.async_forward_entry_setup(
                    self.entry, component
                )
            )
```

to my understanding config_entries.async_forward_entry_setups should be coroutine and shall be therefore shall be called as indicated in HA docs:

```
        for component in PLATFORMS:
            await self._hass.config_entries.async_forward_entry_setup(
                  self.entry, component
            )
```

tested but doesn't work ;-)

PatrikTrestik, 2024-07-08:
Tested and no warning in log.

It was not obvious from HA doc, but correct is to use "plural" functions and await it.

litinoveweedle, 2024-07-08:
Blime 🤣, so shall I file PR for it?

PatrikTrestik, 2024-07-08:
> Blime 🤣, so shall I file PR for it?

Already there. I did it. See 2 lines up 😄

litinoveweedle, 2024-07-08:
Cool, thanks.👍

wills106, 2024-07-09:
> consider allowing contributor

There only seems to be an option to manually add people. Is that the normal way on a private GitHub?

PatrikTrestik, 2024-07-09:
> > consider allowing contributor
> 
> There only seems to be an option to manually add people. Is that the normal way on a private GitHub?

Yes it is. 
This is your repository and only you decide who can contribute.
Normal way is to let all people issue PR as we do now and choose contributors from them.
Only difference between "outside" user and contributor is that contributor can be assigned to work and can have higher rights in your repo (but I suggest to be carefull with PR rights atc.).
Contributor should be someone who constantly (more or less) helps maintain the code. And he should declare it 😃

litinoveweedle, 2024-07-09:
I would have one proposal as well. Looking to Patrik last PR, which brings a lot of strictly not necessary format changes, I would propose to implement CI (GitHub actions based) with black auto formatting. I did this on several HA integration projects myself and it helps tremendously to keep code clean and PR tidy.

wills106, 2024-09-20:
I had a look at VS Code plugin "Black Formatter" for python by MS, but I ended up with comments inside brackets on the end of other comments.

So not sure if the CI approach is different or I didn't have something setup correctly?

wills106, 2024-12-23:
Closing down as have "Black Formatter" added in now.

closed 2024-12-23 by wills106