wills106/homeassistant-solax-modbus #949: [Bug]: After update to 2024.07.0 Beta1 many entities change the syntax

smartminjos, 2024-07-03:
### Describe the bug

After update to 2024.07.0 Beta1 many entities change the syntax and therefore historical data are not visible any more. E.g. new syntax is sensor.solax_export_energy_today (old one was sensor.solax_today_s_export_energy. + many more . . . ) 

### Integration Version

2024.07.0b1

### Homeassistant core version

2024.6.4

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

Firmware: DSP v1.33 ARM v1.31

### Connection Method

PocketWifi 3.0

### Dongle firmware

donno

### Detailed Error Log

log nto needed, there is not error in the log

### Additional context

no

wills106, 2024-07-03:
What version were you on previously?

smartminjos, 2024-07-03:
2024.05.07 i guess the latest non-beta one

wills106, 2024-07-03:
You mustn't have reconfigured the Integration in a long time. I have gone back to August 2023 and it's been the same since.
You would have to keep going back in the releases to find out when it changed.

[image]

smartminjos, 2024-07-03:
I see your point however what i see in version 2024.05.07 (i did rollback) is different syntax on entities. If i do upgrade, i get less a bit less entities and different syntax as depicted. I am puzzled here :)))
[image]

wills106, 2024-07-03:
I see what you mean.

In 2024.07.0 Beta x it's using the `key` as it's ID entity:
https://github.com/wills106/homeassistant-solax-modbus/blob/0ce5572312549cadf67e008aeb142af1c1a477db/custom_components/solax_modbus/plugin_solax.py#L4733

But in 2024.05.7 and earlier it's converting the `name` into it's ID entity:
https://github.com/wills106/homeassistant-solax-modbus/blob/0ce5572312549cadf67e008aeb142af1c1a477db/custom_components/solax_modbus/plugin_solax.py#L4732

smartminjos, 2024-07-03:
ok, what does it mean for me now ? :) sorry for those questions, i am not IT guy and just user ....

wills106, 2024-07-03:
> i am not IT guy

That makes two of us 😅

The change probably came about through #926 
I'll try and compare the list of entities on my inverter between the two versions and see how much they have changed first.

smartminjos, 2024-07-03:
:))))

ah i see, well i noticed around maybe 10 entities where changed and I struggle to see the data after update. Mainly the one s with _s_today , battery capacity etc. 

btw heavy reading in #926 :)

Zwer2k, 2024-07-03:
I think using the key as an ID is the right way to go. It would be possible to simply adapt the few keys to the old spelling so that the statistics are not lost.

For example
```
name = "Today's Export Energy",
key = "export_energy_today",
```

change to
```
name = "Today's Export Energy",
key = "today_s_export_energy",
```

This would allow you to adjust the display names in the future if necessary without losing the statistics

wills106, 2024-07-04:
Yea, that's why I was going to extract the names from both versions to see how many need correcting.

tom3606, 2024-07-10:
Good morning, can I check if I am encountering the same issue? 

I've lost a load of entities and Home Assistant has stopped collating any of the data provided by the inverter since the 4th July (last date data was saved was 3rd July). The keys seem to have changed and no longer contain _today or quite what it was before. Unfortunately, I don't know exactly what it was so can't put them back to how they were. I can do a roll back but I have lost over a week of data.

Thanks,

Tom

romankysely, 2024-07-24:
Hi guys, I have fallen to same trap and had to go back to v.2024.05.7 Is there any progress there? Can I help somehow?

wills106, 2024-07-24:
The commit about an hour corrects most of them for SolaX
https://github.com/wills106/homeassistant-solax-modbus/commit/59b3910a461aae5ed0e5242575fc422ab393fb30

Just working on the rest at the moment and I can push a new Beta out.

wills106, 2024-07-24:
Just released [2024.07.0b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b3] hopefully correcting the SolaX ones.

Not looked into other brands yet. So please feedback.

mbeetle, 2024-07-30:
Hi . For me the old Sensor "sensor.solax_today_s_import_energy" has the new name "sensor.solax_today_s_import_energy_today"
Should this be fixed already in 2024.07.0b3 ? Unfortunately i use this Sensor as Grid Import in Energy Dashboard :(

wills106, 2024-07-31:
> For me the old Sensor "sensor.solax_today_s_import_energy" has the new name "sensor.solax_today_s_import_energy_today"

Corrected that in [2024.07.0b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.07.0b4)

wills106, 2024-08-01:
Has anyone else come across any incorrectly named entities or can I release it as a full release now?

smartminjos, 2024-08-01:
Hi i am not at home to be able to test it. Sorry

D

wills106, 2024-08-03:
I have released [2024.08.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.1) as far as I can tell they are corrected now.

smartminjos, 2024-08-03:
all works perfect !!!!!! well done

wills106, 2024-08-03:
Closing as issue resolved.

closed 2024-08-03 by wills106