wills106/homeassistant-solax-modbus #950: [Bug]: Logbook of EVC Integration is logging the Time

Philipp-X1, 2024-07-03:
### Describe the bug

I am getting a Logbook entry every 10 seconds about the RTC. ( 10 sec is my reading intervall of the integration )
So every 10 seconds it Logs the Date and Time.
I dont know if this behavior is wanted, but the Inverter Integration doesnt do this at all.

Just wondering if this logging can be removed?

### Integration Version

newest

### Homeassistant core version

newest

### Inverter brand

Solax X3-EVC-11K

### Plugin used

plugin_solax_ev_charger.py

### Serial prefix

N/A

### Inverter firmware versions

1.13

### Connection Method

Waveshare RS485 to ETH

### Dongle firmware

-

### Detailed Error Log

As mentioned here https://github.com/wills106/homeassistant-solax-modbus/issues/805 @pat asked me to open up a new Issue.
Would be grateful if you can take a look into it :)

wills106, 2024-07-03:
RTC is normally disabled by default. Have you enabled it then?

Philipp-X1, 2024-07-03:
Actually i dont know. Is there a difference between the Inverter and the EV Charger?

I dont see any possibility to enable or disable it.

I only got this control button:

[image]

[image]

wills106, 2024-07-03:
It should be disabled by default.

Click the sensor so the box pops up
Click the Gear (Settings)
Then toggle "Enabled" so it's turned off.

Philipp-X1, 2024-07-03:
Thank you! 
Yes, I did enable it back then.
Wasn’t even thinking about this…

Disabled it now and it isn’t logging anymore, thanks.

closed 2024-07-03 by Philipp-X1