wills106/homeassistant-solax-modbus #961: [Bug]: Failed to call service select/select_option. Modbus Error: [Connection] Client is not connected

02curls, 2024-07-16:
### Describe the bug

Since updating to the latest versions of 
1. this integration, 
2. the HA, 
3. the Solax wifi dongle,
 
all entities are updating and displaying perfectly fine in HA, 
but whenever I try to change any **"select." entity** (like for example select.solax3_charge_and_discharge_period2_enable) 
I now always get **"Failed to call service select/select_option. Modbus Error: [Connection] Client is not connected"** 

### Integration Version

2024.07.0b2

### Homeassistant core version

2024.7.2

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM 1.31 DSP 1.33

### Connection Method

SolaX Pocket WiFi

### Dongle firmware

3.015.02

### Detailed Error Log

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:553
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 19:20:29 (5 occurrences)
Last logged: 21:00:15

Cancel send, because not connected!

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:575
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 19:20:14 (6 occurrences)
Last logged: 21:00:05

Cancel send, because not connected!

02curls, 2024-07-17:
downgraded to Integration Version **2024.05.7 and changing "select." entities works again**,
nothing else downgraded, kept the latest versions of: Homeassistant core version 2024.7.2, Dongle firmware 3.015.02

lieglw, 2024-07-17:
Same issue here. Have 2nd instance for the same inverters running on the previous version and works.
On the upgraded one have delete integration completely and reinstalled but no inverter could be found.

02curls, 2024-07-31:
I just tried the Integration Version 2024.07.0b4, but getting the same error whenever I try to change any "select." entity (like for example select.solax3_charge_and_discharge_period2_enable)
I now always get "Failed to call service select/select_option. Modbus Error: [Connection] Client is not connected"
so downgraded to Integration Version 2024.05.7 and changing "select." entities works again,

wills106, 2024-08-01:
@02curls when updating to 2024.07.0bx did you remove the Integration, Restart HA and add it back in again?

The reason I ask is that `init.py:553` and `init.py:575` in 2024.05.7 were both in the write register section. But in 2024.07.0b4 `init.py:553` is a comment in a read_registers block. `init.py:575` is part of write_register block but again a comment.

So it looks like HA is still stuck with 2024.05.7 init file, so it's miss configured.

02curls, 2024-08-01:
@wills106 no I did not,
so now I did, in http://homeassistant.local:8123/config/integrations/integration/solax_modbus I deleted my inverter,
restarted,
upgraded to 2024.07.0b4,
restarted,
connected the inverter,
restarted,
the same problem remains

02curls, 2024-08-01:
@wills106 I now even deleted the integration in HACS completely, restarted, downloaded the 2024.07.0b4, restarted, connected the inverter, restarted, the same problem remains

lieglw, 2024-08-01:
Had the same issue. Also when delete and ....

But for me it helped unplug the dongle wait 1min plug and then add the inverter.

Nevertheless before doing this also removed the integration, restart HA and reinstall the integration and then did the unplug thing.

Interesting thing, the other HA instance still had the connection.

02curls, 2024-08-05:
@wills106 I disconnected and deleted everything now again, upgraded to 2024.08.6, reconnected,
I see all entities, 
changing values of entities does work every now and then, 
sometimes yes, but not every time, 
I am unable to find a pattern when that works and when not, 
these are the problematic entities for example: number.luxtronik2_dhw_target_temperature, select.solax_charge_and_discharge_period2_enable

I keep seeing these errors in logs:

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:549
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12:04:59 (1 occurrences)
Last logged: 12:04:59

Cancel send, because not connected!

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:571
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12:04:54 (5 occurrences)
Last logged: 12:05:34

Cancel send, because not connected!

BillyRichards, 2024-08-06:
I'm having this exact same issue as well after updating to 2024.08.6. I deleted the integration, restarted and and added the integration. Restarted again. It reads values from the inverter but most of the time cannot control the inverter, giving the Failed to call service error. I have a Gen 3 inverter which is directly connected via ethernet.

thedrjones, 2024-08-06:
I've had the same problem when trying to upgrade to 2024.08.6. Most register writes were failing with this same Modbus Error: [Connection] Client is not connected message. 

I'm on an X1-Hybrid-Gen3 (H1E302), direct via ethernet.

Reverted to 2024.05.7 in the mean time, will try again on a test instance if I can and see if Wireshark helps explain things.

BillyRichards, 2024-08-06:
I've got the same inverter, connected the same way. How did you revert to 2024.05.7? I can only see 2024.08.x or Main in the list. Downloading Main didn't help me.

Thanks,
Bill.

whitebarns, 2024-08-06:
I also have the same problem after upgrading to 2024.08.6 when issuing a sequence of commands to change the Charger Use Mode followed by changing the Manual Mode Select. 
The second call fails with
_Error: Modbus Error: [Connection] Client is not connected_
I have worked round the issue by adding a 15 second delay between the commands

BillyRichards, 2024-08-06:
Do you get the Modbus error - no response after 3 attempts for the first command? I get that, even though the change seems to actually stick. After that it's the Client not connected error.

02curls, 2024-08-06:
@wills106 

upgraded to 2024.08.7

observations:

when changing the **number**.luxtronik2_dhw_target_temperature, I IMMEDIATELY get the error "Failed to call service number/set_value. Write: Error communicating with device" 
**BUT** after on average 47 second (timed this multiple times), the value of that entity **DOES CHANGE** as entered

when chaning the **select**.solax_charge_and_discharge_period2_enable, NOT IMMEDIATELY, but with some lag, I get the same error "Failed to call service number/set_value. Write: Error communicating with device" and the state of that entity DOES **NOT** CHANGE as wished 

logs here:

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:571
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 20:55:56 (4 occurrences)
Last logged: 20:58:54

Cancel send, because not connected!

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:549
integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 20:57:32 (1 occurrences)
Last logged: 20:57:32

Cancel send, because not connected!

02curls, 2024-08-06:
@BillyRichards I did get the "no response after 3 attempts" when changing a "**select.**" entity ONCE, but many more times on that same entity I just got the "Failed to call service number/set_value. Write: Error communicating with device" and the state of that entity DID NOT CHANGE as wished

thedrjones, 2024-08-06:
> How did you revert to 2024.05.7? I can only see 2024.08.x or Main in the list.

Unfortunately HACS does not go back far enough in versions to go back that far. I reverted by removing from HACS, then manually grabbing the [2024.05.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.7) zip from the releases page and then followed the [manual install procedure](https://homeassistant-solax-modbus.readthedocs.io/en/latest/installation/#manual-installation).

BillyRichards, 2024-08-07:
Thanks, I'll give that a try later.
Cheers.

wills106, 2024-08-07:
If you got to the Integration page, HACS then configure you can change the number of previous versions.
It's defaulted to 5.

[image]

02curls, 2024-08-07:
hello @wills106 is there anything else I can further investigate/try to help solve the problem? thank you

> @wills106
>
> upgraded to 2024.08.7
>
> observations:
>
> when changing the **number**.luxtronik2_dhw_target_temperature, I IMMEDIATELY get the error "Failed to call service number/set_value. Write: Error [...]

wills106, 2024-08-07:
> "no response after 3 attempts"

I'm just looking into increasing the number of attempts. That might help.

02curls, 2024-08-07:
yes,

and yes, I remember seeing the first 2 of those errors

wills106, 2024-08-07:
There was 3 connection attempts each with 3 retires.
I have increased the retries to 6 hopefully in [2024.08.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.8)

02curls, 2024-08-07:
@wills106 not sure what that means in terms of time roughly, but as I mentioned before, 47 seconds is what it took on average

02curls, 2024-08-07:
@wills106 

upgraded to 2024.08.8, restarted, the same issue remains:

when changing (for example) the number.luxtronik2_dhw_target_temperature, I IMMEDIATELY get the error "Failed to call service number/set_value. Write: Error communicating with device" BUT after on average 47 second (timed this multiple times), the value of that entity DOES CHANGE as entered

when chaning the select.solax_charge_and_discharge_period2_enable, NOT IMMEDIATELY, but with some lag, I get the same error "Failed to call service number/set_value. Write: Error communicating with device" and the state of that entity DOES NOT CHANGE as wished at all

Codiwr, 2024-08-08:
This is affecting me as well since 08
7, no.change with 8.8.

Works for a bit after unplugging Wifi stick, then rebooting HA.

Only affects changing settings, can still view entities

Randomly fails which is a disaster as I rely on automations to stop house battery discharging when charging car on Octopus Intelligent

Very grateful if you can identify and address the problem!

Alfred99-HA, 2024-08-11:
I'm having this exact same issue as well after updating to 2024.08.6. I deleted the integration, restarted and and added the integration as suggested. Restarted again. It reads values from the inverter but I cannot control the inverter (e.g. charger use mode), giving the Failed to call service error. I have a Solax Gen 2 inverter which is directly connected via ethernet.

As a side effect of deleting and reinstalling solax modbus my engeryboard lost the price information for all times before new installation. Any hints on how to reestablish historic price information ?

mre150, 2024-08-15:
I had this same issue, I am using latest HA Core and have now rolled back Solax to 2024.05.7  
This version appears to be working with no issues.
Any of the 2024.08 versions can all read OK for me but as soon as I try to write any change to modbus it crashes the connection and requires HA to be restarted to see reads again.

Codiwr, 2024-08-25:
> Has anyone tried 2024.08.10
> or later? As that changed the client.connect behaviour back to 2024.05.7
> behaviour.

Working fine on .10 so far!

Thanks

whitebarns, 2024-08-25:
Working well for me too. 
Many thanks

wills106, 2024-09-07:
Closing down as appears to be fixed.

closed 2024-09-07 by wills106