wills106/homeassistant-solax-modbus #963: [Bug]: No Entities available

whartonr, 2024-07-21:
### Describe the bug

Integration appeared to have got stuck as no figures were changes and report house load of 1024w when there was only the normal load of around 200w being used at the time.
Restarted HA and then all entities reporting not available

### Integration Version

2024.05.7

### Homeassistant core version

2024.7.3

### Inverter brand

Solax X1 Hybird 5.0 Gen 4

### Plugin used

plugin_solax.py

### Serial prefix

H4502AI

### Inverter firmware versions

DSP v1.29 ARM v1.27

### Connection Method

SolaX Pocket WiFi 3.0

### Dongle firmware

3.006.04

### Additional context

[Solax HA Log.txt](https://github.com/user-attachments/files/16323703/Solax.HA.Log.txt)

Holkham, 2024-07-21:
I believe I have a similar issue after updating to Home Assistant Core (2024.7.3)

Wpmuc, 2024-07-22:
I think I have the same issue - there are no entities or sensors available.

jkscz, 2024-07-22:
I found: https://github.com/wills106/homeassistant-solax-modbus/issues/794
It helped me disconnect and reconnect the inverter from the LAN.

MenosChris, 2024-07-28:
Same issue

Holkham, 2024-07-29:
All sorted :-) Power cycled the house and hey presto all working

wills106, 2024-08-01:
Have a look at #872

closed 2024-08-01 by wills106