wills106/homeassistant-solax-modbus #964: [Bug]: Many entities lost after update

viktorp75, 2024-07-24:
### Describe the bug

Updated to 2024.05.7 and about 20% of all entities are gone. I had about 380 entities before and now I have 297 entities. Most of them that were lost not so important but some of them I was using, eg. solax.charger_use_mode

Lat time I updated before was around 2023.9, not sure how to find out exactly.
 

### Integration Version

2024.05.7

### Homeassistant core version

2024.7.3

### Inverter brand

Solax X3 G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM 2.03 DSP 2.07

### Connection Method

Wifi dongle 3.0

### Dongle firmware

3.001.02

wills106, 2024-07-24:
Looks like a duplicate of #909

closed 2024-08-01 by wills106

viktorp75, 2024-08-01:
> <wills106/homeassistant-solax-modbus/issue/964/issue_event/13719728540@
> github.com>

I was checking all lost entities and I do not need them, only thing I
wanted to see the mode for the inverter (SelfUse, Manual...) is there a way
to display it now when the solax.charger_use_mode
 is gone?